enjoy-digital/litex #314: We need to document LiteX CSRs!

gsomlo, 2019-12-10:
I'd like to add a doc/CSR.md file to LiteX, documenting how CSRs work from both the hardware and software (programmer's API) perspective. I'm going to add a new comment below with a .md-ish formatted document, which we can collectively work on until it's more or less ready, after which I'll post a PR to add it to LiteX. If there's a better way to edit a doc before including it under ./doc, please let me know and I can start doing that (I'm a bit awkward with github, having been used to linux's mailing list based model of development and feedback solicitation, so apologies for any clunkiness :).
Comments, feedback, and direct edits of the following comment (if github allows it) much appreciated!

gsomlo, 2019-12-10:
1. LiteX CSR definition and terminology

LiteX CSRs are MMIO Configuration/Status Registers used to interact with
and control various hardware peripheral devices included on a LiteX SoC.
NOTE: LiteX CSRs are *NOT* to be confused with the concept of CSRs as
defined within the RISC-V ISA! To further complicate things, it is
entirely possible for the LiteX SoC to be configured with a RISC-V soft
CPU, in which case the term "CSR" will be used in a RISC-V specific way
when discussing the CPU's behavior and internals, and in a LiteX specific
way (see below) when discussing "hardware peripheral MMIO registers" that
belong to the SoC, and are accessed by the CPU at the direction of various
pieces of system software (e.g., the Bios, bootloader, or OS kernel).

2. LiteX CSR Addressing

2.1. CSR Address Space: Base Address and `csr_address_width`

Currently, LiteX CSRs are placed in an address segment of size 0x1000000
bytes (16MB, see `litex/soc/integration/soc_core.py:SoCCore:__init__()`
-- search for `self.register_mem("csr", ...)`).

This segment starts at a Base Address provided as `mem_map["csr"]`, also
in SocCore, where it defaults to 0x82000000. The Base Address value may
be overridden by other components (typically CPUs) as they are added to
the SoC configuration.

Assuming an address space of 32 bits, the LiteX CSR Base Address occupies
the 8 MSBits, leaving the remaining 24 bits for addressing actual MMIO
registers within the CSR segment. However, assuming that the smallest
addressable individual memory location is 32-bit wide, that also removes
the 2 LSBits, leaving the actual number of bits for addressing individual
CSR elements (`csr_address_width`) to be max. 22 (enforced via an
`assert 2**(csr_address_width + 2) <= 0x1000000` in `SocCore:__init__()`).
**FIXME**: should this be 3 LSBs when alignment is 64?

2.2. CSR Bank ID

All CSRs that belong to the same hardware device are grouped together in
a "bank", and given a common ID or "map address". 

Bank IDs are allocated in `SoCCore:add_csr()` where one (weak) sanity
check ensures an ID can't exceed `2^csr_address_width-1` (which would
reflect the extreme/degenerate case in which each bank contains exactly
one addressable element). CSR Banks are allocated the next available ID
in a natural sequence starting at 0.

Bank IDs (as variable name `mapaddr`) are then multiplied by 0x800 (i.e.,
left-shifted by 11 bits) and added to the CSR Base Address to determine
the physical bank address in the CPU's physical address space. This means
that, effectively, we are allowed bits [23..11], or 13 bits worth of bank
IDs (remember, bits [31..24] are reserved for the overall CSR segment's
Base Address).
**FIXME**: should we modify the assert and checks in `add_csr()` to ensure a
       Bank ID can't exceed `2^13` (obviously using named parameters, not
       raw numeric constants)!!!
       Also, even 13 is only if we utilize the maximum allowed 22 bits
       of `csr_address_width`; if the latter is smaller, then that's even
       fewer than 13 Bank ID bits (again, let's parameterize this)!

A CSR Bank is selected (see `litex/soc/interconnect/csr_bus.py:CSRBank()`)
if its Bank ID (as parameter `address`) matches the CSR Bus address
MSBits, starting with bit 9 and up (See the Figure below for a graphic
example of how CSR Bus address bits are matched up against the address
bits of the CPU's MMIO Bus).

```
    Alignment (CPU XLen): 32
    CSR Base Address = 0xe000_0000
    csr_address_width = 14
    UART Bank ID = 0x04
[... 32 lines cut ...]

Regardless of whether they fit in a single subregister or require being
spread across multiple subregisters, all registers in a CSR Bank will be
laid out as a sequence of adjacent subregisters, and the number of bits
required to enumerate all subregisters will be counted as `decode_bits`
(see `litex/soc/interconnect/csr.py:GenericBank()`).
**FIXME**: we should probably assert `decode_bits <= 9` (for 32bit alignment),
       or `decode_bits <= 8` (for 64bit alignment), to ensure they fit
       within the alloted address size (and, again, parameterize, don't
       hardcode raw numeric magic constants)!!!

Once a bank is matched, see the latter part of the `__init__()` method
in `litex/soc/interconnect/csr_bus.py:CSRBank()` for how an individual
subregister is selected based on the lesser bits of the provided CSR Bus
address.
**FIXME**: right now we have an `addr_shift` of 0 (32-bit-align) or 1 (64bit)
       which determines whether the entire bank is ignored, might that have
       something to do with our "difficulties" in dumping 64bit subregs?

3. CSR Software API

**FIXME**: write this section once the above FIXMEs have been resolved. Mainly,
it is important to know how the intra-subregister endianness works, before
we can either document this behavior authoritatively, or, further use this
information to write a universal OS-kernel bus driver for the LiteX CSR Bus!

mithro, 2019-12-10:
@xobs this is probably of interest to you.

BracketMaster, 2020-02-19:
Nice!

enjoy-digital, 2020-03-04:
@gsomlo: sorry for the delay and thanks for the documentation. I'll try in the next week to create a proper wiki for the project and just prepared things with topics i'd like to document. I just put your initial description here: https://github.com/enjoy-digital/litex/wiki/CSR-Bus. Feel free to edit it and update, i'll also try to work on the FIXME. We can keep this issue for discussing things until we are happy with the CSR documentation.

BracketMaster, 2020-03-04:
@gsomlo I also have a simple working CSR example for ECP [here](https://github.com/BracketMaster/Litex_DMA_Tutorial) if you'd like to reference that.