enjoy-digital/litex #316: Code does not Build for Quartus on Linux

teezzan, 2019-12-14:
Hello,
I am kind of a noob in this space. 
I installed Litex and Migen but have been unable to build for my DE10-Lite board from Terminal. It generates a Verilog file in build folder in the same directory but always returns

```
build_top.sh: line 2: quartus_map: command not found
build_top.sh: line 3: quartus_fit: command not found
build_top.sh: line 4: quartus_asm: command not found
build_top.sh: line 5: quartus_sta: command not found
Traceback (most recent call last):
  File "s002.py", line 134, in <module>
    platform.build(module)
  File "/home/gal3li0/litex/litex/build/altera/platform.py", line 33, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/gal3li0/litex/litex/build/altera/quartus.py", line 221, in build
    _run_script(script)
  File "/home/gal3li0/litex/litex/build/altera/quartus.py", line 157, in _run_script
    raise OSError("Subprocess failed")
OSError: Subprocess failed
```
This error come from trying to run https://github.com/teezzan/my_migen_codes/blob/master/blinky/s002.py which is a modified version of https://github.com/litex-hub/fpga_101/blob/master/lab001/solutions/s002.py . 

I looked around on the internet and it was of no help to me. Is there anything I am doing wrong?
Thank you.

enjoy-digital, 2019-12-16:
Hello @teezzan,

the issue is that Quartus binaries are not in your `$PATH` and LiteX is not able yet to locate it. Doing something like export `PATH=/opt/intel/quartus/bin:$PATH` should solve your issue.

closed 2019-12-18 by enjoy-digital

teezzan, 2019-12-29:
I am very sorry for the late response. Thank you very much. It works. But it didn't flash the code. Is there another command that has to be run to flash the code unto the board?

enjoy-digital, 2019-12-30:
This will indeed just build the design. To load it, you can do something like that:
````
from litex.build.altera.programmer import USBBlaster
prog = USBBlaster()
prog.load_bitstream("bitstream.sof")
````

teezzan, 2019-12-30:
Thank you very much. That worked like a charm. I am very grateful.