enjoy-digital/litex #317: Microwatt does not build

kgugala, 2019-12-18:
I tried building it for Arty using:
```
litex/boards/targets/arty.py --cpu-type microwatt
```

the build fails with:
```
  File "/{...}/.local/lib/python3.7/site-packages/litex/soc/cores/cpu/microwatt/core.py", line 130, in add_sources
    "utils.vhdl"
TypeError: add_source() takes from 2 to 4 positional arguments but 34 were given
```
I used the latest Litex from master (commit bd15f07cf7d8bc349512d9e8dcc37f6f618ac685)

enjoy-digital, 2019-12-18:
@kgugala: thanks for reporting. This should be fixed now, can you try with:
````
./arty.py --cpu-type=microwatt --no-compile-software
````
(with microwatt repo cloned near core.py and renamed to sources, i'll add it as a submodule once things will be working)

kgugala, 2019-12-18:
@enjoy-digital the issue above is now fixed, thanks
but, I think, microwatt vhdl sources submodule is missing (#318)

closed 2019-12-18 by kgugala