enjoy-digital/litex #322: printf formatting for 64bit integers is broken on 32bit cpus

gsomlo, 2019-12-21:
I started with applying the following patch (function to test uint64_t printf formatting:
```
diff --git a/litex/soc/software/bios/main.c b/litex/soc/software/bios/main.c
index 621d377a..64c5af06 100644
--- a/litex/soc/software/bios/main.c
+++ b/litex/soc/software/bios/main.c
@@ -552,6 +552,24 @@ static void boot_sequence(void)
        }
 }
 
+static void _do_test_uint64(void)
+{
+       int i;
+       union {
+               uint8_t  foo_chr[8];
+               uint64_t foo_int;
+       } foo;
+
+       foo.foo_int = 0x0123456789abcdef;
+
+       for (i = 0; i < 8; i++)
+               printf(" %02x", foo.foo_chr[i]);
+       printf("\n");
+       printf("%016x\n", foo.foo_int);
+       printf("%016lx\n", foo.foo_int);
+       printf("%016Lx\n", foo.foo_int);
+}
+
 int main(int i, char **c)
 {
        char buffer[64];
@@ -627,6 +645,7 @@ int main(int i, char **c)
                printf("\n");
        }
 
+       _do_test_uint64();
        printf("--============= \e[1mConsole\e[0m ================--\n");
        while(1) {
                putsnonl("\e[92;1mlitex\e[0m> ");
```
I got the following results:
1. with `litex/litex/tools/litex_sim.py --cpu-type mor1kx`:
```
 01 23 45 67 89 ab cd ef
0000000001234567
0000000001234567
0000000089abcdef
```
2. with `litex/litex/tools/litex_sim.py --cpu-type vexriscv`:
```
 ef cd ab 89 67 45 23 01
0000000000000002
0000000000000002
0000000000000002
```
3. with `litex/litex/tools/litex_sim.py --cpu-type rocket`:
```
 ef cd ab 89 67 45 23 01
0000000089abcdef
0123456789abcdef
0123456789abcdef
```

I'd expect `%016Lx` to work on all three CPU models (the explicit 64bit formatting), but it's broken on 32 bit CPUs. I'll investigate when I get some spare time, but in the mean time, if anyone has any ideas for an easy fix (e.g. refresh whatever code we are using for printf in the bios), I wouldn't mind being preempted :)

geertu, 2021-05-25:
> \+       printf("%016x\n", foo.foo_int);

litex/litex/soc/software/bios/main.c: In function '_do_test_uint64':
litex/litex/soc/software/bios/main.c:90:14: warning: format '%x' expects argument of type 'unsigned int', but argument 2 has type 'uint64_t' {aka 'long long unsigned int'} [-Wformat=]
  printf("%016x\n", foo.foo_int);
          ~~~~^     ~~~~~~~~~~~
          %016llx

Note that as foo.foo_int is a 64-bit integral, it is subject to 64-bit argument passing rules, which depend on architecture and ABI (e.g. passing in a register pair, or aligning the stack to 8 bytes before passing), while "%x" assumes a 32-bit argument.
As a consequence, this may print random data from the skipped register or stack gap, instead of the intuitively expected low or high 32-bits of the value.

If you rally want to print this value as a 32-bit integer (which still gives different results depending on endianness), then please use a cast:

`printf("%016x\n", (unsigned int)foo.foo_int);
`

> \+       printf("%016lx\n", foo.foo_int);

Same issue here on 32-bit platforms.

enjoy-digital, 2021-09-27:
We are now relying on `picolibc` for `printf`so we can probably close this.

PR #988 (2021-08-13): Fix stack alignment
RISC-V requires stack to be aligned to 16 bytes.
https://github.com/riscv/riscv-elf-psabi-doc/blob/1d5384e6690ea39b5395ddc6943882ed2830dbee/riscv-elf.md?plain=1#L183

Right now, in bios/linker.ld, `_fstack` is being set to 8 bytes before the end of sram region.
```
PROVIDE(_fstack = ORIGIN(sram) + LENGTH(sram) - 8);
```

Removing ` - 8` makes it aligned to 16.

Also there are changes in crt0.S for vexriscv, vexriscv_smp and cv32e40p.

Code that was setting up stack, was adding 4 to its address for some reason.

Removing it makes it aligned to 8 bytes, and with change in bio [... cut]
closed 2021-09-27 by enjoy-digital