enjoy-digital/litex #323: Compiling BIOS without M Extension Fails for 64-Bit SoCs

scanakci, 2019-12-23:
Unfortunately, Litex BIOS results in linker error when using "-march=rv64ia   -mabi=lp64" gcc flags.  Linker error can be reproduced using Rocket Processor under standard mode. Since BlackParrot processor does not support Multiplication Extension for now, I need to use these flags.  The detailed errors are provided below:

> riscv64-unknown-elf-ld: ../libcompiler_rt/libcompiler_rt.a(udivmoddi4.o): in function `__udivmoddi4':
./litex/litex/litex/soc/software/compiler_rt/lib/builtins/udivmoddi4.c:94: undefined reference to `__ctzdi2'
riscv64-unknown-elf-ld: 
./litex/litex/litex/soc/software/compiler_rt/lib/builtins/udivmoddi4.c:96: undefined reference to `__clzdi2'
riscv64-unknown-elf-ld: .
/litex/litex/litex/soc/software/compiler_rt/lib/builtins/udivmoddi4.c:102: undefined reference to `__clzdi2'
riscv64-unknown-elf-ld: 
./litex/litex/litex/soc/software/compiler_rt/lib/builtins/udivmoddi4.c:132: undefined reference to `__ctzdi2'
riscv64-unknown-elf-ld: 
./litex/litex/litex/soc/software/compiler_rt/lib/builtins/udivmoddi4.c:136: undefined reference to `__clzdi2'
riscv64-unknown-elf-ld: 
./litex/litex/litex/soc/software/compiler_rt/lib/builtins/udivmoddi4.c:142: undefined reference to `__clzdi2'
riscv64-unknown-elf-ld: 
./litex/litex/litex/soc/software/compiler_rt/lib/builtins/udivmoddi4.c:167: undefined reference to `__clzdi2'
riscv64-unknown-elf-ld: 
./litex/litex/litex/soc/software/compiler_rt/lib/builtins/udivmoddi4.c:176: undefined reference to `__clzdi2'
./litex/litex/litex/soc/software/bios/Makefile:27: recipe for target 'bios.elf' failed"

scanakci, 2019-12-25:
with clang, I did not get linker error and generate the bios executable. Unfortunately, both blackparrot and rocket are stuck after using udivmoddi4 5-6 times. they are stuck when executing "printf(" @ %dMHz\n", CONFIG_CLOCK_FREQUENCY/1000000);" line in main.c.

mithro, 2019-12-25:
@gsomlo Might be able to help here as he did the Rocket work.

futaris, 2019-12-26:
Sounds like there is no soft implementation of clzdi2 in gcc in that build that you are using.

enjoy-digital, 2020-05-13:
@scanakci: do you still have the issue?

scanakci, 2020-05-13:
I found a temporary workaround for this issue by using a different version of udivmoddi4.c. For IA extension, user needs to replace udivmoddi4.c file in compiler-rt with a version that I provide in pythondata-cpu-blackparrot. By sourcing a [bash script](https://github.com/scanakci/litex/blob/master/litex/soc/cores/cpu/blackparrot/setEnvironment.sh#L23), a user can easily replace the file. BP started supporting M extension recently. Therefore, this copy process will not be necessary once I switch to that version. However, the bug for IA extension is still there. You can easily reproduce it with Rocket.

scanakci, 2020-05-13:
I actually forgot to mention another issue. I fixed the compilation problem by using this  [specific line](https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/bios/Makefile#L4). I had to specify the libgcc.a (generated for IA) to the linker directly.