enjoy-digital/litex #329: ICE40: Unable to modify yosys template

xobs, 2020-01-02:
With 0931ccc91985a725d305a5ca5ccc64f501745cdd it appears as though it's no longer possible to modify the Yosys template:

```sh
$ python .\foboot-bitstream.py  --revision pvt --document-only --with-debug=usb   lxbuildenv: v2019.8.19.1 (run .\foboot-bitstream.py --lx-help for help)
Traceback (most recent call last):
  File ".\foboot-bitstream.py", line 462, in <module>
    main()
  File ".\foboot-bitstream.py", line 422, in main
    soc = BaseSoC(platform, cpu_type=cpu_type, cpu_variant=cpu_variant,
  File ".\foboot-bitstream.py", line 259, in __init__
    platform.toolchain.nextpnr_yosys_template[2] += " -relut -abc2 -dffe_min_ce_use 4 -relut"
AttributeError: 'LatticeIceStormToolchain' object has no attribute 'nextpnr_yosys_template'
$
```

It used to be that we could access `platform.toolchain.nextpnr_yosys_template[]` to modify the toolchain flags.  With this patch, this no longer seems possible.

enjoy-digital, 2020-01-02:
Thanks for the feedback, i tried to not change usage when doing the changes but i probably failed on this. I'll have a look.

enjoy-digital, 2020-01-02:
In the previous code there were two templates for the yosys command: `yosys_template` (initial, was used by arachne-pnr) and `nextnpr_yosys_template`. Since i removed arachne-pnr support, i only keep a  `yosys_template` that is now the one used by the recommended flow: Yosys + Nextpnr. 

I just created a PR to foboot to update that: https://github.com/im-tomu/foboot/pull/30 This also provide the template directly to give more flexibility for things to change/evolve in LiteX without breaking foboot.

xobs, 2020-01-02:
Thanks!  I'll use that template in the workshop as well, when I bring that version forward.

closed 2020-01-02 by xobs