enjoy-digital/litex #334: Unable to load BIOS onto LiteX vexriscv core via usb_fifo

feliks-montez, 2020-01-06:
I am attempting to get litex to run on a new board (Mimas A7) and I think I'm really close (I'll submit a PR once it works). I'm pretty sure I've created the platform and target files correctly and the bitstream successfully generates. I'm able to load it onto the board with `openocd`. The main issue is that this board does not support standard UART by default. (I could reflash the EEPROM to configure the FT2232H's Channel A for UART as a last resort, but I want to avoid that if possible.) As far as I can tell, by default on the Mimas A7, Channel A is in async FIFO FT245 mode, for which, according to [this issue](https://github.com/enjoy-digital/litex/issues/231), it appears support has been added. I've mapped the necessary pins including `"usb_fifo"` in the platform file and passed `uart_name="usb_fifo"` in the target.
With the bitstream loaded, I attempt to load the bios: 
```
litex_term --serial-boot --kernel software/bios/bios.bin /dev/ttyUSB0
```
This hangs at `[LXTERM] Starting....`. If I power off the board, I get this error saying that perhaps the board was listening for something, but just never responded: [lxterm-error.txt](https://github.com/enjoy-digital/litex/files/4024885/lxterm-error.txt). I suspect it's just a disconnection error.

I know that there are a lot of layers at work here, and the problem could be just about anywhere, but is there anything obvious that I'm missing? My hunch is that the CPU core itself works, but I just can't communicate with it. Does litex_term need modified to explicitly use FT245 instead? Perhaps I'm forgetting to modify the SoC design somewhere to account for `usb_fifo` instead of `serial`?

enjoy-digital, 2020-01-06:
Hi @feliks-montez,

to be able to use the `serial-boot` the CPU should already be executing the BIOS. So you first need to build a design with the BIOS in integrated ROM, similar to Arty for example: https://github.com/enjoy-digital/litex/blob/master/litex/boards/targets/arty.py#L48-L74

Then be sure the usb_fifo is described as it's done here:
https://github.com/enjoy-digital/litex/blob/master/litex/boards/platforms/minispartan6.py#L66-L74

Then use uart_name="usb_fifo" in SoCSDRAM and build the design.

By just using `litex_term /dev/ttyUSBX`, you should see the BIOS prompt.

If it's still not working, please provide your platform/target files so that i can have a look.

feliks-montez, 2020-01-07:
So the BIOS _is_ supposed to be included in the bitstream. I'm confused about how to load the bios into the integrated ROM then. My BaseSoC class is almost a direct copy of that Arty example, except for the Mimas A7's sdram_module and passing `uart_name="usb_fifo"` through into `SoCSDRAM.__init__`. I can't find any code (explicit or implicit) in the platform definition that loads the bios into the integrated ROM, so I tried generating the bitstream with `./mimas_a7.py --uart-name usb_fifo --integrated-rom-file bios.bin`. No luck. My `"usb_fifo"` platform definition is also based directly on the Arty example.
[mimas_a7_platform.py](https://github.com/enjoy-digital/litex/files/4028885/mimas_a7_platform.py.txt)
[mimas_a7_target.py](https://github.com/enjoy-digital/litex/files/4028886/mimas_a7_target.py.txt)

enjoy-digital, 2020-01-07:
If your target is similar to Arty, you can just do: `./mimas_a7.py --uart-name usb_fifo`. The ROM initialization is done here: https://github.com/enjoy-digital/litex/blob/master/litex/soc/integration/builder.py#L172.

The reset seem active high on the on the Mimas A7, so instead of:
`self.comb += pll.reset.eq(~platform.request("cpu_reset"))`
you should use:
`self.comb += pll.reset.eq(platform.request("cpu_reset"))`

If it's still not working, i would first recommend testing with the standard UART, by reconfiguring the FTDI chip or using an external RS232<-->USB dongle.

feliks-montez, 2020-01-07:
I tried inverting the pll reset, but I still can't get a connection. I think first I'll implement an FT245 anync FIFO loopback in the FPGA and see if I can make a connection that way. That may give me some more leads. If that doesn't help, I'll probably try configuring the FT2232H Channel A as standard UART.

feliks-montez, 2020-01-14:
Got it working! Turns out the issue was that the FTDI chip wasn't actually programmed for async FIFO 245 like I thought it was. The documentation for the Mimas A7 is a little confusing to me. After using FT_PROG to set Channel A to 245 FIFO and D2XX Direct, I was able to connect using a similar technique as [this](https://rohitksingh.in/ftdi-asynchronous-fifo-usb-communication-with-fpga.html) article. In case it's helpful for anyone in the future: I generated the bitstream with ```targets/mimas_a7.py --uart-name usb_fifo``` and then loaded the bitstream with ```openocd -f numato_mimas_a7.cfg -c "transport select jtag; init; pld load 0 soc_basesoc_mimas_a7/gateware/top.bit; shutdown"```

closed 2020-01-14 by feliks-montez

enjoy-digital, 2020-01-15:
Great, thanks for the feedback.