enjoy-digital/litex #335: Use of signed CSRStatus

DurandA, 2020-01-06:
I wanted to store signed value in a CSRStatus from a comparator. However, it seems that using signed signals is unsupported since it triggers a warning (e.g. `Warning: Literal has a width of 32 bit, but value requires 33 bit.`).

Here is a simple module to demonstrate the issue:
```
class SignedCSRModule(Module, AutoCSR):
    def __init__(self, cell_sets, clock_domain="sys"):
        self.bit_value = comparator = Signal((32, True))
        self._bit_value = CSRStatus(32, reset=0)

        self.specials += [
            MultiReg(self._reset.storage, self.reset, clock_domain),
            MultiReg(comparator, self._bit_value.status, clock_domain),
        ]
```

Note that reducing the signal width of the comparator to 31 bit will not solve the issue (i.e. `Warning: Literal has a width of 31 bit, but value requires 32 bit.`).

Since I just use addition and subtraction, it doesn't matter much as operations are unified in two's complement and I can just use a signed value.

enjoy-digital, 2020-01-12:
Thanks for reporting. https://github.com/enjoy-digital/litex/commit/d92bd8ffaa9a9d1f3d51498c8e382f7041e69274 should fix it.

closed 2020-01-12 by enjoy-digital

DurandA, 2020-01-12:
Many thanks!