enjoy-digital/litex #341: Add support for memtest on non-DDR memories.

mithro, 2020-01-13:
It would be good to have support for memtest functionality on non-DDR memories including;
 - [ ] HyperRAM
 - [ ] SPI RAM (like used on the hackaday badge)
 - [ ] Static-RAM
 - [ ] Single Port RAM inside the Fomu (128 kbytes).

mithro, 2020-01-14:
The memory bandwidth testing would be great too!

enjoy-digital, 2020-12-22:
With the recent changes, generic mem_test and mem_speed commands were already available. With https://github.com/enjoy-digital/litex/commit/1a338b602a125ba07ed348b7d63db49cb73f0bd8, a mem_list command is added to list the available memory regions. This way all regions of the SoC can be easily tested (write/read accesses and speed):

List the memory regions:
```
    litex> mem_list
    Available memory regions:
    ROM       0x00000000 0x8000
    SRAM      0x01000000 0x2000
    MAIN_RAM  0x40000000 0x10000000
    CSR       0x82000000 0x10000
```

 Test 0x1000 bytes of MAIN_RAM:
   
```
    litex> mem_test 0x40000000 0x1000
    Memtest at 0x40000000 (4KiB)...
      Write: 0x40000000-0x40001000 4KiB
       Read: 0x40000000-0x40001000 4KiB
    Memtest OK
```
    
Test speed on 0x1000 bytes of MAIN_RAM:
    
```
    litex> mem_speed 0x40000000 0x1000
    Memspeed at 0x40000000 (4KiB)...
      Write speed: 352KiB/s
       Read speed: 288KiB/s
```

closed 2020-12-22 by enjoy-digital