enjoy-digital/litex #342: Import the RSD out of order RISC-V core

mithro, 2020-01-14:
> RSD is a 32-bit RISC-V out-of-order superscalar processor core. RSD is very fast due to
> aggressive OoO features, while it is very compact and can be synthesized for small FPGAs.
> 
[image]

It would be awesome to support an out of order RISC-V core for comparison to in-order cores like VexRISCV and Rocket.

mithro, 2020-01-14:
I logged the following upstream bug at https://github.com/rsd-devel/rsd/issues/6

mithro, 2020-01-14:
Related issues;
 - https://github.com/enjoy-digital/litex/issues/177 - Import the Taiga high performance RISC-V core
 - https://github.com/enjoy-digital/litex/issues/245 - Import the microwatt PowerPC core
 - https://github.com/enjoy-digital/litex/pull/278 - Pull request to import BlackParrot
 - https://github.com/enjoy-digital/litex/pull/186 - Experimental Support for 64-bit RocketChip

enjoy-digital, 2020-05-19:
There are no ongoing efforts to import/support this core and now have others RISC-V linux capable alternatives (Rocket, BlackParrot), so i'm closing but please re-open if someone is willing to put efforts into this.

closed 2020-05-19 by enjoy-digital