enjoy-digital/litex #344: Enable breakpoints to vexriscv-debug

xobs, 2020-01-15:
The current `+debug` variants all have the `DebugPlugin` added, however none of them have any breakpoints.  Having breakpoints can be very useful, and adding them is relatively easy.

We should add at least two breakpoints.

An example of how to do this is available at https://github.com/xobs/VexRiscv-verilog/blob/master/src/main/scala/vexriscv/GenFomu.scala#L285

gregdavill, 2020-03-11:
I've added the changes I've had to do to enabled hardware breakpoints with the LiteX vexriscv configuration. https://github.com/litex-hub/litex-data-cpu-vexriscv/pull/5

I see @mithro has recently refactored the prebuilt vexriscv layout. So I'm not sure if I altered the correct file. Basically you just need to add `hardwareBreakpointCount` to the `DebugPligin` in the scala configuration/generation file as @xobs has mentioned.

I've also added the parameter so this can be changed in the makefile, I've set all debug platforms with 2 breakpoints.

enjoy-digital, 2020-03-12:
Thanks @gregdavill.

enjoy-digital, 2020-03-23:
@gregdavill: i've merged your changes in https://github.com/litex-hub/litex-data-cpu-vexriscv and regenerated the verilog, but https://github.com/litex-hub/litex-data-cpu-vexriscv seems corrupted since the change from vexriscv_verilog to litex-data-cpu-vexriscv. We'll have to finish https://github.com/enjoy-digital/litex/issues/394 and/or investigate before being able to have breakpoints enabled by default.

enjoy-digital, 2020-04-28:
With #399 merged, we are now using the updated Vexriscv verilog, so with the hardware breakpoints.

closed 2020-04-28 by enjoy-digital

kaolpr, 2022-01-11:
I'm having a problem with those hardware breakpoints and https://github.com/litex-hub/pythondata-cpu-vexriscv/blob/2fe1d49bdd2fc02cf2ef279af70c6fd232d6f681/pythondata_cpu_vexriscv/verilog/VexRiscv_Debug.yaml#L1 suggests they are not really enabled, am I right?

tcal-x, 2022-01-11:
> I'm having a problem with those hardware breakpoints and https://github.com/litex-hub/pythondata-cpu-vexriscv/blob/2fe1d49bdd2fc02cf2ef279af70c6fd232d6f681/pythondata_cpu_vexriscv/verilog/VexRiscv_Debug.yaml#L1 suggests they are not really enabled, am I right?

How many hardware breakpoints do you specify?

There may still be a specific bug in the software (not VexRiscv or LiteX) that happens when hwbp count is 1.  The workaround is to use 2 or more hwbp's.

https://github.com/litex-hub/wishbone-utils/issues/30

kaolpr, 2022-01-11:
Well, it turns out that hardware breakpoints were not enabled at all. They were introduced once, but than, after some repository refactoring they were lost. 

After enabling them (PR https://github.com/litex-hub/pythondata-cpu-vexriscv/pull/16/files) it works like a charm.