enjoy-digital/litex #345: Support register_mem autoselecting appropriate memory location

mithro, 2020-01-15:
Currently you have to do a;
```python
    def add_memory_region(self, name, origin, length, type="cached", io_region=False):
        ...
```
Which means you need to manually choose an `origin` for the for the memory.

Compared to adding an interrupt;
```python
    def add_interrupt(self, interrupt_name, interrupt_id=None, allow_user_defined=False):
        ...
```
You can choose to supply an `interrupt_id` value or you can just let the SoC decide on the best option.

Also compared to adding a csr region;
```python
    def add_csr(self, csr_name, csr_id=None, allow_user_defined=False):
        ...
```
You can again choose to supply a `csr_id` value or you can just let the SoC decide on the best option.

Keeping the `add_memory_region` API compatible while allowing a user to *just* provide a length like `xxx.add_memory_region("testing", 12)` is somewhat hard. I think it can be done like the following;
```python
    def add_memory_region(self, name, *args, *, origin=None, length=None, type="cached", io_region=False):
        # Support the old (name, origin, length) form
        if len(arg) == 2:
            # Make sure that origin / length wasn't also provided as a keyword argument
            assert not origin, (args, origin)
            assert not length, (args, length)
            origin = arg[0]
            length = arg[1]
        # Support just specify length (name, length)
        # Support just specify length (name, length, origin=XXXX)
        elif len(arg) == 1:
            # Make sure that origin wasn't also provided as a keyword argument
            assert not length
            length = arg[0]
       elif len(args) != 0:
            raise ArgumentError("Wrong number of arguments. add_mem_region(name, {})".format(args))
      # Auto select a origin if not provided...
      if origin is None:
          ...        
```

Thoughts?

enjoy-digital, 2020-01-20:
@mithro: thanks for the feedback and implementation. I was also thinking the same and also wanted to change the name of the function to reflect with the internal names (`mem_regions`). As you say, keeping API compatibility is not easy for this, so i suggest just using another name: `add_mem_region` for the new API while keeping `add_memory_region` support for now with the old API (that we could deprecate later). I implemented  `add_mem_region` in https://github.com/enjoy-digital/litex/commit/53bc18cc3f739ed02b95a9e9e530915e22b3e9d2 and will do more tests on SoC with automatic memory mapping. If it's working fine, i'll adapt the LiteX/LiteX-boards targets to it (old targets will still be functional).

mithro, 2020-01-21:
Two functions with the similar names and behaviour is a fraught with danger -- Which one do I use? Which one is the right one?. It is better to take the hit to make the function work harder.

enjoy-digital, 2020-01-21:
Both will work for now, we will add a deprecated warning to add_memory_region and will remove it in the future.

enjoy-digital, 2020-02-18:
This is now supported by the new SoC class. For now we are only using it in retro-compatibility mode with SoCCore. We'll keep the SoCCore API for retro-compatibility for old designs (with only `add_memory_region` not 'add_mem_region'), new design will use the new SoC Class and existing examples/design adapted progressively.

closed 2020-02-18 by enjoy-digital