enjoy-digital/litex #346: Support for multiple DRAMs in SoCSDRAM

kgugala, 2020-01-15:
We've been working on adding Litex support for Pano Logic G2 board (https://github.com/tomverbeure/panologic-g2) we've added basic support in Litex-buildenv (https://github.com/timvideos/litex-buildenv/pull/223).

The board has 2 DDRAM chips connected to separate banks. To handle this we should instantiate two DDRAM controllers, but current SoCSDRAM implementation assumes there is single DDRAM PHY in the system.

CC @mithro

mithro, 2020-01-15:
@kgugala You should just be able to use the non-`SoCSDRAM` and just instantiate the block twice?

@enjoy-digital Might have a suggestion on how to make the 2 x DDR chips driven via a single DDR controller?

enjoy-digital, 2020-01-16:
@mithro: i've been thinking for some time about having a single SoC class that could cover SoCCore/SoCSDRAM needs and that would be more flexible and would remove this kind of limitations. I'll prototype something.

enjoy-digital, 2020-03-05:
When we have multiple DRAM chips connected to the board using separate command lines we have the choice to:
1. Have 2 PHYs and a single Controller, DFI commands will just be replicated on each PHY and the only change for the Controller and system is that the data-width/capacity/bandwidth is doubled.
2. Have a Controller per PHY that will be mapped at a different location and each Controller/PHY will be fully independent from the other.

Both solutions can have advantages/disadvantages depending the use case.

To support 1. we could create a wrapper that will encapsulate the 2 PHYs and create a single DFI interface from the 2 DFIs of each PHY.

To support 2. we could use the new SoC class that has a `add_sdram` method that we could use to add the 2 Controllers, but this would still need some changes to the software to do the initialization of the 2 Controllers. The new `add_sdram ` method is currently prototyped on the nexys4ddr target:
Nexys4DDR targets: https://github.com/enjoy-digital/litex/blob/master/litex/boards/targets/nexys4ddr.py#L74

enjoy-digital, 2020-03-05:
@mglb submitted https://github.com/enjoy-digital/litex/pull/411 and https://github.com/enjoy-digital/litedram/pull/157 just after i wrote my previous comment. I'm a bit uncomfortable with this, because i realize how much work it is, but i'm not sure we have a real use case for it. From my previous comment, on the Pano Logic G2 we clearly want to implement solution 1. while it is doing so implementing 2.

1. could be implemented with just some transformations on the pads, the PHY/Controller/SOC will adapts itself after that, here is for example a  proof of concept:

In LiteDRAM, add a function to gather the pads from the different chips:
````python3
class LiteDRAMPadsGatherer:
    def __init__(self, pads):
        if not isinstance(pads, list):
            self.groups = [pads]
        else:
            self.groups = pads
        self.sel = 0

    def sel_group(self, n):
        self.sel = n

    def __getattr__(self, name):
        if name in ["dm", "dq", "dqs_p", "dqs_n"]:
            return Cat(*[getattr(self.groups[i], name)
                for i in range(len(self.groups))])
        else:
            return getattr(self.groups[self.sel], name)
````

In the PHYs: transform the pads:
````python3
class S7DDRPHY(Module, AutoCSR):
    def __init__(self, pads, with_odelay,
        memtype          = "DDR3",
        nphases          = 4,
        sys_clk_freq     = 100e6,
        iodelay_clk_freq = 200e6,
        cmd_latency      = 0):
        pads = LiteDRAMPadsGatherer(pads)
       [...]
````

and just do a loop on the command signals:
````python3
        for g in range(len(pads.groups)):
            pads.sel_group(g)
           # Clock ------------------------------------------------------------------------------------
           [...]
           # Addresses and Commands -------------------------------------------------------
          [...]
````

and instiantiate the PHY like this:
````python3
self.submodules.ddrphy = s7ddrphy.A7DDRPHY(
      #pads = platform.request("ddram", 0), # 1 PHY
       pads = [platform.request("ddram", 0), platform.request("ddram", 1)], # 2 PHYs
       memtype      = "DDR3",
       nphases      = 4,
       sys_clk_freq = sys_clk_freq)
````
Also that with this solution, the the raw throughput seen from the CPU or from a DRAM port is 2x the one from the other solution.

So i'm not sure we should really support multiple SDRAM controller at the same time in LiteXSoC and if so, it will generally be on very large FPGAs where the cost of a CPU for the initialization is minimal and so where LiteDRAM will be instantiated with as a standalone core that will initialize itself , that will be in fact a LiteXSoC with its own CPU + ROM + RAM + LiteDRAM controller and be part of a main SoC as a Wishbone/AXI slave peripheral. So maybe we should restrict the BIOS to only support one ROM, one SDRAM, one Ethernet, one SDCard etc... and consider that for more complex systems, multiple LiteXSoC needs to be combined together or that user has to write software for it. Otherwise, i fear this is going to be un-maintanable or someone is going to become crazy (hoping it will not be me :))

@mglb  @kgugala @mithro: what do you think?

mithro, 2020-03-05:
@enjoy-digital -- I guess I don't understand why this isn't as simple as just being a `for X in Y` where in most cases Y is a list with a single element? (I haven't had a chance to review the stuff you linked.)

Of the peripherals you listed, only DDR really needs BIOS support (LiteEth's BIOS stuff is more about increased usefulness rather than *actually* required).

One of LiteX + LiteDRAM's power is the extreme resource efficiency, it seems silly to take a step backwards just because you have more than one of anything....

enjoy-digital, 2020-03-05:
@mithro: it's not about taking a step backwards it's about being pragmatic: i'm not sure we we need to complicate that much the code if the use case is only the Pano Logic G2 and would like that we test the solution i'm suggesting that can be done more easily and provide better performance for the actual use case. If you have others use cases in mind i'd like to be aware of it. If not, i'm just saying we should probably wait to have a real use case and then think if that's what we really want to implement.

From my experience, each time i had to have multiple LiteDRAM controllers on an FPGA:
1. It was on systems with 64-bit or 72-bit SO-DIMM where the bus is very large which then create a large controller and where the CPU + ROM + RAM resource usage is only a small fraction of the LiteDRAM controller resource usage.
2. Total resource usage with CPU + ROM + RAM was still a lot less than 50% of MIG
3. It was on very large FPGA XCU440 where your CPU almost comes for free. 
4. From a system point of view, it was a lot easier to have standalone cores that just behave themselves, so a distributed system, than centralized system with a common CPU.
5. On these systems where DRAM controllers are far from each others on the chip, sharing a common CPU+Bus to control 3 or 4 controllers could create more congestions/constraints than just using separate CPU and a bit more resources.

My idea is just that rather spending lots of energy maintaining a centralized system that can support all configurations, maybe we should keep the SoCs simple and focus on the interconnection between simple SoCs to create complex systems.

Also for the BIOS, i'm also wondering if we should keep it minimal and switch to a proper boot-loader (Barebox or U-Boot) to support more easily the various boot options (Ethernet, Flash, SD-Card, etc...).

mithro, 2020-03-05:
I see three use cases here;
 - (a) A single DDR controller that is of arbitrary width.
 - (b) Multiple independent (but small) DDR controllers mapped to different memory regions.
   * This example is were you have multiple data streams that need independent bandwidth. 
   * In the video world this would be things like using controller A for buffering for Input 1 / Output 1 and controller B for buffering Input 2 / Output 2.
   * In data processing applications this would be where you have a flow of data that goes from one controller to another. IE You stream the data in a A-->B, then B-->A, then A-->B -- meaning one controller is only doing reading, while the second controller is only doing writing.
   * Another video example would be having a DDR controller dedicated to frame buffering operations (probably connected to a quite small sized module) while the other DDR controller is used for general data processing operations. Think of it like having one DDR controller for your video card and one for your CPU.
 - (c) Large devices which have many DDR controllers.

For option (c) I think we agree that the model should be to use LiteDRAM in an independent core fashion. They are not highly sensitive to the cost overhead of multiple RISC-V cores and similar issues.

As option (b) is frequently cost sensitive there is frequently a lot of sensitivity to resource usage. I don't think the independent core option is a great fit here. I'm still missing something about why it isn't as simple as just mapping the to separate memory regions and then calling the DDR init code with two different base addresses?

For option (a) the software doesn't really even see any of the results? It just looks like one big DDR controller?

For *both* option (a) and (b) I think we want it to be an easily supported default config.

mithro, 2020-03-05:
FYI - I'm pretty supportive of replacing the LiteX BIOS with a more traditional bootloader like U-boot. The less that is unique and special about LiteX on the software side, the more likelihood that people already have experience with the tools and can get started quicker. The primary question is if U-boot can get small enough and be modified easily enough to fit our usage model.....

daveshah1, 2020-03-05:
Full U-boot would have to be on SPI flash, but iirc U-Boot has something called SPL (it's a while since I've last used it) that is closer to the LiteX BIOS basically doing DRAM and SPI init type stuff, and might fit in BRAM.

mithro, 2020-03-05:
Let's split the BIOS question into a separate issue - created https://github.com/enjoy-digital/litex/issues/413

mithro, 2020-03-05:
@enjoy-digital - FYI I took a look at https://github.com/enjoy-digital/litex/pull/411/files and I believe it can be simplified significantly with a very small amount of code refactoring. I don't believe multiple DDR support needs to have as dramatic effect as that pull request shows.

enjoy-digital, 2020-03-09:
With https://github.com/enjoy-digital/litedram/commit/5e068f412b8574696aae1e4a5785440ec727abbc and https://github.com/enjoy-digital/litedram/commit/26564ba93cf342a191dc68d31194caf81c7c0c6a it should now be possible to create a configuration for the Pano Logic G2 board using the 2 external DRAM chips  managed by a single controller. This could probably be tested/used for now until we decide if we have a need for multiple Controller now and how to do it without introducing too much changes.

mglb, 2020-03-24:
@enjoy-digital looks like the use of a single controller is not possible with pano logic g2. Memory chips are connected to different edges and IO banks of the FPGA, and serdes instances located on different edges can't be controlled by clock and strobe signal from single BUFPLL.

enjoy-digital, 2020-03-24:
@mglb: ok thanks for the feedback. I suggest we wait a little bit before deciding what is the best to do.

enjoy-digital, 2020-12-01:
The current PRs have just been closed since won't be merged (work has probably been done too early):
- https://github.com/enjoy-digital/litex/pull/411
- https://github.com/enjoy-digital/litedram/pull/157

This will however be useful in the future for reference.

enjoy-digital, 2021-01-08:
As explained in my previous comments, I don't think this is a good idea to provide build-in support for this in LiteX BIOS. This use case is very occasional and when required can be implemented differently (multiple standalone controllers using SERV CPU for example for the initialization). In the future, LiteX MPSoC support will simplify these use cases while still only supporting only a single DRAM controller per LiteX SoC in the BIOS.

closed 2021-01-08 by enjoy-digital