enjoy-digital/litex #350: Allow specifying CSR spacing

xobs, 2020-01-20:
Currently, CSRs appear to be hardcoded to a spacing of `0x800`, meaning each new CSR is spaced `0x800` above its previous value.  For example, if CSR[0] is placed at `0xe0000000`, then CSR[1] will be placed at `0xe0000800`.

If this were allowed to be configured, then it could be changed so that CSR[1] would instead be placed at 0xe0001000`.

The primary usecase here is to allow CSRs to be managed by the RISC-V MMU, such that each CSR block is managed by a different userspace process.  By placing CSRs on a page boundary (i.e. `0x1000` or `4096`), each CSR block gets its own access control entry.

enjoy-digital, 2020-01-20:
@xobs: indeed that would be useful. I had a very quick look at that some time ago to make it modular but haven't finished it. I'll have another look.

enjoy-digital, 2020-02-17:
This is now supported with:  https://github.com/enjoy-digital/litex/commit/67e8a042f83dd64daa338de6d55d442223e0b684, https://github.com/enjoy-digital/litex/commit/0497f3ca714a5ffa33d667d4eb687379dd7c4b8f,  https://github.com/enjoy-digital/litex/commit/5b3808cb810f8596c73fdac81f0c37781865e116 and https://github.com/enjoy-digital/litex/commit/6ed0f445b609945742abfc41b81c9f0ab8b83c6f.

I tested it with:
lxsim --csr-paging=2048 --csr-address-width=14
lxsim --csr-paging=4096 --csr-address-width=14
lxsim --csr-paging=8192 --csr-address-width=14
lxsim --csr-paging=16384 --csr-address-width=15

@xobs: can you confirm it works for you?

bunnie, 2020-02-17:
Regression tested successfully to hardware with 4096-byte pages: https://github.com/betrusted-io/betrusted-soc/commit/aa9109b2ba4a6a0a0424f9ffb876a7eed568b943

enjoy-digital, 2020-02-17:
Great, thanks for the feedback.

closed 2020-02-17 by enjoy-digital