enjoy-digital/litex #355: litex_sim: add LPDDR/DDR/DDR2/DDR3 support

enjoy-digital, 2020-01-23:
Requires https://github.com/enjoy-digital/litex/issues/352 and https://github.com/enjoy-digital/litex/issues/353.

Each SDRAM module has a `memtype` attribute that can be used to identify the type of SDRAM module.

For each type of memory, `phy_settings` will be created based on settings already used on hardware:
- DDR/LPDDR: Pipistrello settings.
- DDR2: Nexys4DDR settings.
- DDR3: Arty settings.

To get the inital `phy_settings` values, some traces can be added to the LiteDRAM PHY:
- https://github.com/enjoy-digital/litedram/blob/master/litedram/phy/s6ddrphy.py#L63
- https://github.com/enjoy-digital/litedram/blob/master/litedram/phy/s7ddrphy.py#L71

Once simulation is working with fixed values for each configuration, computation done inside the PHY to generate the `phy_settings` should be integrated in the simulation.

While doing this work, doing first point of https://github.com/enjoy-digital/litedram/issues/60 would also be useful to avoid duplicating code for the  `phy_settings` computation.

This change will allow simulating the various SDRAM configurations we are already using on hardware and will ease benchmarking memory accesses with LiteX/LiteDRAM.

piotr-binkowski, 2020-01-24:
From what I see the first point of enjoy-digital/litedram#60 might be already done because latency/phases computation code used by `s7ddrphy` is already in `litedram/common.py`:

- https://github.com/enjoy-digital/litedram/blob/master/litedram/common.py#L26
- https://github.com/enjoy-digital/litedram/blob/master/litedram/common.py#L48
- https://github.com/enjoy-digital/litedram/blob/master/litedram/common.py#L51

enjoy-digital, 2020-01-24:
@piotr-binkowski: indeed, sorry i forgot to update https://github.com/enjoy-digital/litedram/issues/60 when i did it.

enjoy-digital, 2020-01-24:
With https://github.com/enjoy-digital/litex/pull/358, we can now simulate LPDDR/DDR/DDR2/DDR3, but memtest reports errors for DDR2/DDR3 which needs to be investigated. (could be a configuration issue or DFI model limitation).

piotr-binkowski, 2020-01-28:
I did some more testing and it looks like configuring `write_latency` as 0 in PhyConfig fixes the issues with DDR2/DDR3

enjoy-digital, 2020-01-28:
OK great thanks, it could be that `write_latency` is not modelized correctly in the DFI model. I will also look a bit later, if you find things please continue posting here.

enjoy-digital, 2020-01-28:
WIth https://github.com/enjoy-digital/litedram/pull/111, memtest is now passing for SDR/DDR/LPDDR/DDR2 and DDR3. Let's wait https://github.com/enjoy-digital/litedram/issues/102 and https://github.com/litex-hub/linux-on-litex-vexriscv/issues/84 to be done before closing.

PR #358 (2020-01-24): tools/litex_sim: add support for other sdram types
This PR will allow litex_sim to run simulations using other types of sdram memory (DDR, LPDDR, DDR2, DDR3) instead of being limited to SDR only

Closes #355
closed 2020-02-03 by enjoy-digital