enjoy-digital/litex #367: Make lxsocdoc part of LiteX directly?

mithro, 2020-01-29:
[lxsocdoc](https://github.com/litex-hub/lxsocdoc) from @xobs is really cool. 

[It generates really nice documentation for CSR registers](https://rm.fomu.im/). It also generates nice SVD files for usage by things like rust bindings. 

As it seems **generally** useful to anyone using the litex CSR registers, I think it should be included as part of LiteX. It will also help promote the idea of better documenting the CSR registers too.

@xobs / @enjoy-digital - What do you think?

enjoy-digital, 2020-01-29:
@mithro: I also find it really cool and already started using it on different projects, so would be happy to have it directly integrated if that's the intention of @xobs. (I also like a lot lxbuildenv :))

xobs, 2020-01-30:
Sure thing.  Once you get used to documenting registers, life becomes so much simpler!  Future-you will thank today-you.

The current codebase is a bit of a mess, but that can be fixed over time.  I've just fixed basic support for non-8-bit CSRs (it was mostly compatible, just had `8` hardcoded in a few places), and need to add support for register lanes.

What would be the process to bring in lxsocdoc?

enjoy-digital, 2020-01-30:
@xobs we could integrate the current version with the others tools in: https://github.com/enjoy-digital/litex/tree/master/litex/tools and install it with LiteX as we are doing with the others tools: https://github.com/enjoy-digital/litex/blob/master/setup.py#L45-L48. I'm happy to work with you on improving it over time. Can you create a PR for that if that's fine for you?

xobs, 2020-02-04:
All of the tools under `litex/tools` are executables, whereas `lxsocdoc` is more of a pipeline flow.  That is, it's meant to be added just following execution of `Builder`.

Might it make more sense to put it under a new directory `litex/gen/doc/`?

enjoy-digital, 2020-02-04:
In fact `litex/gen` is more related to Migen level specific features. What about `litex/soc/doc`? We could also move the current `litex/soc/integration/doc.py` to it (eventually renamed to` autodoc.py`?). This also makes things consistent: `lxsocdoc` would be the short name for `litex/soc/doc` and the import could be done with: `from litex.soc.doc import *` or just as you were already doing `from lxsocdoc import *`

enjoy-digital, 2020-02-07:
Integrated with https://github.com/enjoy-digital/litex/pull/375.

closed 2020-02-07 by enjoy-digital