enjoy-digital/litex #368: minerva currently broken

mithro, 2020-01-29:
See pull request https://github.com/timvideos/litex-buildenv/pull/301
```
./build/arty_base_minerva/output.20200129-184738.log
---------------------------------------------
Traceback (most recent call last):
  File "/home/travis/build/timvideos/litex-buildenv/third_party/litex/litex/soc/cores/cpu/minerva/verilog/cli.py", line 114, in <module>
    main()
  File "/home/travis/build/timvideos/litex-buildenv/third_party/litex/litex/soc/cores/cpu/minerva/verilog/cli.py", line 110, in main
    cli.main_runner(parser, args, cpu, name="minerva_cpu", ports=ports)
  File "/home/travis/build/timvideos/litex-buildenv/third_party/nmigen/nmigen/cli.py", line 46, in main_runner
    fragment = Fragment.get(design, platform)
  File "/home/travis/build/timvideos/litex-buildenv/third_party/nmigen/nmigen/hdl/ir.py", line 67, in get
    obj = obj.elaborate(platform)
  File "/home/travis/build/timvideos/litex-buildenv/third_party/nmigen/nmigen/hdl/dsl.py", line 484, in elaborate
    fragment.add_subfragment(Fragment.get(self._named_submodules[name], platform), name)
  File "/home/travis/build/timvideos/litex-buildenv/third_party/nmigen/nmigen/hdl/ir.py", line 67, in get
    obj = obj.elaborate(platform)
  File "/home/travis/build/timvideos/litex-buildenv/third_party/litex/litex/soc/cores/cpu/minerva/verilog/minerva/units/decoder.py", line 77, in elaborate
    fmt = Signal(max=Type.J)
TypeError: __init__() got an unexpected keyword argument 'max'
Traceback (most recent call last):
  File "./make.py", line 169, in <module>
    main()
  File "./make.py", line 151, in main
    vns = builder.build(**dict(args.build_option))
  File "/home/travis/build/timvideos/litex-buildenv/third_party/litex/litex/soc/integration/builder.py", line 167, in build
    self.soc.finalize()
  File "/home/travis/build/timvideos/litex-buildenv/third_party/migen/migen/fhdl/module.py", line 156, in finalize
    subfragments = self._collect_submodules()
  File "/home/travis/build/timvideos/litex-buildenv/third_party/migen/migen/fhdl/module.py", line 149, in _collect_submodules
    r.append((name, submodule.get_fragment()))
  File "/home/travis/build/timvideos/litex-buildenv/third_party/migen/migen/fhdl/module.py", line 102, in get_fragment
    self.finalize()
  File "/home/travis/build/timvideos/litex-buildenv/third_party/migen/migen/fhdl/module.py", line 157, in finalize
    self.do_finalize(*args, **kwargs)
  File "/home/travis/build/timvideos/litex-buildenv/third_party/litex/litex/soc/cores/cpu/minerva/core.py", line 112, in do_finalize
    verilog_filename = verilog_filename)
  File "/home/travis/build/timvideos/litex-buildenv/third_party/litex/litex/soc/cores/cpu/minerva/core.py", line 102, in elaborate
    raise OSError("Unable to elaborate Minerva CPU, please check your nMigen/Yosys install")
OSError: Unable to elaborate Minerva CPU, please check your nMigen/Yosys install
```

Seems to be caused by newer nMigen and the current litex?

enjoy-digital, 2020-01-30:
Minerva is not building correctly with nMigen master, i just created an issue for that: https://github.com/lambdaconcept/minerva/issues/7

enjoy-digital, 2020-01-30:
Fixed with https://github.com/lambdaconcept/minerva/commit/fb296e4e48e5ced8dd05f2228d84b4bc18f54f75 and https://github.com/enjoy-digital/litex/commit/c6b9676db8d9846f55e2c85261b666bb5149e283.

closed 2020-01-30 by enjoy-digital