enjoy-digital/litex #371: Increase in BIOS+firmware size increase with latest LiteX revision

mithro, 2020-02-01:
As of 3350d33 (see https://github.com/timvideos/litex-buildenv/pull/314) - the LiteX-BuildEnv is showing a lot of failures see https://travis-ci.com/timvideos/litex-buildenv/builds/146976853

This seems to be a result of BIOS / firmware seeming to use a lot more SRAM. This seems to affect all of `lm32`, `mor1kx` and `vexriscv` configurations;

## lm32 example
```
cp /home/travis/build/timvideos/litex-buildenv/third_party/litex/litex/soc/software/bios/boot-helper-lm32.S boot-helper-lm32.S
 CC       boot-helper-lm32.o
 LD       firmware.elf
lm32-elf-ld: firmware.elf section `.bss' will not fit in region `sram'
lm32-elf-ld: region `sram' overflowed by 10136 bytes
```

## vexriscv example

https://travis-ci.com/timvideos/litex-buildenv/jobs/282427202
```
 CC       /home/travis/build/timvideos/litex-buildenv/build/arty_base_vexriscv/software/include/../..//software/firmware/version_data.o
 CC       /home/travis/build/timvideos/litex-buildenv/build/arty_base_vexriscv/software/include/../..//software/firmware/hdmi_in1.o
cp /home/travis/build/timvideos/litex-buildenv/third_party/litex/litex/soc/software/bios/boot-helper-vexriscv.S boot-helper-vexriscv.S
 CC       boot-helper-vexriscv.o
 LD       firmware.elf
riscv32-unknown-elf-ld: firmware.elf section `.bss' will not fit in region `sram'
riscv32-unknown-elf-ld: region `sram' overflowed by 10136 bytes
/home/travis/build/timvideos/litex-buildenv/firmware/Makefile:93: recipe for target 'firmware.elf' failed
make[1]: *** [firmware.elf] Error 1
```

## mor1kx example

https://travis-ci.com/timvideos/litex-buildenv/jobs/282427196
```
 CC       boot.o
 LD       bios.elf
or1k-elf-ld: bios.elf section `.text' will not fit in region `rom'
or1k-elf-ld: region `rom' overflowed by 6424 bytes
/home/travis/build/timvideos/litex-buildenv/third_party/litex/litex/soc/software/bios/Makefile:27: recipe for target 'bios.elf' failed
make[1]: *** [bios.elf] Error 1
```

mithro, 2020-02-03:
The issue seems to *actually* be https://github.com/enjoy-digital/litex/commit/83a7225ccceb7f6f930441dc6b01aeaf0c8a2b06 when combined with https://github.com/enjoy-digital/litex/commit/7a05353aa7949707f23453b6db013e2a62ecc8d9 that causes this problem. The code in LiteX-BuildEnv does the following;
```python
        if 'integrated_rom_size' not in kwargs:
            kwargs['integrated_rom_size']=0x8000
        if 'integrated_sram_size' not in kwargs:
            kwargs['integrated_sram_size']=0x8000
```

enjoy-digital, 2020-03-04:
@mithro, @mateusz-holenko: is it now resolved on litex-buildenv?

mateusz-holenko, 2020-03-04:
I prepared a PR that fixes buildenv's platforms after bumping to the current LiteX: https://github.com/timvideos/litex-buildenv/pull/348.

It reworks the way LiteX-BuildEnv configures integrated rom/sram size and it should solve the issue.

enjoy-digital, 2020-03-16:
From the recent discussions, it seems to be related to the way the arguments are handled in litex-buildenv, so we can probably close this.

closed 2020-03-16 by enjoy-digital