enjoy-digital/litex #391: Black Parrot cleanups

enjoy-digital, 2020-02-20:
- [ ] Fix submodule commit issue (we can't do a checkout of https://github.com/black-parrot/pre-alpha-release/tree/8aa6b6259308105872e19675c1cd5aee22283913) 
- [ ] Cleanup core.py.
- [ ] Cleanup integration.
- [ ] Avoid cloning too much submodules (only https://github.com/black-parrot/pre-alpha-release and https://github.com/black-parrot/basejump_stl should be needed).

@scanakci: Could you help me with the first point? I'm not able to checkout the code you used for validation and would like to start from that. Can you update to a new version of the CPU or provide a commit that we can use for the checkout?

scanakci, 2020-02-20:
@enjoy-digital thanks for letting me know. I am currently using another commit since that version is not supporting some features required for Linux boot. Once I make it work, I will let you know.

enjoy-digital, 2020-02-20:
OK thanks, i'll wait for that then. I just did a first cleanup pass of `core.py`: https://github.com/enjoy-digital/litex/commit/53ee9a5e059b60d59c17d9d2d8a1249796695e73.

scanakci, 2020-05-02:
I made a PR which includes a version of BP that supports Linux and simulates BIOS .
https://github.com/litex-hub/pythondata-cpu-blackparrot/pull/1

Since I am using a commit from a developer branch of BP, I did not put as a submodule for now. I will add those submodules later when I make my BP version upstream.

I will make another PR for LiteX repo changes sometimes this weekend.

scanakci, 2020-05-02:
I will update the PR based on @mithro's suggestion.

scanakci, 2020-05-12:
I updated the BP version that boots up Linux with this [PR](https://github.com/enjoy-digital/black-parrot/pull/1). The recent PR](https://github.com/enjoy-digital/litex/pull/506) in LiteX repo updates the README and core.py in addition to a minor change in vivado command for properly generating bitstream for system verilog.

scanakci, 2020-05-12:
Can you please merge [PR](https://github.com/enjoy-digital/black-parrot/pull/1) for BlackParrot repo as well unless there is an issue?

enjoy-digital, 2020-05-13:
Sorry, that's now merged. Now that things are working, we should also probably see if we could generate https://github.com/litex-hub/pythondata-cpu-blackparrot directly from the official BlackParrot repository. I mostly created https://github.com/enjoy-digital/black-parrot to avoid the recursive clone of the black-parrot repository (that was done automatically since LiteX was clone with --recursive), but this should no longer be an issue with the pythondata packages.

scanakci, 2020-05-13:
The latest commit in official blackparrot repo will most probably not gonna work since there are some architectural changes that I should adapt to LiteX. My next step is to integrate most recent version of BlackParrot and generate pythondata-cpu-blackparrot using the recent version. I will make another PR once it is done.

enjoy-digital, 2020-05-13:
Ok thanks i understand. Let's keep things like this then for now.

enjoy-digital, 2021-06-17:
Closing, this issue is obsolete.

closed 2021-06-17 by enjoy-digital