enjoy-digital/litex #392: LiteX and LiteSDCard support

gsomlo, 2020-02-20:
With PR #390 I am able to add SDCard support to a generic LiteX SoC (for now only on the nexys4ddr). This was made possible by a recent LiteSDCard commit (https://github.com/enjoy-digital/litesdcard/commit/0beb92fd93c7cf98315668d0a1e173f5b9846035) allowing `SDClockerS7` to use a dedicated 100MHz clock signal that's not necessarily `sys_clk` (which, depending on the cpu being used, might be significantly lower).

@enjoy-digital pointed out a recent commit by @mateusz-holenko and @kamilrakoczy in the antmicro tree (https://github.com/antmicro/litesdcard/tree/sdcard-fixes) which seems to suggest further improvements to LiteSDCard (removing some CSRs configuring sdcard clock speed via software, adding some CSRs to send/receive data, etc.), and suggested we start this github issue to capture thoughts and ideas for improving litesdcard support in LiteX.

Personally, I'm interested in booting 64bit linux from a preinstalled partition laid out on a microSD card, specifically using the ECP5-based trellisboard. For now, though, nexys4ddr seems a more commonly available target on which to smooth out the overall rough edges.

Any ideas/comments/braninstorming much appreciated.

mateusz-holenko, 2020-02-20:
We have a working Linux driver for LiteSDCard on our repository: https://github.com/antmicro/litex-linux-riscv/tree/sd_card_driver that you might try. It's developed for 32-bits, so some adaptation will probably be needed - e.g., CSRs offsets - but hopefully not much.

After pushing fixes to LiteX and clearing the code a bit we will PR it to linux-on-litex-vexriscv.

rob-ng15, 2020-03-17:
A similar goal, but using SPI bitbanging for the SD Card. Maybe the FAT16 code could be reused.

https://github.com/enjoy-digital/litex/pull/429

enjoy-digital, 2020-11-09:
LiteSDCard is now fully supported in LiteX designs and is demonstrated on several LiteX-Boards targets.

closed 2020-11-09 by enjoy-digital

gsomlo, 2020-11-09:
Any word on what causes it to get stuck after a couple of insert-init-eject cycles?

@enjoy-digital: Do you think that's likely to be a simple fix, or should we have a (separate) issue as a reminder and/or place to track that behavior?

enjoy-digital, 2020-11-09:
@gsomlo: sorry i was closing this one since a bit outdated and generic, can you create a specific issue for the insert-init-eject issue?