enjoy-digital/litex #398: Litex BIOS SD Card Booting via SPI (bitbanging)

rob-ng15, 2020-02-22:
I've generated code, shown to work on a de10nano with a MiSTer SDRAM board that will load  EMULATOR.BIN, IMAGE, ROOTFS~1.CPI and RV32.DTB from a SD Card connected to GPIO pins in SPI mode.

Demonstration below:

`--============= Console ================--
litex> gpiospisdboot
SD Card via SPI Initialising
Reading MBR
Partition Information: Active=0x00, Type=0x06, LBAStart=0x00000800

Read FAT16 Boot Sector
  Jump Code:              0xeb 0x3c 0x90
  OEM Code:               [mkfs.fat]
  Sector Size:            512
  Sectors Per Cluster:    128
  Reserved Sectors:       128
  Number of Fats:         2
  Root Dir Entries:       2048
  Total Sectors Short:    0
  Media Descriptor:       0xf8
  Fat Size Sectors:       128
  Sectors Per Track:      32
  Number of Heads:        64
  Hidden Sectors:         2048
  Total Sectors Long:     4194304
  Drive Number:           0x80
  Current Head:           0x01
  Boot Signature:         0x29
  Volume ID:              0xebdc1fad
  Volume Label:           [RISCV      ]
  Volume Label:           [FAT16   ]
  Boot Sector Signature:  0xaa55

sdCardFatTable = 0xc7ff0000  Reading Fat16 Table (128 Sectors Long)

sdCardFat16RootDir = 0xc7fe0000  Reading Root Directory (128 Sectors Long)

Root Directory
  File 0 [RISCV   .   ] @ Cluster 0 for 0 bytes
  File 2 [EMULATOR.BIN] @ Cluster 3 for 9600 bytes
  File 4 [IMAGE   .   ] @ Cluster 4 for 4545524 bytes
  File 6 [ROOTFS~1.CPI] @ Cluster 74 for 8029184 bytes
  File 8 [RV32    .DTB] @ Cluster 197 for 1837 bytes

Reading File [IMAGE.] into 0xc0000000 : File starts at Cluster 4 length 4545524
Clusters: 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 32 33 34 35 36 37 38 39 40 41 42 43 44 45 46 47 48 49 50 51 52 53 54 55 56 57 58 59 60 61 62 63 64 65 66 67 68 69 70 71 72 73 

Reading File [ROOTFS~1.CPI] into 0xc0800000 : File starts at Cluster 74 length 8029184
Clusters: 74 75 76 77 78 79 80 81 82 83 84 85 86 87 88 89 90 91 92 93 94 95 96 97 98 99 100 101 102 103 104 105 106 107 108 109 110 111 112 113 114 115 116 117 118 119 120 121 122 123 124 125 126 127 128 129 130 131 132 133 134 135 136 137 138 139 140 141 142 143 144 145 146 147 148 149 150 151 152 153 154 155 156 157 158 159 160 161 162 163 164 165 166 167 168 169 170 171 172 173 174 175 176 177 178 179 180 181 182 183 184 185 186 187 188 189 190 191 192 193 194 195 196 

Reading File [RV32.DTB] into 0xc1000000 : File starts at Cluster 197 length 1837
Clusters: 197 

Reading File [EMULATOR.BIN] into 0x20000000 : File starts at Cluster 3 length 9600
Clusters: 3 

Executing booted program at 0x20000000

--============= Liftoff! ===============--
VexRiscv Machine Mode software built Jan 10 2020 11:39:55
--========== Booting Linux =============--`

Code is available at [https://github.com/rob-ng15/litex-mister](url)

enjoy-digital, 2020-02-22:
@rob-ng15: Great! I'll have a closer look and will try to reproduce your results next week. How long does it take to copy the files? That would be an awesome PR :)

rob-ng15, 2020-02-22:
I guess it is dependent on the RAM speed. In the code the SDcard is run at
400HHz, but the processor and RAM run at 50MHz, so the RAM writing speed is
slow.

It takes about 10 minutes to transfer the Linux files on the DE10nano due
to the above. It may well be possible to run the SDcard at a higher speed,
but I need to be able to drive the RAM at a higher speed.

For reference, I could not get the same code working with the spi
bitbanging module, it would never give a MISO return from the card. The way
that the code is modularised the actual interaction with the hardware is
contained in a few modules, near the top of the code, so it should be
adaptable to other hardware.

As a proof of concept / first bash, it works, but can no doubt be improved.

mithro, 2020-02-23:
@rob-ng15 - This is supercool (even if slow)!

mithro, 2020-02-23:
@rob-ng15 There are a bunch of people on the #tomu IRC channel who have been looking at the soft cpu performance and might also be able to offer tips about where performance might be going. The Fomu is running the CPU at 16MHz, so things are a little bit more important there :-) (FYI Would you like a Fomu board?)

rob-ng15, 2020-02-23:
Many thanks Tim.

Although I got this working, sadly I didn't reach one of my design goals,
which was to have a quicker booting system than lxterm.

I'll keep on tidying up the code, adding comments (I found one optimisation
yesterday), as it has been a long time since I've done any coding it is a
bit rough. Plus, as it wasn't exactly straight forward talking to an SDcard
via SPI there has been a lot of testing to see how to get it going.

The Fomu board looks kind of cool! I'm old enough to remember when IRC was
brand new. One of the little beasts would be cool.

Rob.

rob-ng15, 2020-02-23:
Code tidied and comments added

rob-ng15, 2020-02-24:
Have tried running the card at 25MHz, i.e. half the speed of the processor, as fast as possible, and the speed is still slow.

Does anyone know of a way to set the SDRAM speed faster than 50MHz, i.e. to 100MHz which is still in spec?

enjoy-digital, 2020-02-24:
@rob-ng15: its not yet possible with SDRAM, we were mostly focusing work on DDRX but now that there is more interest in FPGA boards with SDRAM, that's something we should indeed support. That should not be too complicated and i'll try to add support for that soon.

rob-ng15, 2020-02-24:
Many thanks. I feel that whilst, obviously, FPGA implementations of a RiscV
core will be slow, we should try and get the best performance that we can.

Your efforts are much appreciated. I'll carry on playing!

enjoy-digital, 2020-02-24:
In fact, the actual bottleneck is more probably the bitbanging on the SDCard and not the RAM speed since from what i understand, you are executing the code from ROM and on SDRAM system with similar SDRAM/CPU frequency, it's possible to boot linux from SPI Flash in less than a minute:
https://twitter.com/GregDavill/status/1231082623633543168

You could maybe use the hardware SPI Master core: https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/spi.py. It's very easy to use, in your design create a SPI Master, similar to: https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_gen.py#L59-L74 Set data_width to the maximum size you'll need to transfer.

The core will provide you the following registers:
https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/spi.py#L127-L135

To do a SPI transfer:
- write data to the `mosi` register.
- write `start` and `length` to the control register.
- wait `done` register to be set.
- read the `miso` register if it was a read.

But if you are only doing byte transfers, not sure this will improve speed a lot due to the overhead of controlling the gateware SPI Master.

rob-ng15, 2020-02-25:
Many thanks for that information. I was looking at SPIMaster from spi.py and SPIMaster from bitbang.by and failed to get either working. I'll certainly give them a go, as all that will need to be modified are the functions towards the top that directly interact with the SDCARD.

I could do with a way of creating a delay loop, the one I use is probably very inefficient in that it sets the timer0 and checks that it hasn't depleted. As the BIOS is single threaded (I believe) I didn't see this as an issue.

Further tests do support bother of our theories:

1 RAM speed is a bottleneck
2 bitbanging is a bottleneck, especially at 400KHz, 1MHz seems to give a slight improvement. 25MHz would be ideal, but the processor isn't fast enough to keep up!

I'll let you know how I progress.

rob-ng15, 2020-02-25:
BTW can you help with this conundrum?

To intialise the SDCARD you need to hit the card with a series of pulses at 100-400MHz, but the card can operate at 25MHz. Any idea on how this can be achieved using SPIMaster, as that requires a specified clock speed for the SDCARD?

enjoy-digital, 2020-02-25:
@rob-ng15: It could be a modified SPIMaster that has CSR registers to this using bitbanging?

rob-ng15, 2020-02-25:
Sorry, I don't understand?

I tried SPIMASTER from bitbang.py, but failed to get it working, the code
being identical to this, but with the bits shifted into the registers
appropriately. I tested the protocol via gpio, resulting in this code.

I'll try again using the bitbang.py and see if I can get that working.

I'll then try SPIMASTER from spi.py and see if I can get that working.

enjoy-digital, 2020-02-25:
@rob-ng15: sorry, i was indeed saying to have an hybrid solution that mixes bitbang and hardware.
I just found a typo in the bitbanging SPI code: https://github.com/enjoy-digital/litex/commit/ee27a9e5340113de9fcbb91c160803c3677b6930 can you do another test with the updated code? This was probably your issue.

rob-ng15, 2020-02-25:
> I just found a typo in the bitbanging SPI code: ee27a9e

Thanks, I'll try and get to test that tomorrow. It would explain why I
couldn't get a MISO response.

I'd prefer to use the official Litex cores than bodge my own using GPIO.

At least I now no I wasn't mad! Thankfully the way I structured the code
the changes will be minimal.

rob-ng15, 2020-02-27:
Code has been rebased on the new SPIMaster from bitbang.py and is show to be working. I'll update my github repository ASAP. A few modifications have increased the speed slightly.

Glad the bug in bitbang.py has been found, but the experiment via GPIO pins was worthwhile to prove that the code actually worked!

enjoy-digital, 2020-02-27:
Great :)

rob-ng15, 2020-02-27:
My github has been updated. Hopefully someone can make use of, and improve this code.

rob-ng15, 2020-03-02:
OK, I am ready to try implementing SPIMaster from spi.py but I'll still need to be able to set the SD CARD to SPI by hitting it with the MOSI CS and CLK pulses at 100KHz to initialise SPI mode. Once that is done, the card can be driven at 25MHz using SPIMaster.

enjoy-digital, 2020-03-02:
With the patch attached you should be able to configure the clock divider from software with the added `clk_divide` register.

[0001-cores-spi-SPIMaster-make-clk_divider-configurable.patch.txt](https://github.com/enjoy-digital/litex/files/4275805/0001-cores-spi-SPIMaster-make-clk_divider-configurable.patch.txt)

rob-ng15, 2020-03-05:
Many thanks.

Finally got it working with the unpatched SPIMaster at 400KHz. It does give a significant speed boost, nowhere near the 1 minute quoted for booting from flash, but certainly less than 5 minutes.

I'll try the patch above, and see if I can get it working at 25MHz once the card has been initialised at 100 or 400KHz. I'll update my github.com accordingly, then the code is ready and waiting for anyone who wishes to use it.

PS: Some documentation to state that the length signal is the number of bits to transfer would be good!

enjoy-digital, 2020-03-05:
Great! Thanks for the feedback on the length, i'll update the description.

rob-ng15, 2020-03-05:
Thank you.

I've given up with the clock divider version. So I'll post code running the card at 400KHz, or whatever is set in the board file that initialises the SPIMaster.

I'll stick with the names spisdbb for software bitbanging and spisd for hardware bitbanging, and use these to conditionally pull in the code into the BIOS main.c and boot.c files, unless you have any suggestions? As SPI can be used for other purposes, I thought that might be for the best.

I'll update github.com shortly and then the code is ready for general testing and can be incorporated into Litex main repository if desired.

Goal of faster boot than via UART has finally been achieved using the hardware driver.

Thanks for all of your help.

rob-ng15, 2020-03-05:
github updated with spisdbb (software bitbanging) and spisd (hardware bitbanging).

Changed files are in the ALTERED folder.

When linux boots, does this matter [    0.000000] No DTB passed to the kernel?

enjoy-digital, 2020-03-05:
Thanks. For the "No DTB passed to the kernel" that something we also have: https://asciinema.org/a/tvvAQPzH29IsEEdmTOUTLEKeF

rob-ng15, 2020-03-05:
Now the code is working with the current SPIMaster from spi.py and I'm happy to try again with the clock dividable SPIMaster.

If I set the SPIMaster with a 25e6 = 25MHz, what value do I need to write to the clock divider to provide a 100khz signal? What value do I need to write to the clock divider to provide a 25MHz signal? 

I was probably writing the wrong values?

rob-ng15, 2020-03-13:
Probably, final change, unless we can work out the reclocking of the SPIMaster... New version, using SPI csr so as to be visible to the linux kernel...

https://github.com/rob-ng15/litex-mister/tree/master/SPI-Master

rob-ng15, 2020-03-17:
Pull request generated.

rob-ng15, 2020-03-19:
Excellent.

Glad it works for another as well. First bit of coding I have done for years. Enjoyed the challenge.

enjoy-digital, 2020-03-19:
Great you enjoyed doing this, that's really a nice contribution! The integration changes are:
- https://github.com/enjoy-digital/litex/commit/37f25ed37a246fc5397f4f3732322084b33477c1
- https://github.com/litex-hub/litex-boards/commit/6ab13a0661d8a629c8c4bc3486d54e707d1ce0d7
- https://github.com/litex-hub/linux-on-litex-vexriscv/commit/2d232e9cb1932d0add2926fe8afb1b15f1993fb1

Could you do a test with upstream LiteX/LiteX-boards and Linux-on-LiteX-Vexriscv to verify it's working on De10Nano + MiSTer?

Also by curiosity i did a quick test and increased the `spisdcard_clk_freq` from 400KHz to 16MHz here: https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/soc_linux.py#L119 and got things working with a very nice speed boost! you should probably do the test :) This means that supporting dynamic clock dividers will be very useful, i'll work on this.

rob-ng15, 2020-03-19:
>    - 37f25ed
>    - litex-hub/litex-boards@6ab13a0
>    - litex-hub/linux-on-litex-vexriscv@2d232e9

OK.

I'll test that shortly. I'll rebase to the master git, I need to download
it all again after my messing around!

I agree, on the Mister system I was able to boot in about 90 seconds at
400khz. Changing the speed may inhibit some cards initialising, as they
need a <400khz pulse to switch to spi mode. If reclocking can work, I'll
happily add in the 2 lines of code needed.

Definitely enjoyed. Very different working with others code.

Where do you recommend getting a FOMU from, and which one?

Rob.

enjoy-digital, 2020-03-19:
For the 400KHz, that was indeed just a test, the default value is still 400KHz. I just wanted to see if it was working to confirm if `spisdcard_clk_freq` was the main bottleneck and it seems to be the case.

For the Fomu @mithro has boards to give away for contributions and was willing to send you one if you were interested.

rob-ng15, 2020-03-19:
Ok, many thanks. Keep me posted when the the glitches with reclocking are dealt with.and I will be happy to test.

A set slow clock/set high clock can be put at the start/end of the spi_sdcard_goidle() function.

rob-ng15, 2020-03-19:
Just tried with a 16MHz spi sd card frequewncy, and yep it works on my sd card. Less than 30 seconds to boot!

Definitely looking forward to a more robust/portable implementation with dynamic reclocking. In the protocol there is a mechanism for reading the maximum speed of the sd card (apparently anyway), could be worth investigating when SPIMaster is up and running. Will happily add the support into the code.

enjoy-digital, 2020-03-20:
@rob-ng15: https://github.com/enjoy-digital/litex/pull/435 adds a configurable clock divider. The default value will be the one configured when building the gateware but it can be reconfigured by calling: `spisdcard_clk_divider_write(x)` I tested if by reconfiguring the SDCard SPI clock frequency just after the `spi_sdcard_readMBR()` and it seems to work correctly. Could you test it on your setup?

rob-ng15, 2020-03-20:
Yeah, will have to be tomorrow. I was going to put it at the end of the
goidle function.

Rob.

rob-ng15, 2020-03-21:
Tested and pull request #436 generated.

I guess we can close this off now.

Many thanks, Rob.

enjoy-digital, 2020-03-21:
Thanks a lot for this work #435 and #436 are now merged and SPI-Mode SDCard has been tested on various boards (Xilinx/Altera/Lattice). Support can be added easily to Linux-on-LiteX-Vexriscv by just defining the pins in the platform and adding `spisdcard` capability to the board. This will be very useful. In the future, we'll probably use a proper bootloader to support more easily different filesystems on the SDCard (see https://github.com/enjoy-digital/litex/issues/413), but that will still be useful to have FAT16 support directly in the BIOS for simple systems.

closed 2020-03-21 by enjoy-digital