enjoy-digital/litex #404: Regression in #401

sajattack, 2020-03-02:
Using a litex commit after PR #401 with either the sifive riscv toolchain v9.1.0 or the gnu toolchain v9.2.0 results in the following error message when building linux-on-litex-vexriscv 
`riscv-sifive-elf-ld: unrecognised emulation mode: abi=ilp32`

enjoy-digital, 2020-03-03:
@sajattack: when building emulator.bin of linux on litex-vexriscv, are you also using https://github.com/litex-hub/linux-on-litex-vexriscv/commit/e6ee5b401eda653fbb54eb37e71ac01280d00bad?

sajattack, 2020-03-03:
Perhaps not.

enjoy-digital, 2020-03-04:
When testing without https://github.com/litex-hub/linux-on-litex-vexriscv/commit/e6ee5b401eda653fbb54eb37e71ac01280d00bad the same error is reported. So this is not a regression but Linux-on-LiteX-Vexrsicv that is not in sync with LiteX.

closed 2020-03-04 by enjoy-digital

mithro, 2020-03-05:
It might be worth considering `linux-on-litex-vexriscv` do some type of minimum version check on the LiteX version in use?

enjoy-digital, 2020-03-05:
@mithro: for now `linux-on-litex-vexriscv` is tested and should be use upstream LiteX. In the future we could add a version check if necessary. Here LiteX has been updated but not `linux-on-litex-vexriscv`.