enjoy-digital/litex #407: xilinx: ModuleNotFoundError: No module named 'distutils.spawn'

xobs, 2020-03-03:
When building on a stock Ubuntu 18.04 Python distribution, litex produces the following error:

```
Traceback (most recent call last):
  File "./betrusted-sim.py", line 18, in <module>
    from litex.build.xilinx import XilinxPlatform
  File "/mnt/c/Users/smcro/Code/Betrusted/sim/deps/litex/litex/build/xilinx/__init__.py", line 2, in <module>
    from litex.build.xilinx.programmer import UrJTAG, XC3SProg, FpgaProg, VivadoProgrammer, iMPACT, Adept
  File "/mnt/c/Users/smcro/Code/Betrusted/sim/deps/litex/litex/build/xilinx/programmer.py", line 10, in <module>           from distutils.spawn import find_executable
ModuleNotFoundError: No module named 'distutils.spawn'
```

I have set `compile_gateware=False` as I'm just trying to generate Verilog code.  Commenting out `from distutils.spawn import find_executable` in both `vivado.py` and `programmer.py` gets it working again.

enjoy-digital, 2020-03-03:
This seems similar to https://github.com/pypa/pipenv/issues/2922, can you try installing python3-distutils with: `sudo apt-get install python3-distutils` ?

xobs, 2020-03-04:
That does seem to solve the issue.

It looks like that module isn't included by default on Ubuntu, nor is it included in the official Windows Embedded distribution.  Should this module be used here, given that it's not universally installed, and it doesn't appear to be designed for this usecase, given it's not being used to install additional Python modules:

```
The distutils package provides support for building and installing additional modules into a Python installation. The new modules may be either 100%-pure Python, or may be extension modules written in C, or may be collections of Python packages which include modules coded in both Python and C.

Most Python users will not want to use this module directly, but instead use the cross-version tools maintained by the Python Packaging Authority. 
```

enjoy-digital, 2020-03-04:
Thanks for the feedback. `find_executable` from `distutils.spawn` has been used as part of https://github.com/enjoy-digital/litex/commit/fda18fd6ef20459eacd61365fe7840c599ccc77b since `distutils` was already used for auto-sourcing ISE/Vivado. With https://github.com/enjoy-digital/litex/commit/0b923aa49757e777302ce8d28e35414977fc9db0, auto-sourcing is removed (since not consistent for all build backends, difficult to do correctly without real additional value and since it's in fact probably better to let the user install and choose the toolchain that will be used) and this also removes distutils dependency.

closed 2020-03-04 by enjoy-digital