enjoy-digital/litex #413: Consider alternative options for LiteX BIOS

mithro, 2020-03-05:
Rather than having our own code base, maybe we can reuse something that people are more likely to be familiar with?

mithro, 2020-03-05:
@daveshah1 said at https://github.com/enjoy-digital/litex/issues/346#issuecomment-595422821
> Full U-boot would have to be on SPI flash, but iirc U-Boot has something called SPL (it's a while since I've last used it) that is closer to the LiteX BIOS basically doing DRAM and SPI init type stuff, and might fit in BRAM.

enjoy-digital, 2020-03-05:
@frantony did an initial Barebox port last year: http://lists.infradead.org/pipermail/barebox/2019-June/038284.html from what i can read and feedback from others, this could be a better option than U-boot, but we would need to evaluate that more closely.

xobs, 2020-03-06:
I would very much like "none" to be an option.  In the past it's been a challenge to find the right combination of configuration flags to tell litex to have the PC start at an address other than 0, particularly one that's not aligned to a 256 MB boundary, such as following a bitstream image.

Alternately, making it configurable so that a different build system can work.  For example, in Fomu I had to replace a program called `bios` with an external Makefile.  It would be nice to be able to replace the command as well, so rather than running `make -f ... -C ...`, it would run a custom command.  This would be useful when doing a BIOS in Rust.

enjoy-digital, 2020-03-06:
With https://github.com/enjoy-digital/litex/commit/e801dc0261d7101b148135933f441ed82ca61da8 it's should now be possible to create a SoC without the BIOS that will just resets the CPU at `cpu_reset_address`.

enjoy-digital, 2020-03-09:
@frantony kindly offered to push his initial Barebox work to https://github.com/litex-hub/barebox and wrote instructions to test it in https://github.com/litex-hub/barebox/commit/e66ff6010e6232edd976082dde96f6300e7a3a21

mithro, 2020-03-10:
@enjoy-digital - Awesome! How should do the evaluation of Barebox to the existing BIOS?

frantony, 2020-03-10:
Sorry!

There is an error in the instructions: 

use 
```
$ git clone -b 20200309.litex-vexriscv-linux https://github.com/litex-hub/barebox
```
instead of
```
$ git clone -b 20200309.litex https://github.com/litex-hub/barebox
 ```

I have added instruction to wiki page: https://github.com/litex-hub/barebox/wiki

frantony, 2020-03-10:
@mithro 

> How should do the evaluation of Barebox to the existing BIOS?

Mainline RISC-V linux kernel is intended to run in S-mode (as opposed to MIPS and ARM, e.g.).

Hereby there are at least two possibilities:

* existing BIOS runs in M-mode and provides supervisor execution environment (SEE) for barebox and linux kernel; both linux kernel and barebox run in S-mode;
* the "no BIOS" situation: barebox runs in M-mode and provides SEE for linux kernel.

We have to consider [opensbi](https://github.com/coreboot/coreboot/tree/master/src/arch/riscv) and [coreboot](https://github.com/coreboot/coreboot/tree/master/src/arch/riscv) code.

frantony, 2020-04-04:
I have pushed updated barebox version to github, please see https://github.com/litex-hub/barebox/tree/20200404.litex-vexriscv-linux

This version can run linux from https://github.com/litex-hub/linux-on-litex-vexriscv-prebuilt
please see the docs in the https://github.com/litex-hub/barebox/commit/6889b19424da09d32871f47a7e899f941e749272 commit message.

Here is a small demo:
  * openocd loads fpga bitstream into ARTY board,
  * litex bios loads barebox via tftp and runs it,
  * barebox starts and goes into dialog mode,
  * user enters several barebox commands to load linux files into RAM via tftp and runs linux:

[image]

Barebox image used in this demo (github prevents uploading bin-file, please uncompress before use):
[litex-barebox-20200404-boot.bin.gz](https://github.com/enjoy-digital/litex/files/4431825/litex-barebox-20200404-boot.bin.gz)

enjoy-digital, 2020-04-04:
Thanks a lot @frantony! i'll test it very soon.

DurandA, 2020-05-15:
Just my two cents, but I like the fact that the current BIOS is so minimal that it can be completely understood very quickly.

enjoy-digital, 2020-05-15:
@DurandA: i think the title of the issue is missleading, we are not going to remove the current BIOS, but the idea is to keep it minimal and for complex features, use a proper bootloader.

mithro, 2020-05-15:
@enjoy-digital I renamed the issue to "Investigate a two stage bootloader (keeping LiteX BIOS minimal and using an external bootloader for complex features)"

enjoy-digital, 2020-05-15:
@mithro: ok thanks.