enjoy-digital/litex #414: Support svd generation prior to building software

xobs, 2020-03-06:
Using `litex.soc.doc`, it is possible to build an svd file.  This file can then be used to generate an equivalent of a `csr.h` file in an external program.

However, this process currently needs to be run after the `Builder.build()` step, and therefore needs to be run after the software and gateware are built.

A workaround is to do two passes of `Builder.build(), with the first setting `build_gateware=False, build_software=False`, then generating the svd file, then doing a second pass of `Builder.build()`.

1. Is it difficult to add svd generation to occur before software generation?
2. Should we unconditionally generate svd files, much like we unconditionally generate `csr.h`?

esden, 2020-03-06:
This is related to the icebreaker soc that does not generate a bios as it is not needed. All we want are the CSR header files and SVD file.

It would be nice if the system just generated the SVD file the same way it is done for the csr.h header. We need the SVD file to be able to write Rust programs for the generated SOC.

For reference: https://github.com/icebreaker-fpga/icebreaker-litex-examples/tree/master/soc

bunnie, 2020-03-06:
+1 here, i am implementing a Rust BIOS and we need the SVD file to generate the Bios outside the Litex flow so that we can have the mem.init file for the $readmemh statement...

There is probably also a corollary issue that should be opened as to trigger BIOS inclusion I point Litex to a dummy "do nothing" C Makefile. Without that, it won't try to include the Rust BIOS at all.

enjoy-digital, 2020-03-06:
@xobs, @esden, @bunnie: can you check if https://github.com/enjoy-digital/litex/pull/416 is covering your needs? It also fixes https://github.com/enjoy-digital/litex/issues/415.

enjoy-digital, 2020-03-06:
Implemented by https://github.com/enjoy-digital/litex/pull/416.

closed 2020-03-06 by enjoy-digital