enjoy-digital/litex #415: Regression: SVD generation no longer works

xobs, 2020-03-06:
As a result of https://github.com/enjoy-digital/litex/commit/4f935714de39d65d586b8b01efae322b04bd3db1#diff-1620348e4c19d354b94149c972966387 svd file generation is broken:

```
Generate the documentation by running `sphinx-build -M html build/documentation/ build/documentation/_build`
Traceback (most recent call last):
  File "./icebreaker.py", line 261, in <module>
    main()
  File "./icebreaker.py", line 257, in main
    lxsocdoc.generate_svd(soc, "build/software", vendor="1BitSquared", name="iCEBESOC")
  File "/home/esdentem/projects/icebreaker/icebreaker-litex-examples-git/soc/deps/litex/litex/soc/doc/__init__.py", line 35, in generate_svd
    svd.write(export.get_svd(soc, **kwargs))
  File "/home/esdentem/projects/icebreaker/icebreaker-litex-examples-git/soc/deps/litex/litex/soc/integration/export.py", line 350, in get_svd
    csr_region, csr_data_width=soc.csr_data_width))
TypeError: __init__() missing 1 required positional argument: 'region'
```

esden, 2020-03-06:
If you need to test this, you can try building the icebreaker example soc here: https://github.com/icebreaker-fpga/icebreaker-litex-examples/tree/master/soc

Just uncomment the following line: https://github.com/icebreaker-fpga/icebreaker-litex-examples/blob/master/soc/icebreaker.py#L257

enjoy-digital, 2020-03-06:
Thanks for reporting, this is fixed as part of https://github.com/enjoy-digital/litex/pull/416,  sorry i wasn't testing SVD generation when doing the changes. This will now be easier to test with the `--csr-svd` parameters available on all targets and i will also add it to Travis-CI.

closed 2020-03-06 by enjoy-digital