enjoy-digital/litex #443: Add a release

FFY00, 2020-03-27:
Hey, would it be possible to add a release so that litex can be distributed? Same goes for the other components. Thanks.

mithro, 2020-03-27:
@FFY00 -- How are you planning on distributing LiteX? Distributing it through a Linux distros package management solution is a **very** bad idea. We could be better at auto publishing packages on PyPi however.

FFY00, 2020-03-27:
There's no big difference between distributing it in a distro and in pypi. Could you elaborate please?

mithro, 2020-03-27:
Uploading to PyPi would be automatic as part of the Travis CI on every git commit -- this keeps the delta between people using the git repo and people using PyPi very small. Uploading to distros is out of the LiteX communities hands and hence will happen at random intervals leading to large deltas (which leads to large support costs).

FFY00, 2020-03-27:
> Uploading to PyPi would be automatic as part of the Travis CI on every git commit -- this keeps the delta between people using the git repo and people using PyPi very small. Uploading

Now, that is what I would consider to be a bad idea. It will let users run into issues like this: ae6ef923afc0e9832acae88492b748c601755049. But I supposed it's the same as right now.

I see no reason for this project to not have a release model (should I?). Forcing people to use the latest version will only result in more instability. You can still have a rapid release model (eg. release every other week), that is not an issue. I don't see the need to force the users to use the latest master commit which was merged 1h ago, hasn't had much testing, might have a typo and will maybe break your project.

Just tag a release, if there's a major bug, apply a fix and tag another. Let patches sit in master a few days before asking the end user to use them. By doing this the stability should increase a lot (would have probably prevented #404, #409, #377, etc.). This also allows you to list breaking changes on the release changelog so that users can easily adapt their projects, avoiding issues like #428.

> Uploading to distros is out of the LiteX communities hands and hence will happen at random intervals leading to large deltas (which leads to large support costs).

Not entirely true. Some distros like Debian, Ubuntu, etc. will update the packages at random times, but this is not true for all distros. Distributions like Arch, Gentoo, Majrao, Void, etc. follow the upstream closely and will update the packages pretty quickly.

If the user's distro is too slow to update then it's not really your problem, it's theirs. I have talked about this with a lot of people, the repositories are the most relevant and essential part of a Linux distribution, if your distro is too slow updating packages then you should consider switching to a distro that does effectively fit your needs. I would go as far as saying that the repositories define the distro, all the rest is just configuration defaults, which shouldn't block you in any way.

> which leads to large support costs

Just deny support: "Version X will be EOL in Y month(s), please move to a newer version. If your distribution does not provide an up-to-date package please consider installing from PyPI (via `pip`: `pip install litex`)".

enjoy-digital, 2020-03-27:
@FFY00 I was also thinking about doing releases for quite some time. This will indeed improve stability for users and will also ease development and eventual API changes that will be listed for each release. For me, stability will only increase if distributed packages are based on proper releases and not updated at each commit, so we should probably only distribute the released versions. If developers wants the up to date code, they can use the git repositories and setup scripts. I'll work on that in the next weeks to find a way to create releases but keep this process simple.

FFY00, 2020-03-27:
@enjoy-digital that would be great. While you are figuring out how it would work, please consider following [semver](https://semver.org/) to easily document API changes.

enjoy-digital, 2020-03-28:
@FFY00 for the version number, i don't think we'll use something like MAJOR.MINOR.PATCH, but something simpler like what is done in buildroot: do regular releases (every 3 months?) and just use YYYY.MM for the version with a log of the major changes and API changes. I'd like to avoid something complicated, just have released versions that have been well tested and that users can use if they want something stable and have the major functional/ API changes documented. This should already be a great improvement.

FFY00, 2020-03-28:
That is reasonable. Like you said, just having a release would be a great improvement.

enjoy-digital, 2020-04-28:
A first `2020.04` release has been created with https://github.com/enjoy-digital/litex/commit/56aa7897df99d7ad68ea537ab096c3abdc683666. Merging https://github.com/enjoy-digital/litex/pull/399 introduced a breaking change and it was probably a good opportunity to create the first release. The version number is YYYY.MM and we'll probably do regular releases every 3 months.

closed 2020-04-28 by enjoy-digital

FFY00, 2020-04-28:
In a similar fashion, can https://github.com/litex-hub/pythondata-software-compiler_rt also get a release? This way I can actually package litex :)

enjoy-digital, 2020-04-28:
@FFY00 yes sure, i just did it.

FFY00, 2020-04-28:
Awesome, thanks!