enjoy-digital/litex #453: lxsim --cpu-typ=vexriscv Make exit status 2

opendbf, 2020-04-07:
After following the installation guide in the readme, I ran the lxsim command that is detailed in this readme and recieved an error that appears to trace back to a make call on a make.  I am stumped as I am not familiar with this yet.  This is the last line of the traceback when I run lxsim --cpu-typ=vexriscv fails

subprocess.CalledProcessError: Command '['make', '-C', '/home/me/soc_simsoc_litex_sim/software/libcompiler_rt', '-f', '/home/me/litex/litex/soc/software/libcompiler_rt/Makefile']' returned non-zero exit status 2.

mithro, 2020-04-07:
Please include the full traceback. This just tells us that make faile.

opendbf, 2020-04-07:
Traceback (most recent call last):
  File "/home/haines/.local/bin/lxsim", line 11, in <module>
    load_entry_point('litex', 'console_scripts', 'lxsim')()
  File "/home/haines/litex/litex/tools/litex_sim.py", line 371, in main
    trace_end   = int(args.trace_end))
Sorry, I would give you guys a screenshot but I do not have an editor for it at the moment on this machine.  The following is the fill traceback.  This is on Ubuntu.  

Traceback (most recent call last):
  File "/home/me/.local/bin/lxsim", line 11, in <module>
    load_entry_point('litex', 'console_scripts', 'lxsim')()
  File "/home/me/litex/litex/tools/litex_sim.py", line 371, in main
    trace_end   = int(args.trace_end))
  File "/home/me/litex/litex/soc/integration/builder.py", line 197, in build
    self._generate_rom_software(not self.soc.integrated_rom_initialized)
  File "/home/me/litex/litex/soc/integration/builder.py", line 177, in _generate_rom_software
    subprocess.check_call(["make", "-C", dst_dir, "-f", makefile])
  File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '-C', '/home/me/soc_simsoc_litex_sim/software/libcompiler_rt', '-f', '/home/me/litex/litex/soc/software/libcompiler_rt/Makefile']' returned non-zero exit status 2.

enjoy-digital, 2020-04-07:
@opendbf: can you check that `compiler_rt` has been cloned correctly by looking at `litex/soc/software/compiler_rt` and see if it's not empty? If not, can you try to reinstall LiteX using: https://github.com/enjoy-digital/litex/wiki/Installation?

opendbf, 2020-04-07:
The directory is not empty, however it contains two files: A make file and mulsi3.c.  Running ./litex_setup.py gcc does not appear to do anything as the command exits with no output to the terminal.  I inspected the litex_setup file and there does not appear to be a gcc command option.

enjoy-digital, 2020-04-07:
The two files should be in `libcompiler_rt`, not `compiler_rt`, so if you don't see it, it's probably that the clone has failed. `litex_setup.py` has just been updated to support gcc install, can you make sure you have the last version?

opendbf, 2020-04-08:
I deleted all the files and completed the install again.  The problem seems to be resolved but now there is a new traceback upon running lxsim --cpu-type=vexriscv.  The program cannot find any cross compilation toolchains.  My path is as follows:
$ echo $PATH
/home/me/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin

Traceback (most recent call last):
  File "/home/me/.local/bin/lxsim", line 11, in <module>
    load_entry_point('litex', 'console_scripts', 'lxsim')()
  File "/home/me/enjoy_digital/litex/litex/tools/litex_sim.py", line 371, in main
    trace_end   = int(args.trace_end))
  File "/home/me/enjoy_digital/litex/litex/soc/integration/builder.py", line 191, in build
    self._generate_includes()
  File "/home/me/enjoy_digital/litex/litex/soc/integration/builder.py", line 87, in _generate_includes
    for k, v in export.get_cpu_mak(self.soc.cpu, self.compile_software):
  File "/home/me/enjoy_digital/litex/litex/soc/integration/export.py", line 84, in get_cpu_mak
    ("TRIPLE", select_triple(triple)),
  File "/home/me/enjoy_digital/litex/litex/soc/integration/export.py", line 79, in select_triple
    raise OSError(msg)
OSError: Unable to find any of the cross compilation toolchains:
- riscv64-unknown-elf
- riscv32-unknown-elf
- riscv-none-embed
- riscv64-linux
- riscv-sifive-elf

enjoy-digital, 2020-04-08:
Can you make sure the RISC-V toolchain is in your path? Here is how Travis-CI is doing it:
https://github.com/enjoy-digital/litex/blob/master/.travis.yml#L38-L41

opendbf, 2020-04-08:
The previous issue is resolved after running the commands on the TravisCI script.  Now I am getting the following problem.

chmod -x bios.elf
 OBJCOPY  bios.bin
chmod -x bios.bin
python3 -m litex.soc.software.mkmscimg bios.bin --little
make: Leaving directory '/home/me/enjoy_digital/soc_simsoc_litex_sim/software/bios'
Traceback (most recent call last):
  File "/home/me/.local/bin/lxsim", line 11, in <module>
    load_entry_point('litex', 'console_scripts', 'lxsim')()
  File "/home/me/enjoy_digital/litex/litex/tools/litex_sim.py", line 379, in main
    trace_end   = int(args.trace_end)
  File "/home/me/enjoy_digital/litex/litex/soc/integration/builder.py", line 204, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
  File "/home/me/enjoy_digital/litex/litex/soc/integration/soc.py", line 899, in build
    return self.platform.build(self, *args, **kwargs)
  File "/home/me/enjoy_digital/litex/litex/build/sim/platform.py", line 44, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/me/enjoy_digital/litex/litex/build/sim/verilator.py", line 209, in build
    _compile_sim(build_name, verbose)
  File "/home/me/enjoy_digital/litex/litex/build/sim/verilator.py", line 149, in _compile_sim
    raise OSError("Subprocess failed with {}\n{}".format(p.returncode, "\n".join(error_messages)))
OSError: Subprocess failed with 1
make: Entering directory '/home/me/enjoy_digital/soc_simsoc_litex_sim/gateware'
mkdir -p /home/me/enjoy_digital/soc_simsoc_litex_sim/gateware/obj_dir
make -C /home/me/enjoy_digital/litex/litex/build/sim/core/modules
make[1]: Entering directory '/home/haines/enjoy_digital/litex/litex/build/sim/core/modules'
make -C xgmii_ethernet
make[2]: Entering directory '/home/me/enjoy_digital/litex/litex/build/sim/core/modules/xgmii_ethernet'
cc -c -Wall -O3 -ggdb -fPIC -Werror -Itapcfg/src/include -I../.. -o /home/me/enjoy_digital/soc_simsoc_litex_sim/gateware/obj_dir/xgmii_ethernet.o xgmii_ethernet.c
make[2]: cc: Command not found
../rules.mak:4: recipe for target '/home/me/enjoy_digital/soc_simsoc_litex_sim/gateware/obj_dir/xgmii_ethernet.o' failed
make[2]: *** [/home/me/enjoy_digital/soc_simsoc_litex_sim/gateware/obj_dir/xgmii_ethernet.o] Error 127
make[2]: Leaving directory '/home/me/enjoy_digital/litex/litex/build/sim/core/modules/xgmii_ethernet'
Makefile:8: recipe for target 'xgmii_ethernet' failed
make[1]: *** [xgmii_ethernet] Error 2
make[1]: Leaving directory '/home/me/enjoy_digital/litex/litex/build/sim/core/modules'
/home/me/enjoy_digital/litex/litex/build/sim/core/Makefile:55: recipe for target 'modules' failed
make: *** [modules] Error 2
make: Leaving directory '/home/me/enjoy_digital/soc_simsoc_litex_sim/gateware'
cp: cannot stat 'obj_dir/*.so': No such file or directory

Your help is greatly appreciated! Thank you!

enjoy-digital, 2020-04-10:
It seems you don't have a C compiler installed. Can you do (if on Ubuntu)?:
````
$sudo apt install build-essential
````

opendbf, 2020-04-10:
Sure can! I’ll post the result later. Thanks again!

opendbf, 2020-04-10:
That worked! LxSim works perfectly now! Great work!

Thanks!

enjoy-digital, 2020-05-14:
Closing since resolved.

closed 2020-05-14 by enjoy-digital

pradyu152002, 2022-06-15:
> Can you make sure the RISC-V toolchain is in your path? Here is how Travis-CI is doing it: https://github.com/enjoy-digital/litex/blob/master/.travis.yml#L38-L41

Hey, i seem to be hacing the same issue, where the command litex_sim --cpu-type=vexriscv gives an error:
OSError: Unable to find any of the cross compilation toolchains:
- riscv64-pc-linux-musl
- riscv64-unknown-elf
- riscv64-unknown-linux-gnu
- riscv64-elf
- riscv64-linux
- riscv64-linux-gnu
- riscv-sifive-elf
- riscv64-none-elf
- riscv32-unknown-elf
- riscv32-unknown-linux-gnu
- riscv32-elf
- riscv-none-embed
- riscv-none-elf

The link you gave as a solution to this problem doesnt seem to exist. Could you help me out. Thank you!