enjoy-digital/litex #461: SDRAM failure with current Yosys

mubes, 2020-04-09:
Hi,
when using current master Yosys (42e7e4420768ae4bdf0a7ae32234a9c10939c8f0) to create a ulx3s litex configuration (`ulx3s.py --device=LFE5U-25F`) memtest fails as follows;
```
Initializing SDRAM...
SDRAM now under hardware control
Memtest bus failed: 128/256 errors
Memtest data failed: 393500/524288 errors
Memtest addr failed: 4096/8192 errors
Memory initialization failed
```
Yosys 9b982e92 does not exhibit this problem and bisecting leads to 164dd0f6b298e416bd1ef882f21a4d0b5acfd039 as being the first commit that creates the issue, but that commit appears benign. Also, replacing VexRiscv with picorv32 removes the problem for that Yosys build, at least with some nextpnr seeds. The problem still occurs at current master though.

Conversation with daveshah and tnt (See 1BitSquared discord fpga at 08:00 GMT on 9 Apr 2020) suggests that the issue is potentially timing issues in the SDRAM module.

Not quite sure how to progress with this. If there is testing I can do to help then please let me know. For now this is submitted just so the issue isn't lost, and to offer a test setup if it can be useful.

enjoy-digital, 2020-04-09:
Hi @mubes,

i also experienced instabilities with the SDRAM on the ULX3S. That's probably not directly related to the hardware and tools but more of clocking settings in the targets. I was planning to investigate this more with a scope to understand, i'll do it soon.

mubes, 2020-04-09:
No problem.  If I can help then drop me a note. I've got a ULX3S and 
5A-75B board here I can use, and a couple of half decent scopes.

Regards

enjoy-digital, 2020-04-10:
The issue was in Nextpnr:
- https://twitter.com/fpga_dave/status/1248514139959504897
- https://github.com/YosysHQ/nextpnr/commit/a8111bba830524fd7d572836d07d72a137cc32f0

But this has useful to improve LiteX/LiteDRAM while investigating :)

closed 2020-04-10 by enjoy-digital