enjoy-digital/litex #463: litex_setup.py update - fails on new repos

gojimmypi, 2020-04-12:
Perhaps petty, but [litex_setup.py update](https://github.com/enjoy-digital/litex/blob/master/litex_setup.py) doesn't handle new repos gracefully:

```
...[snip etc]
[updating litejesd204b]...
remote: Enumerating objects: 35, done.
remote: Counting objects: 100% (35/35), done.
remote: Compressing objects: 100% (20/20), done.
remote: Total 29 (delta 18), reused 20 (delta 9), pack-reused 0
Unpacking objects: 100% (29/29), done.
From https://github.com/enjoy-digital/litejesd204b
   8a83b9e..d4b4af9  master     -> origin/master
Updating 8a83b9e..d4b4af9
Fast-forward
 .travis.yml          |  8 +++++++-
 README => README.md  | 51 ++++++++++++++++++++++++++++++++++-----------------
 litejesd204b/core.py | 34 ++++++++++++++++++++++++----------
 litejesd204b/link.py |  8 +++++---
 setup.py             | 19 +------------------
 5 files changed, 71 insertions(+), 49 deletions(-)
 rename README => README.md (69%)
[updating litespi]...
Traceback (most recent call last):
  File "./litex_setup.py", line 125, in <module>
    os.chdir(os.path.join(current_path, name))
FileNotFoundError: [Errno 2] No such file or directory: '/mnt/c/workspace/litespi'
```
of course, a quick manual clone resolved:

```
git clone https://github.com/litex-hub/litespi.git
```

enjoy-digital, 2020-04-12:
@gojimmypi: thanks for reporting. An exception is now raised with https://github.com/enjoy-digital/litex/commit/a8bf02167aa26934aa64605c6f12bc7142cd2313 indicating that you need to re-run `./litex_setup.py init` and `./litex_setup.py install`.

closed 2020-04-12 by enjoy-digital

gojimmypi, 2020-04-12:
@enjoy-digital thanks for the prompt response. Seems to have worked.

Note when I initially ran `./litex_setup.py init` I saw the errors below. In my case, I needed to add the `--user` parameter. I wonder if there's some way to detect the permission problem and prompt the user with something like "_perhaps you meant to add the --user parameter?_". Fortunately I had this in a script to remind me.

```
$ ./litex_setup.py init
[cloning migen]...
fatal: destination path 'migen' already exists and is not an empty directory.
Traceback (most recent call last):
  File "./litex_setup.py", line 95, in <module>
    shell=True)
  File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command 'git clone https://github.com/m-labs/migen --recursive' returned non-zero exit status 128.
```
also, I'm pretty sure the first time I ran `./litex_setup.py install` I did not do it with `sudo`, yet I see this error.

```
 ./litex_setup.py install
[installing migen]...
/usr/lib/python3.6/distutils/dist.py:261: UserWarning: Unknown distribution option: 'long_description_content_type'
  warnings.warn(msg)
running develop
error: can't create or remove files in install directory

The following error occurred while trying to add or remove files in the
installation directory:

    [Errno 13] Permission denied: '/usr/local/lib/python3.6/dist-packages/test-easy-install-18288.write-test'

The installation directory you specified (via --install-dir, --prefix, or
the distutils default setting) was:

    /usr/local/lib/python3.6/dist-packages/

Perhaps your account does not have write access to this directory?  If the
installation directory is a system-owned directory, you may need to sign in
as the administrator or "root" account.  If you do not have administrative
access to this machine, you may wish to choose a different installation
directory, preferably one that is listed in your PYTHONPATH environment
variable.

For information on other options, you may wish to consult the
documentation at:

  https://setuptools.readthedocs.io/en/latest/easy_install.html

Please make the appropriate changes for your system and try again.

Traceback (most recent call last):
  File "./litex_setup.py", line 112, in <module>
    shell=True)
  File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command 'python3 setup.py develop' returned non-zero exit status 1.
```

enjoy-digital, 2020-04-12:
Thanks for the feedback. I think we'll try to improve things after merging https://github.com/enjoy-digital/litex/pull/399, that is also doing some changes in `litex_setup.py`.