enjoy-digital/litex #465: migen conv_output.py FileNotFoundError: [Errno 2] No such file or directory: 'top.v'

gojimmypi, 2020-04-12:
I updated my toolchain today... broke it. I re-ran `./litex_setup.py` (with `--user param`) as noted in https://github.com/enjoy-digital/litex/issues/463#issuecomment-612652220 - everything up to date, but the `top.v` file seems to not be generated anymore. Any suggestions where to look? Thanks

```
gojimmypi:/mnt/c/workspace/litex-boards/litex_boards/targets
$ ./ulx3s.py --device LFE5U-45F --cpu-type picorv32
INFO:SoC:        __   _ __      _  __
INFO:SoC:       / /  (_) /____ | |/_/
INFO:SoC:      / /__/ / __/ -_)>  <
INFO:SoC:     /____/_/\__/\__/_/|_|
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2020-04-12 12:18:58)

[...snip...]

make: Leaving directory '/mnt/c/workspace/litex-boards/litex_boards/targets/soc_basesoc_ulx3s/software/libbase'
make: Entering directory '/mnt/c/workspace/litex-boards/litex_boards/targets/soc_basesoc_ulx3s/software/libnet'
 CC       microudp.o
 AR       libnet.a
make: Leaving directory '/mnt/c/workspace/litex-boards/litex_boards/targets/soc_basesoc_ulx3s/software/libnet'
make: Entering directory '/mnt/c/workspace/litex-boards/litex_boards/targets/soc_basesoc_ulx3s/software/bios'
 CC       isr.o
 CC       sdram.o
 CC       sdcard.o
 CC       main.o
 CC       boot.o
 LD       bios.elf
chmod -x bios.elf
 OBJCOPY  bios.bin
chmod -x bios.bin
python3 -m litex.soc.software.mkmscimg bios.bin --little
make: Leaving directory '/mnt/c/workspace/litex-boards/litex_boards/targets/soc_basesoc_ulx3s/software/bios'
Traceback (most recent call last):
  File "./ulx3s.py", line 99, in <module>
    main()
  File "./ulx3s.py", line 96, in main
    builder.build(**builder_kargs)
  File "/mnt/c/workspace/litex/litex/soc/integration/builder.py", line 204, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
  File "/mnt/c/workspace/litex/litex/soc/integration/soc.py", line 899, in build
    return self.platform.build(self, *args, **kwargs)
  File "/mnt/c/workspace/litex/litex/build/lattice/platform.py", line 34, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/mnt/c/workspace/litex/litex/build/lattice/trellis.py", line 202, in build
    v_output.write(top_file)
  File "/mnt/c/workspace/migen/migen/fhdl/conv_output.py", line 31, in write
    with open(main_filename, "w") as f:
FileNotFoundError: [Errno 2] No such file or directory: 'top.v'
gojimmypi:/mnt/c/workspace/litex-boards/litex_boards/targets
```

enjoy-digital, 2020-04-12:
It seems Migen is not able to create the `top.v` file, but i'm not sure to understand why. I would recommend doing a clean re-install.

gojimmypi, 2020-04-12:
I was doing this on WSL and noticed in task manager that Python was taking up a lot of CPU, even though my sessions all seemed to be sitting idle. I closed all my WSL prompts and the Python process continued. (I was not using Python anywhere else). I ended up rebooting, and all appears to be better now. I did not re-install Migen, so I don't know what the root problem was. 

I'm happy to close this, but perhaps there's some way to detect and warn about the problem? I don't know how we'd check this, now that it is not misbehaving. Perhaps close and re-open if/when it reoccurs?

mithro, 2020-04-13:
Windows can't deal with two programs trying to write to the same file and it sounds like you had another Python process that had hung writing to the file in the background. The other option is something went wrong earlier when the code tried to create the gateware directory.

xobs, 2020-04-13:
It could also be Windows Defender blocking access to the file.  I've seen that, but I'm not sure how that would interact with WSL.

gojimmypi, 2020-04-13:
It seems to behaving now. Well, other than my new [LiteX memtest](https://github.com/litex-hub/litex-boards/issues/66) problem that seems to have crept in while I was fussing with upgrading the toolchain. But the Windows OS does indeed "see" the underlying WSL filesystem. When doing a big git clone / build - my AV software is usually quite busy scanning. Indeed that may likely have been the problem with access being blocked.

During the build process - are background threads spawned? If so, that may have been how Python got stuck. At one point, I did do a radical git change to go back in time, as I didn't have memtest problems in the past. A bunch of files were deleted and replaced. If there was a background process running on one of them, that could be bad.  

Also: I did have nearly a dozen concurrently open WSL sessions. I don't _think_ that I tried to build concurrently. At least when I closed each one - they were all idle.

enjoy-digital, 2020-04-13:
@gojimmypi: good if it's behaving correctly now. I'm not sure we can prevent this from happening if the OS is not in a clean stage before running the scripts. I suggest we close this and you could re-open it in case you see the situation happen again. I'll look at https://github.com/litex-hub/litex-boards/issues/66.

closed 2020-04-13 by enjoy-digital