enjoy-digital/litex #468: GenericProgrammer: no module named `requests`

xobs, 2020-04-16:
This got picked up by CI last night:

```
08:05:25 + python3 ./foboot-bitstream.py --lx-ignore-git --revision pvt --with-dsp --placer heap --with-debug usb --seed 4
08:05:25 Traceback (most recent call last):
08:05:25   File "./foboot-bitstream.py", line 19, in <module>
08:05:25     from litex.build.lattice.platform import LatticePlatform
08:05:25   File "/var/lib/jenkins/workspace/foboot/hw/deps/litex/litex/build/lattice/__init__.py", line 2, in <module>
08:05:25     from litex.build.lattice.programmer import LatticeProgrammer
08:05:25   File "/var/lib/jenkins/workspace/foboot/hw/deps/litex/litex/build/lattice/programmer.py", line 8, in <module>
08:05:25     from litex.build.generic_programmer import GenericProgrammer
08:05:25   File "/var/lib/jenkins/workspace/foboot/hw/deps/litex/litex/build/generic_programmer.py", line 8, in <module>
08:05:25     import requests
08:05:25 ModuleNotFoundError: No module named 'requests'
```

Probably caused by https://github.com/enjoy-digital/litex/commit/5e149ceda2e60ffce2740f8ef60f66428bbba449

Full log at: https://ci.betrusted.io/job/foboot/43/console

mithro, 2020-04-16:
@xobs Guess the setup.py / requirements.txt files should add requests as a requirement?

enjoy-digital, 2020-04-16:
Thanks for catching it, requests has already been added to setup.py with https://github.com/enjoy-digital/litex/commit/2d018826532e486b94615c6b9cee4f16a924dba2 but i just moved the import to only do it when needed with https://github.com/enjoy-digital/litex/commit/c0c5ae558ace64d37741c135af0d5c7be05c23c8.

closed 2020-04-16 by enjoy-digital