enjoy-digital/litex #471: Add SERV support

enjoy-digital, 2020-04-23:
SERV is a bit-serial  RISC-V core that could be useful in some designs to replace a complex FSM by a CPU while still minimizing resource usage. This would be a perfect fit for LiteDRAM standalone core where the CPU is only doing the DRAM initialization.

Initial SERV integration has been pushed to https://github.com/enjoy-digital/litex/tree/serv but needs to be finished.

The LiteX SoC with  SERV CPU can be simulated with: `litex_sim --cpu-type=serv` and waveforms can be generated by adding `--trace` to the command. The waveform will then be located in `soc_simsoc_litex_sim/gateware/dut.vcd`.

For bring up, the UART is configured in polling mode with `self.add_constant("UART_POLLING")`. This allow doing the initial tests without interrupt support.

gregdavill, 2020-04-26:
So I just setup a project here using SERV. So I wish I'd seen this a few hours ago.

One thing I hit when bringing up my project was the wishbone address bus. LiteX uses a word addressed bus, Serv a byte addressed one, which provides the lowest 2 bits. So I had to drop these on both instruction and data buses.
```python
i_addr = Signal(32)
d_addr = Signal(32)

self.comb += [
    ibus.sel.eq(0xF),
    ibus.stb.eq(ibus.cyc),
    dbus.stb.eq(dbus.cyc),
    ibus.adr.eq(i_addr[2:]),
    dbus.adr.eq(d_addr[2:]),
]

self.cpu_params = dict(
...
        o_o_ibus_adr      = i_addr,
...
        o_o_dbus_adr      = d_addr,
    )
```

When I get a chance I'll take a look at your initial integration.

enjoy-digital, 2020-04-26:
@gregdavill: thanks for the feedback. I just fixed these points in the initial integration https://github.com/enjoy-digital/litex/commit/2efd939d06ba7903892fc5a742838b49ab1728fb.
There is still probably something wrong somewhere since the simulation is still not booting, i will have a look in the next days. Is your project available somewhere to have a look?

gregdavill, 2020-04-26:
I think we need to use `serv_rf_top` here:
```diff
- self.specials += Instance("serv_top", **self.cpu_params)
+ self.specials += Instance("serv_rf_top", **self.cpu_params)
```
https://github.com/enjoy-digital/litex/compare/serv#diff-9f586f8936c1ea0b2741be8735346161R91

This adds the register file to the CPU, providing just the wishbone interface out. I think it's separated like this so that potentially the RF and instruction memory can be combined into a single RAM block for an even smaller footprint.

gregdavill, 2020-04-26:
Also I'm not sure if this is just a bug I'm seeing. But yosys throws an error on this default parameter.
```diff
greg@xps:~/projects/DiVA-firmware/hw/deps/serv (master) $ git diff
diff --git a/rtl/shift_reg.v b/rtl/shift_reg.v
index 64b5ed7..5c4d644 100644
--- a/rtl/shift_reg.v
+++ b/rtl/shift_reg.v
@@ -1,5 +1,5 @@
 module shift_reg
-  #(parameter LEN = 0,
+  #(parameter LEN = 2,
     parameter INIT = 0)
   (
    input wire           clk,
```

enjoy-digital, 2020-04-26:
Thanks @gregdavill, i indeed made a typo on the top level name, with `serv_rf_top` this is now working in simulation :) I'll do some tests with Yosys and on hardware tomorrow.

gregdavill, 2020-04-27:
An option for overcoming the issues with that default parameter, from https://github.com/olofk/serv/issues/24

We can use the `-defer` option for `read_verilog` in yosys. This avoids the issue.
I'll make a PR so you can see the changes.
```
    -defer
        only read the abstract syntax tree and defer actual compilation
        to a later 'hierarchy' command. Useful in cases where the default
        parameters of modules yield invalid or not synthesizable code.
```

enjoy-digital, 2020-04-27:
SERV just landed in LiteX, thanks @gregdavill, @olofk for fixing the issues, finding the missing pieces.

@olofk: in case you want to test, if LiteX and Verilator are installed, you can just do: `litex_sim --cpu-type=serv` or if you want to test on hardware, you can select one of the targets in https://github.com/litex-hub/litex-boards/tree/master/litex_boards/targets and do `./target.py --cpu-type=serv`. It should also be directly usable with the LiteDRAM generator (but since it's slow, we'll maybe have to reduce the memtest size).

The submodule is not yet properly integrated (for now we are just doing a git clone to get the sources) and i will wait that we merge https://github.com/enjoy-digital/litex/pull/399 and create a SERV python package to close this.

olofk, 2020-04-27:
I'll give it a try and let you know how it goes. Interesting about the python packages. Haven't looked at them closely but it sounds like it would make sense to parse FuseSoC .core files to get the info you need. Especially for more complex verilog/vhdl cores or subsystems which potentially depend on other cores

enjoy-digital, 2020-04-28:
Thanks @olofk, when using external Verilog/VHDL, relying on FuseSoC is something we should indeed explore.

enjoy-digital, 2020-04-28:
#399 has been merged and https://github.com/litex-hub/pythondata-cpu-serv has been created, so SERV now comes pre-installed and can just be used as any other CPU with just `--cpu-type=serv` :)

closed 2020-04-28 by enjoy-digital