enjoy-digital/litex #480: Tests failing (minerva target)

FFY00, 2020-04-28:
It complains about `/build/python-litex/src/litex-2020.04/litex/soc/cores/cpu/minerva/verilog/cli.py` being missing, which is true (https://github.com/enjoy-digital/litex/tree/2020.04/litex/soc/cores/cpu/minerva).

Seems like the CI is skipping this target and so the error sneaked in.

Just a side note, if you want, you can use an arch docker image in the CI as we should have all the required dependencies.

```
============================= test session starts ==============================
platform linux -- Python 3.8.2, pytest-5.4.1, py-1.8.1, pluggy-0.13.1
rootdir: /build/python-litex/src/litex-2020.04
collected 77 items

test/test_axi.py ............                                            [ 15%]
test/test_bitbang.py ..                                                  [ 18%]
test/test_clock.py ..........                                            [ 31%]
test/test_code_8b10b.py .....                                            [ 37%]
test/test_csr.py ...                                                     [ 41%]
test/test_ecc.py .....                                                   [ 48%]
test/test_emif.py .                                                      [ 49%]
test/test_gearbox.py .                                                   [ 50%]
test/test_hyperbus.py ...                                                [ 54%]
test/test_i2s.py .                                                       [ 55%]
test/test_icap.py ..                                                     [ 58%]
test/test_packet.py ....                                                 [ 63%]
test/test_prbs.py ..                                                     [ 66%]
test/test_spi.py .....                                                   [ 72%]
test/test_spi_opi.py .                                                   [ 74%]
test/test_stream.py ..                                                   [ 76%]
test/test_targets.py .............F....                                  [100%]

=================================== FAILURES ===================================
______________________ TestTargets.test_variants_minerva _______________________

self = <test.test_targets.TestTargets testMethod=test_variants_minerva>

    @unittest.skipIf(RUNNING_ON_TRAVIS, "No nMigen/Yosys on Travis-CI")
    def test_variants_minerva(self):
>       self.run_variants("minerva", ('standard',))

test/test_targets.py:172:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
test/test_targets.py:150: in run_variants
    self.run_variant(cpu, v)
test/test_targets.py:161: in run_variant
    subprocess.check_output(cmd, shell=True)
/usr/lib/python3.8/subprocess.py:411: in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

input = None, capture_output = False, timeout = None, check = True
popenargs = ('litex/boards/targets/simple.py litex.boards.platforms.arty     --cpu-type=minerva              --cpu-variant=standard           --no-compile-software       --no-compile-gateware       --uart-name=stub        ',)
kwargs = {'shell': True, 'stdout': -1}
process = <subprocess.Popen object at 0x7fccee6f5e50>, stdout = b''
stderr = None, retcode = 1

    def run(*popenargs,
            input=None, capture_output=False, timeout=None, check=False, **kwargs):
        """Run command with arguments and return a CompletedProcess instance.

        The returned instance will have attributes args, returncode, stdout and
        stderr. By default, stdout and stderr are not captured, and those attributes
        will be None. Pass stdout=PIPE and/or stderr=PIPE in order to capture them.

        If check is True and the exit code was non-zero, it raises a
        CalledProcessError. The CalledProcessError object will have the return code
        in the returncode attribute, and output & stderr attributes if those streams
        were captured.

        If timeout is given, and the process takes too long, a TimeoutExpired
        exception will be raised.
[... 110 lines cut ...]
  File "/build/python-litex/src/litex-2020.04/litex/soc/integration/builder.py", line 193, in build
    self.soc.finalize()
  File "/usr/lib/python3.8/site-packages/migen/fhdl/module.py", line 156, in finalize
    subfragments = self._collect_submodules()
  File "/usr/lib/python3.8/site-packages/migen/fhdl/module.py", line 149, in _collect_submodules
    r.append((name, submodule.get_fragment()))
  File "/usr/lib/python3.8/site-packages/migen/fhdl/module.py", line 102, in get_fragment
    self.finalize()
  File "/usr/lib/python3.8/site-packages/migen/fhdl/module.py", line 157, in finalize
    self.do_finalize(*args, **kwargs)
  File "/build/python-litex/src/litex-2020.04/litex/soc/cores/cpu/minerva/core.py", line 109, in do_finalize
    self.elaborate(
  File "/build/python-litex/src/litex-2020.04/litex/soc/cores/cpu/minerva/core.py", line 104, in elaborate
    raise OSError("Unable to elaborate Minerva CPU, please check your nMigen/Yosys install")
OSError: Unable to elaborate Minerva CPU, please check your nMigen/Yosys install
=============================== warnings summary ===============================
test/test_ecc.py::TestECC::test_ecc
  /build/python-litex/src/litex-2020.04/litex/gen/sim/core.py:278: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated since Python 3.3, and in 3.9 it will stop working
    if (isinstance(v, collections.Iterable)

-- Docs: https://docs.pytest.org/en/latest/warnings.html
=========================== short test summary info ============================
FAILED test/test_targets.py::TestTargets::test_variants_minerva - subprocess....
============== 1 failed, 76 passed, 1 warning in 93.56s (0:01:33) ==============
```

FFY00, 2020-04-28:
For people running into the same issue, you can just pass `-k-test_variants_minerva` to pytest to ignore this test :smile:.

enjoy-digital, 2020-04-28:
@FFY00: the Minerva support is still experimental and requires manual installation of nMigen/Yosys to generate the Verilog, so is indeed not tested in CI. That's something we could work on for the next release :)

mithro, 2020-04-28:
@enjoy-digital - FYI -- If we used conda on CI, then we could easily test nMigen as it provides Yosys....

The conda steps these days are pretty simple (and almost work on Mac / Win / Linux);
`conda-instal.sh`
```
# Download and install miniconda
wget https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-x86_64.sh -O miniconda.sh
bash miniconda.sh -b -p $HOME/miniconda

# Set up environment
source "$HOME/miniconda/etc/profile.d/conda.sh"
hash -r
conda config --set always_yes yes --set changeps1 no # Only needed on CI.
conda env create --file environment.yml
```

`conda-enter.sh`
```
source "$HOME/miniconda/etc/profile.d/conda.sh"
hash -r
conda activate litex
<run commands here>
```

The `environment.yml` sets up which channels to use and binaries to install. It can also reference a normal `requirements.txt` for Python requirements.

Conda also has nextpnr + vpr (with symbiflow for series 7) so we could also do proper place and route runs......

mithro, 2020-04-28:
FYI - @FFY00 Also did an experiment around creating a github-action which I believe is docker based and could be used as another alternative?

enjoy-digital, 2020-04-28:
@mithro: the question is is not necessarily CI here, it's that nMigen/Minerva support in LiteX is still experimental and nMigen is also not yet installed with `litex_setup.py`. We will first need to think about how we'll do it in `litex_setup.py` (but without Conda if possible) and this will also solves the issue with CI.

enjoy-digital, 2020-04-29:
I just created https://github.com/enjoy-digital/litex/issues/489 to fix the Minerva dependencies. For now you have to skip this unit-test since this was experimental. @FFY00: are you able to skip it on your side for packaging or should we do the modifications here?

FFY00, 2020-04-29:
Yes, I can skip the test, as I said the my 2nd comment :smile:.

closed 2020-05-02 by enjoy-digital