enjoy-digital/litex #486: Denoting active low in pin definitions?

TomKeddie, 2020-04-29:
The arduino platform has done a good job of centralising a few definitions like LED_BUILTIN that mean the same piece of code will build for most (all?) boards.  It would be great to come up with a few baseline pins that most platforms support (like a led and a button) so we can use the same simple examples for large number of platforms.

One issue I see with this is many platforms define their io as active low for obvious reasons. Am wondering if we could consider an object like ActiveLow() that generates an inversion in the logic at the input/output.  Perhaps this results in more gates since I think we'd need all outputs to be combinatorial.

For example the icebreaker defintions would change from 
```
    ("user_led_n",    0, Pins("11"), IOStandard("LVCMOS33")),
    ("user_led_n",    1, Pins("37"), IOStandard("LVCMOS33")),
    ("user_btn_n",    0, Pins("10"), IOStandard("LVCMOS33")),
```
to
```
    ("user_led",    0, Pins("11"), ActiveLow(), IOStandard("LVCMOS33")),
    ("user_led",    1, Pins("37"), ActiveLow(), IOStandard("LVCMOS33")),
    ("user_btn",    0, Pins("10"), ActiveLow(), IOStandard("LVCMOS33")),
```
I realise most (all?) of the _io definitions feed into the constraints file so perhaps this is an architectural sin.  I think the concept is a good one perhaps these is a better way to do this?  I'm a novice litex/python hacker.

Sorry if this has been discussed already, I did a bit of a search and found nothing.

enjoy-digital, 2020-04-29:
@TomKeddie: thanks for the feedback, in fact i'd like to keep the Platform files very low level and avoid too much abstrations there: try to keep a direct mapping from/to a Xilinx .xdc file, Quartus .qsf, Lattice .lpf, etc...

The usual rule here in the `Platforms` is to:
- don't use suffix for active high signals.
- use `_n` suffix for active low signals.

That's also a common practice when doing FPGA development with others tools.

We could probably manage the abstraction you want by keeping the current naming rule and do the following:
- if the user do a `platform.request("user_led", 0)` and the platform only define `user_led_n:0`, we could return a `Signal` that could be used by the user, that would be linked to `user_led_n` and would have an `active_low` attribute that means it needs to be inverted when generating the verilog. (we can only do the inversion in the end since we need to know if it's an input or output).
- if the user do a `platform.request("user_led_n", 0)`, then we can consider the user will handle the fact that this signal is active low directly in the design.

The main advantages i see with this are that:
- we don't break current designs.
- we keep common naming practices.
- experimented users can still request the signal with the `_n` suffix and handle this inversion in their logic.

Would it be fine for you?

TomKeddie, 2020-04-29:
> Would it be fine for you?

Hi Florent, thanks for being open to the feedback.  I think you're right, the current model is very clean.  As much as auto-sensing the _n in example code is ambigous it's probably the way to go.  I might have a go at a multiplatform example to kick this off.   Thanks.

enjoy-digital, 2020-04-29:
@TomKeddie: I think we can keep this open to continue discussing and try to implement the solution i was suggesting. This will also be an opportunity to verify that all `user_leds`/`user_btns` are defined with the right polarity (which i'm not sure for all platforms).

TomKeddie, 2020-04-29:
> This will also be an opportunity to verify that all user_leds/user_btns are defined with the right polarity (which i'm not sure for all platforms).

Good idea thanks, I'll take this on.  Will update progress here.