enjoy-digital/litex #493: DRAM tests fail for Genesys board at 125MhZ

scanakci, 2020-05-02:
Unfortunately, memory tests for Genesys boards are failing for Rocket, VexRiscv, (and for BlackParrot) at 125 MhZ. I used almost the latest versions of LiteX and LiteDRAM. If I try lower frequencies, LiteX BIOS does not come to the screen and I observe meaningless characters. I hope my FPGA did not die :)

LiteDRAM(dfe6f90569e7bb505c7b2a1fe0938308019eed1b)
LiteX(e853cac6b618a226d8213789724be6b3786d65ed)

`    __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on May  1 2020 18:20:17
 BIOS CRC passed (1cf4edb6)

 Migen git sha1: 5b5e4fd
 LiteX git sha1: e853cac

--=============== SoC ==================--
CPU:       VexRiscv @ 125MHz
ROM:       32KB
SRAM:      4KB
L2:        8KB
MAIN-RAM:  1048576KB

--========== Initialization ============--
Initializing SDRAM...
SDRAM now under software control
Write leveling:
Command/Clk scan:
|11111111000000000001111111111111| best: 21
Data scan:
m0: |00000111111111111000000000| delay: 05
m1: |00000011111111111100000000| delay: 06
m2: |00011111111111100000000000| delay: 03
m3: |00011111111111100000000000| delay: 03
Read leveling:
m0, b0: |00000000000000000000000000000000| delays: -
m0, b1: |00000000000000000000000000000000| delays: -
m0, b2: |00000000000000000000000000000000| delays: -
m0, b3: |00000000000000000000000000000000| delays: -
m0, b4: |00000000000000000000000000000000| delays: -
m0, b5: |00000000000000000000000000000000| delays: -
m0, b6: |00000000000000000000000000000000| delays: -
m0, b7: |00000000000000000000000000000000| delays: -
best: m0, b0 delays: -
m1, b0: |00000000000000000000000000000000| delays: -
m1, b1: |00000000000000000000000000000000| delays: -
m1, b2: |00000000000000000000000000000000| delays: -
m1, b3: |00000000000000000000000000000000| delays: -
m1, b4: |00000000000000000000000000000000| delays: -
m1, b5: |00000000000000000000000000000000| delays: -
m1, b6: |00000000000000000000000000000000| delays: -
m1, b7: |00000000000000000000000000000000| delays: -
best: m1, b0 delays: -
m2, b0: |00000000000000000000000000000000| delays: -
m2, b1: |00000000000000000000000000000000| delays: -
m2, b2: |00000000000000000000000000000000| delays: -
m2, b3: |00000000000000000000000000000000| delays: -
m2, b4: |00000000000000000000000000000000| delays: -
m2, b5: |00000000000000000000000000000000| delays: -
m2, b6: |00000000000000000000000000000000| delays: -
m2, b7: |00000000000000000000000000000000| delays: -
best: m2, b0 delays: -
m3, b0: |00000000000000000000000000000000| delays: -
m3, b1: |00000000000000000000000000000000| delays: -
m3, b2: |00000000000000000000000000000000| delays: -
m3, b3: |00000000000000000000000000000000| delays: -
m3, b4: |00000000000000000000000000000000| delays: -
m3, b5: |00000000000000000000000000000000| delays: -
m3, b6: |00000000000000000000000000000000| delays: -
m3, b7: |00000000000000000000000000000000| delays: -
best: m3, b0 delays: -
SDRAM now under hardware control
Memtest bus failed: 256/256 errors
Memtest data failed: 524273/524288 errors
Memtest addr failed: 8192/8192 errors
Memory initialization failed

--============= Console ================--
litex>
`

enjoy-digital, 2020-05-02:
With the new automatic Command/Clk scan, it's possible you will need to set `cmd_latency=1` here: https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/genesys2.py#L55-L58, similar to what is done on the KC705 for example: https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/kc705.py#L61.

scanakci, 2020-05-03:
Thanks @enjoy-digital. I did some tests after setting cmd_latency=1. Here are my results.
Overall, 100MhZ and 125MhZ worked fine but 60-75MhZ failed. I am using 100MhZ for now without any issues.

               | VEX   | Rocket | BP
60MhZ  | N/A    | failed   | failed
75MhZ   | failed | N/A     | failed
100MhZ | pass  | N/A     | pass
125MhZ | pass  | pass   | pass

enjoy-digital, 2020-05-03:
Thanks for the feedback @scanakci, that's good you have a working version, we are currently working on avoiding the manual adjustements that are currently needed. I have a Genesys2 board and will be able to test this.

enjoy-digital, 2020-05-05:
Closing since fixed.

closed 2020-05-05 by enjoy-digital