enjoy-digital/litex #495: Vivado settings sourcing

mateusz-holenko, 2020-05-04:
0b923aa removes the support for automatic sourcing of Vivado settings - now all vendor tools are required to be in PATH.

Unfortunatelly this approach causes problems for ISE/Vivado tools as they come with/require many standard libraries in specific versions. As a result adding Vivado directory to PATH allows to run the tool itself, but might cause other software (e.g., vim) to crash rendering the shell session less usable for the user.

The point of sourcing Vivado settings in the `build_top.sh` script is to avoid those problems.

mateusz-holenko, 2020-05-04:
cc @mithro

enjoy-digital, 2020-05-04:
@mateusz-holenko: the comment in the commit was maybe no complete: Vivado can be in the PATH or the setting file sourced manually. When sourced manually, the behavior will be similar to the previous one. Have you tried that?

Sourcing things automatically is not something simple if we want to do it correctly and i'm not sure it should be in the scope of LiteX.

mithro, 2020-05-04:
@enjoy-digital - The problem we have run into is that sourcing the vivado settings causes lots of other non-vivado tools to crash (including Python), thus we need to source the settings *just* before we run Vivado and no sooner than that.

mithro, 2020-05-04:
It seems like sourcing the vivado settings at the top of `build.sh` just before we run vivado commands would fix that.

BTW sourcing settings.sh can also cause Yosys to crash.

enjoy-digital, 2020-05-04:
We could use environment variables like: `LITEX_ENV_VIVADO`, `LITEX_ENV_ISE` let LiteX integrate the sourcing in the `build.sh` if the environment variables are defined, but the automatic sourcing was complicated and causing others issues so that was not a good solution.

mithro, 2020-05-04:
@enjoy-digital - We are happy to provide a "insert these commands before running vivado" type script....

enjoy-digital, 2020-05-04:
@mithro: thanks, i would be fine with it. With 0b923aa i just wanted to avoid the automatic sourcing that cause others issue when a specific version of the tools is required. Here it would still be manual (The user will have to specify the environment that will be used) and sourced at the right time to avoid others issues.

mithro, 2020-05-04:
@enjoy-digital I guess we could monkey patch the `_build_script` method?

enjoy-digital, 2020-05-04:
@mithro: yes this seems fine.

mithro, 2020-05-04:
@mateusz-holenko Can you take a stab at that?

mateusz-holenko, 2020-05-05:
Sure, I'll work on this

enjoy-digital, 2020-05-19:
@mateusz-holenko: can you check https://github.com/enjoy-digital/litex/commit/b4267a79011fa7dd7cb84961196875d3be81f077 is doing what you wanted? On Linux distribution we you can now just do:
````
export LITEX_ENV_VIVADO=/opt/Xilinx/Vivado/201x.x/
./arty.py --build
````
And sourcing will be done automatically just before running Vivado.

enjoy-digital, 2020-05-26:
Closing since a solution has been implemented with https://github.com/enjoy-digital/litex/commit/b4267a79011fa7dd7cb84961196875d3be81f077.

closed 2020-05-26 by enjoy-digital

mateusz-holenko, 2020-05-26:
Sorry, I haven't test it yet, but if I see any problems I'll just open a new issue. Thanks for the support!