enjoy-digital/litex #501: Memory-mapped ECP5SPIFlash

DurandA, 2020-05-09:
While trying to add SPI flash support to the [_ecp5_evn_](https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/ecp5_evn.py), I noticed that the [`ECP5SPIFlash`](https://github.com/enjoy-digital/litex/blob/fbbbdf03b583a5571069fb747170e6749ad118e6/litex/soc/cores/spi_flash.py#L372) is not memory-mapped.

Does it mean no BIOS support to boot images from flash?

Is using SPI flash from ECP5 just (compared to "normal" `SpiFlash`) a matter of driving the clock from the _USRMCLK_ hardware block or is there more to it?

enjoy-digital, 2020-05-09:
`ECP5SPIFlash` is indeed not MMAP. If you want to to have MMAP SPI Flash support, you can use the `add_spi_flash` method (https://github.com/enjoy-digital/litex/blob/master/litex/soc/integration/soc.py#L1185-L1201)  as done here: https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/make.py#L375-L379

I will also have a look to integrate `SPIFlash` to targets in the next days.

DurandA, 2020-05-09:
Many thanks. I missed the fact that `add_spi_flash` takes care of the clock primitive for the ECP5.

DurandA, 2020-05-11:
What are the reasons to have disabled 1x mode support?
https://github.com/enjoy-digital/litex/blob/1e610600f6a088941700d6736e0621010527ef64/litex/soc/integration/soc.py#L1193

The SPI flash ([_Macronix MX25L12833F_](https://www.macronix.com/Lists/Datasheet/Attachments/7447/MX25L12833F,%203V,%20128Mb,%20v1.0.pdf)) of my _ecp5_evn_ board requires prior configuration to enable QIO, so I forced the 1x mode in LiteX instead and I observe a very strange behaviour when reading/writing memory from BIOS.

For example, I wrote 0xFF, 0xFE, 0xFD, 0xFC at offset 0 on the flash (using an external tool) and when I read the memory mapped value from the bios using `mr 0x20000000 4` it reads _00 ff fd fc_ and writing memory seems to have no effect.

Are there known defects with the 1x mode or do you suggest that I investigate on the issue with a logic analyser?

enjoy-digital, 2020-05-11:
The 1x mode should work but hasn't been tested integrated with `add_spi_flash` and some checks should probably be added in the integration. If you enable 1x mode, make sure to use an IO definition similar to this: https://github.com/litex-hub/litex-boards/blob/master/litex_boards/platforms/minispartan6.py#L28-L33 and to use the correct `dummy_cycles` value.

DurandA, 2020-05-11:
I see, the 1x mode uses *fast* read/write while I used 0 zero dummy cycles and this explains the offset. I am still not able to write memory from firmware/BIOS but I probably misinterpreted page/sector size from the datasheet.

Do you want me to make a PR to re-enable 1x support? I am thinking of something along the lines of the following (unless you want to accept only `spiflash` and not `spiflash1x`):
```python
try:
    pads = self.platform.request(name + mode)
except ConstraintError as e:
    if mode == "1x":
        pads = self.platform.request(name)
    else:
        raise
```

enjoy-digital, 2020-05-11:
For now the SPIFlash module only support MMAP reads. The writes need to be done by the bitbang interface. Are you trying to do write with MMAP? The 4X mode should have write support over MMAP soon with: https://github.com/enjoy-digital/litex/pull/478.

If you confirm that the reads are working, we could indeed enable it, but we should probably avoid too much try/except, so maybe we should just use the current naming used in the platforms which is probably `spiflash`.

DurandA, 2020-05-11:
Excellent, thanks. I confirm that read is working so I will prepare a PR. I am also preparing some documentation for bare metal programming on the [_litex-buildenv_ wiki](https://github.com/timvideos/litex-buildenv/wiki/Bare-Metal).

I guess I will try to configure 4X mode with my SPI chip in order to load kernels on flash with `litex_term`. Is there an existing script to add length/CRC header on firmware images/kernels when flashing with external tools?

enjoy-digital, 2020-05-11:
For the length/CRC, you probably want to use this: https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/make.py#L417-L423.

closed 2020-05-11 by enjoy-digital

DurandA, 2020-05-13:
There seems to be further issues with the `SPIFlash` on some cores. I switched my CPU from _serv_ to _vexriscv_ and the data appeared to be scrambled again, similarly to when I had the wrong `dummy_cycle` value.

I noticed that the [_versa_ecp5_](https://github.com/litex-hub/linux-on-litex-vexriscv/blob/b5afdeb534236f51d6c2e2f9a89914cb04aa14c6/make.py#L183) is configured with 11 `dummy_cycles`. However, the [Micron N25Q128A datasheet](https://www.micron.com/-/media/client/global/documents/products/data-sheet/nor-flash/serial-nor/n25q/n25q_128mb_1_8v_65nm.pdf) mention user-configurable dummy cycles <= 10 (I assume it's 10 by default).

Increasing `dummy_cycle` from 8 to 9 on my _ecp5_evn_ fixed the issue (the [Macronix MX25L12833F datasheet](https://www.macronix.com/Lists/Datasheet/Attachments/7447/MX25L12833F,%203V,%20128Mb,%20v1.0.pdf) mentions 8 dummy cycles by default).

The SPI core seems to behave differently on some CPU cores which is weird.

PR #505 (2020-05-11): Enable 1x mode on SPI flash
Fix #501