enjoy-digital/litex #512: build fails on Win platform

AnttiLukats, 2020-05-12:
it like:

memusage.py

fails fatal on WIN platform, all the rest seems working!

Please fix or just add option to exlude memusage from build to let the build succeed

enjoy-digital, 2020-05-12:
@AnttiLukats: thanks for reporting, do you have more information on the fail? If it because `readelf` is not present or if the readelf output parsing is failing?

AnttiLukats, 2020-05-13:
Very weird, on different install on fresh new PC the problem does not appear, the difference is that first PC that had this failure used "make" from Cygwin, while the second PC takes make from MSYS2 install folder. This is as much as I can describe the issue. There really is or may be an issue, but it may also just work if lucky.

xobs, 2020-05-13:
Try setting `SHELL` to `C:\Windows\system32\cmd.exe`.  If `make` finds `bash` in your PATH, it will use that as the `SHELL`.  Otherwise, if you're on Windows, it will use `cmd.exe`.  It sounds like it's finding `sh.exe` and using that, but it's not a complete environment so it won't work.

enjoy-digital, 2020-05-15:
This seems to be the case described in the README:
`Note: On Windows, it's possible you'll have to set SHELL environment variable to SHELL=cmd.exe.`

closed 2020-05-15 by enjoy-digital