enjoy-digital/litex #515: Improve standalone cores

enjoy-digital, 2020-05-13:
To continue the discussion from https://github.com/enjoy-digital/litex/pull/511

On Tue, 2020-05-12 at 07:28 -0700, enjoy-digital wrote:
> Thanks, it's merged with some changes: 3a6dd95. I don't think it's
> necessarily a good idea to use a PointToPoint Interconnect or adapt
> the Wishbone address width to the CSR Bus since these changes are
> specific to the use case and will break others, it's probably better
> to let the synthesis tool do the optimization here automatically.

Except it doesn't work that way unfortunately, at least with the stuff
that's in git right now.

The main problem is that since LiteDRAM creates the exported wishbone
interface with an address_size matching the CSR address size, we lose
all the top bits. But the SharedInterconnect uses them for address
decoding, so all CSR accesses fail (I think Vivado even optimizes out
the entire CSR bus as it realizes that the address decode can never
match, the resulting non-working bitfile is ... small).

The "obvious" fixes are problematic:

 - Setting CSR base to 0 means all the csr.h definitions have the wrong
address, unless I fold that base address into the accessors themseslves
but that leads to other problems if I try to use more than one csr-
using module such as litdram and liteeth. In fact, even then, I'm not
sure if that will help, what is the result of comparing unwired bits of
a vector to 0 ?

 - I could make LiteDRAM create a full size address bus instead of one
that only has the bits necessary for the CSR bus. The problem here then
becomes two folds:

    * This does create extra hardware and wires, and the Arty is really
struggling with all those big busses going around. We get the extra
comparators of the address decode in the timing path.

    * The way we are changing our interconnect in Microwatt is that the
"IO" bus where all the dumb slaves like the LiteDRAM CSRs live will
only have a subset of the address bits going down to it. IE. We have a
"master" address decoder that handles the fast path (core -> bram/dram)
combinationally, and an IO leg that has pipeline buffers, and goes to
IOs like CSRs, UART, etc.... This helps a LOT with timing. However I
have only about 24 bits of address going down that bus (again to
relieve pressure on the Vivado router), so having the address decoder
based on the full 32-bit address inside LiteDRAM breaks.

Generally speaking, when creating a standalone IO core like this, we
really don't need any fancy interconnect with address decode in it.
Well it still has enough decoding for the actual CSRs but that's on the
CSR bus itself.

So I see two ways out of this quagmire:

One which Im not sure how to do yet since I don't have much experience
with that part of LiteX, would be to be able to specify separate
"csr_base" addresses for the generated HW vs generated code. That would
allow to leave the HW one at 0 and add an offset to csr.h. I would
*also* have to extend I think the exported wishbone to the full 32-bits
as to be able to match addresses properly even against 0 unless I can
also reduce the size of the intermediary wishbone.

We would still get the extra HW for the SharedInterconnect & address
decoder, but with some hope, the synth tool would get rid of most of it
(fingers crossed) since it would just unwired compare bits to 0... but
maybe not. Unclear. Unless we still have to route the whole bus up and
force the bits to 0 "upstream".

The other approach would be to keep the option of generating P2P
interconnect. But instead of doing it implicitly based on #masters &
#slaves being both 1, make it explicit via a parameter that
litedram_gen can set. That or make the implicit case even more tight:
#masters=1,#slaves=1,cpu=None for example. Out of curiosity, what use
cases are broken here ?

I've attached a LiteX patch the does the second approach (explicit
parameter) along with the corresponding LiteDRAM patch. What do you
think ?

Cheers,
Ben.

enjoy-digital, 2020-05-13:
On Tue, 2020-05-12 at 07:28 -0700, enjoy-digital wrote:
> Thanks, it's merged with some changes: 3a6dd95. I don't think it's
> necessarily a good idea to use a PointToPoint Interconnect or adapt
> the Wishbone address width to the CSR Bus since these changes [...]

Another small nit. You changed my updating of self.mem_map["csr"]
to instead do this:

        self.add_csr_bridge(self.mem_map["csr"] if csr_base is None else csr_base)

This works, however the log printed when generating still has the
0x82000000 base all over rather than the specified one. Not a huge deal
but can be confusing esp. when debugging.

Cheers,
Ben.

enjoy-digital, 2020-05-13:
@ozbenh: we can continue the discussion here.

I just made changes that simplify things should make your LiteDRAM case working again:
https://github.com/enjoy-digital/litex/commit/9d1443c1a8fe5b13d53f23b8165c423f3e6d629a and https://github.com/enjoy-digital/litedram/commit/6fb8396d8e63114eb8708437c3396aaeb65a5abb.

Now when cpu is set `CPUNone`, the `CSRs` will automatically be mapped to `0x00000000`. Before doing the PointToPoint simplification in LiteX, i'd like it to be evaluated in synthesis because i'm pretty sure the resource usage overhead will be minimal: if you only connect the LSB bits of the `wb_ctrl` interface when doing your integration, i'm pretty sure the synthesis tool will be able to already do most of simplifications. So if you could evaluate the resource usage saving with or without the `PointToPointInterconnect` we could decide if it's worth handling this specific case explicitly.

ozbenh, 2020-05-13:
On Wed, 2020-05-13 at 00:56 -0700, enjoy-digital wrote:
> @ozbenh: we can continue the discussion here.
> I just made changes that simplify things should make your LiteDRAM
> case working again:
> 9d1443c and enjoy-digital/litedram@6fb8396.
> Now when cpu is set CPUNone, the CSRs will automatically be mapped to
> 0x00000000. Before doing the PointToPoint simplification in LiteX,
> i'd like it to be evaluated in synthesis because i'm pretty sure the
> resource usage overhead will be minimal: if you only connect the LSB
> bits of the wb_ctrl interface when doing your integration, i'm pretty
> sure the synthesis tool will be able to already do most of
> simplifications. So if you could evaluate the resource usage saving
> with or without the PointToPointInterconnect we could decide if it's
> worth handling this specific case explicitly.

I'll try later tonight but this will still have these issues:

 - csr.h will now have 0-based constants, which will break the init
code. Possible solutions I can think of for that for that would be:

    * Introduce some kind of sw_csr_base argument that gets added to
the constants generated in csr.h but not to the hdl

    * For me to build the offset into the csr_{read,write}_simple
accessors that I provide to the init code (I use externally provided
ones). But that will become problematic when I want both litedram and
some other IO core like LiteEth since they'll both have csr.h mapped at
0 and will need different offsets.

 - The address decoder in SharedInterconnect will still try to match
all the upper bits against '0', won't it ? I don't know what happens in
Verilog in that case, will that actually match ?

Cheers,
Ben.

ozbenh, 2020-05-13:
So I got it working after doing a bunch of changes to hw/common.h so I can provide my own CSR accessors which include the necessary offset.

The address decoding work because of the change you did to litedram: wb_bus = wishbone.Interface(self.bus.address_width) is basically equivalent it seems to wb_bus = wishbone.Interface(), I'm not sure what you intended here, but we get a 29:0 wishbone, which I then pad in my wrapper with '0' so that works. I've taken out the argument to Interface() completely locally. Either we pass the csr address width in which case we end up with a smaller bus or we don't.

The resulting generated bit file is sadly a bit bigger than what I got with a P2P interconnect,
enough for me to worry a bit (more than 120 slices more) for no added functionnality.
`
  +----------------------------+-------+-------+-----------+-------+
 |          Site Type         |  Used | Fixed | Available | Util% |
 +----------------------------+-------+-------+-----------+-------+
-| Slice LUTs                 | 10110 |     0 |     20800 | 48.61 |
-|   LUT as Logic             |  9191 |     0 |     20800 | 44.19 |
-|   LUT as Memory            |   919 |     0 |      9600 |  9.57 |
+| Slice LUTs                 | 10215 |     0 |     20800 | 49.11 |
+|   LUT as Logic             |  9295 |     0 |     20800 | 44.69 |
+|   LUT as Memory            |   920 |     0 |      9600 |  9.58 |
 |     LUT as Distributed RAM |   620 |     0 |           |       |
-|     LUT as Shift Register  |   299 |     0 |           |       |
-| Slice Registers            |  5587 |     0 |     41600 | 13.43 |
-|   Register as Flip Flop    |  5587 |     0 |     41600 | 13.43 |
+|     LUT as Shift Register  |   300 |     0 |           |       |
+| Slice Registers            |  5607 |     0 |     41600 | 13.48 |
+|   Register as Flip Flop    |  5607 |     0 |     41600 | 13.48 |
 |   Register as Latch        |     0 |     0 |     41600 |  0.00 |
 | F7 Muxes                   |   222 |     0 |     16300 |  1.36 |
-| F8 Muxes                   |    66 |     0 |      8150 |  0.81 |
+| F8 Muxes                   |    68 |     0 |      8150 |  0.83 |
 +----------------------------+-------+-------+-----------+-------+
`

enjoy-digital, 2020-05-13:
Thanks for the feedback, can you do a test by changing https://github.com/enjoy-digital/litex/blob/master/litex/soc/integration/soc.py#L831-L832  to:
````python3
register = False,
timeout_cycles = None
````
I'd like to see the impact of the extra-features vs the InterconnectShared. If the extra-feature are causing this, we could disable them for your use case.

ozbenh, 2020-05-13:
A bit better. Still... 77 extra slices and a few more F8 muxes. 

``` +----------------------------+-------+-------+-----------+-------+
 |          Site Type         |  Used | Fixed | Available | Util% |
 +----------------------------+-------+-------+-----------+-------+
-| Slice LUTs                 | 10110 |     0 |     20800 | 48.61 |
-|   LUT as Logic             |  9191 |     0 |     20800 | 44.19 |
-|   LUT as Memory            |   919 |     0 |      9600 |  9.57 |
+| Slice LUTs                 | 10186 |     0 |     20800 | 48.97 |
+|   LUT as Logic             |  9283 |     0 |     20800 | 44.63 |
+|   LUT as Memory            |   903 |     0 |      9600 |  9.41 |
 |     LUT as Distributed RAM |   620 |     0 |           |       |
-|     LUT as Shift Register  |   299 |     0 |           |       |
-| Slice Registers            |  5587 |     0 |     41600 | 13.43 |
-|   Register as Flip Flop    |  5587 |     0 |     41600 | 13.43 |
+|     LUT as Shift Register  |   283 |     0 |           |       |
+| Slice Registers            |  5588 |     0 |     41600 | 13.43 |
+|   Register as Flip Flop    |  5588 |     0 |     41600 | 13.43 |
 |   Register as Latch        |     0 |     0 |     41600 |  0.00 |
-| F7 Muxes                   |   222 |     0 |     16300 |  1.36 |
-| F8 Muxes                   |    66 |     0 |      8150 |  0.81 |
+| F7 Muxes                   |   221 |     0 |     16300 |  1.36 |
+| F8 Muxes                   |    67 |     0 |      8150 |  0.82 |
 +----------------------------+-------+-------+-----------+-------+
```

enjoy-digital, 2020-05-13:
Thanks for the results, so what we could do in the SoC is using `InterconnectPointToPoint` when:
- len(bus_masters) == 1
- len(bus_slaves) == 1
- when there is no address translation, ie the only bus slave is mapped at 0x00000000.

Do you want to look at this or do you want me to implement it?

ozbenh, 2020-05-13:
I can look at it (provided I figure out where to get the right mem_map entry :-)

This leaves us with the problem of CSR base being 0 in the generation of csr.h

Can you think of a way to provide an offset that gets added to the values in csr.h ? I haven't looked at all in that part of the generator at this point.

enjoy-digital, 2020-05-13:
For the origin, you can get it with the origin of the only bus region that will be declared. For the CSR offset, not sure this should be handled directly in LiteX/LiteDRAM since this is related to the way the core is integrated, but when generating the csr.h we could define and use a `CSR_BASE` that could be overriden by the user. I will work on that.

ozbenh, 2020-05-13:
A CSR_BASE would work. The main issue will be if/when mingling multiple csr.h from different generated cores but I can handle that by "wrapping" the inclusion via another file that defines the corresponding CSR_BASE.

ozbenh, 2020-05-13:
BTW. You may want to completely remove the argument to wishbone.Interface() in litedram, I don't think the way you changed it does what you intended (unless I'm mistaken), and if you intention was to pass the actual bus address width (32) then that's not great, we definitely don't want 32-bits addr lines on a 32-bit wide wishbone, only 30. So the default is probably ok to keep as it used to be.

ozbenh, 2020-05-13:
Ok, I tested the patch below and I'm back to the exact same usage as my original patch series, despite not shrinking the wishbones down from 30 bits:
```--- a/litex/soc/integration/soc.py
+++ b/litex/soc/integration/soc.py
@@ -824,7 +824,18 @@ class SoC(Module):
         # SoC Bus Interconnect ---------------------------------------------------------------------
         bus_masters = self.bus.masters.values()
         bus_slaves  = [(self.bus.regions[n].decoder(self.bus), s) for n, s in self.bus.slaves.items()]
-        if len(bus_masters) and len(bus_slaves):
+
+        use_p2p = False
+        if len(bus_masters) == 1 and len(bus_slaves) == 1:
+            n = list(self.bus.slaves)[0]
+            if self.bus.regions[n].origin == 0:
+                use_p2p = True
+
+        if use_p2p:
+            self.submodules.bus_interconnect = wishbone.InterconnectPointToPoint(
+                master = list(bus_masters)[0],
+                slave  = list(self.bus.slaves.values())[0])
+        elif len(bus_masters) and len(bus_slaves):
             self.submodules.bus_interconnect = wishbone.InterconnectShared(
                 masters        = bus_masters,
                 slaves         = bus_slaves,
```
Are you happy with the above ? If yes, I'll send a PR

ozbenh, 2020-05-13:
BTW. If you add CSR_BASE, make it so that it's added to all the various xxx_BASE definitions, not in the accessors themselves. On Microwatt I need to use my own accessors defined externally, but the same accessor pair will have to deal with litedram and other IP blocks we might use (liteeth and sdcard are on the radar :-)

enjoy-digital, 2020-05-13:
@ozbenh: can you review https://github.com/enjoy-digital/litex/pull/518?

enjoy-digital, 2020-05-13:
@ozbenh: the changes are fine yes, i'll maybe do little changes but you can send a PR.

ozbenh, 2020-05-13:
Ok. I also already sent one for the hw/common.h changes. As for CSR_BASE, I need to hit the sack, I'll try them out tomorrow.

ozbenh, 2020-05-14:
CSR_BASE change working fine. The only issue is that I need to use either gcc -D or -include option to force-define as csr.h doesn't #include anything I could use to set that for now (see other discussion for options here). But feel free to merge this for now.

enjoy-digital, 2020-05-14:
@ozbenh: i think everything should be good now, thanks.

closed 2020-05-14 by enjoy-digital