enjoy-digital/litex #520: Wrong SPI flash dummy cycles on VexRiscv

DurandA, 2020-05-13:
There seems to be further issues with the `SPIFlash` on some cores. I switched my CPU from _serv_ to _vexriscv_ and the data appeared to be scrambled again, similarly to when I had the wrong `dummy_cycle` value.

I noticed that the [_versa_ecp5_](https://github.com/litex-hub/linux-on-litex-vexriscv/blob/b5afdeb534236f51d6c2e2f9a89914cb04aa14c6/make.py#L183) is configured with 11 `dummy_cycles`. However, the [Micron N25Q128A datasheet](https://www.micron.com/-/media/client/global/documents/products/data-sheet/nor-flash/serial-nor/n25q/n25q_128mb_1_8v_65nm.pdf) mention user-configurable dummy cycles <= 10 (I assume it's 10 by default).

Increasing `dummy_cycle` from 8 to 9 on my _ecp5_evn_ fixed the issue (the [Macronix MX25L12833F datasheet](https://www.macronix.com/Lists/Datasheet/Attachments/7447/MX25L12833F,%203V,%20128Mb,%20v1.0.pdf) mentions 8 dummy cycles by default).

The SPI core seems to behave differently on some CPU cores which is weird.

xobs, 2020-05-13:
VexRiscv is a little-endian system, so you need to set `endianness="little"`.  For example:

https://github.com/im-tomu/foboot/blob/master/hw/foboot-bitstream.py#L213

```python
        spi_pads = platform.request("spiflash4x")
        self.submodules.lxspi = spi_flash.SpiFlashDualQuad(spi_pads, dummy=platform.spi_dummy, endianness="little")
        self.register_mem("spiflash", self.mem_map["spiflash"], self.lxspi.bus, size=platform.spi_size)
```

DurandA, 2020-05-13:
I used [`add_spi_flash`](https://github.com/enjoy-digital/litex/blob/78413cc09e5a0e944d1c1aa9b8c753413fbc54e3/litex/soc/integration/soc.py#L1204) which set endianness to `self.cpu.endianness` so I don't think this is the issue. Also _SERV_ is little-endian as well.

enjoy-digital, 2020-05-13:
@DurandA: could you do a test with PicoRV32 to have another result and see if it has a similar behavior than one of the previously tested CPU or if it's different? Note that if you are aware of the content of the SPIFlash, you can adjust the dummy byte by looking at the shift you see when doing a `mr` command to the SPIFlash region of the SoC with the BIOS, but switching the CPU should indeed not affect that.

DurandA, 2020-05-13:
_PicoRV32_ behaves like _SERV_ with the SPI flash (= correctly). The issue seems to be with _VexRiscv_ that requires one extra dummy cycle.

DurandA, 2020-05-14:
Note that fixing this issue will require the update of https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/make.py.

DurandA, 2020-05-14:
I saw that in 3fb99b7 you reverted to the read-only _older_ SPI core by default. I am not absolutely sure if I had a this change integrated so I will make the experiments again and edit this post.

From version after 3fb99b7 I also noticed some difference in behaviour when booting a firmware directly without bios from flash (using `--cpu-reset-address 0x20000000` and `--integrated-rom-size 0`) with `dummy_cycles = 8` for all cores (including _VexRiscv_ :flushed:). I modified `.text` and `.rodata` to point to the `spiflash` segment and the firmware runs fine using _VexRiscv_ while it quickly hangs on _SERV_ or _PicroRV32_ (it sometimes print one or more characters of the first puts/printf before crashing).

DurandA, 2020-05-14:
I also tested with _Minerva_ (and failed to compile the BIOS with _Microwatt_). Here is a recap of my results with the _Macronix MX25L12833F_ flash:

| CPU core | variant | adjusted `dummy_cycles` (tested using `mr` from BIOS) | execute in place
| --- | --- | --- | --- |
| VexRixcv | standard | 9 | ✓ (with 8 `dummy_cycles`)
| SERV | standard | 8 | ✗
| PicoRV32 | standard | 8 | ✗
| Minerva | standard | 9 | ✗

Maybe it has something to do with interrupts.

DurandA, 2020-05-14:
Results are inconsistent and vary as I [change the Trellis/nextpnr seed](https://github.com/enjoy-digital/litex/pull/484).

Maybe an issue with clock domain crossing? This is way beyond my expertise. Maybe @daveshah1 has a suggestion on how this could be debugged.

enjoy-digital, 2020-05-14:
@DurandA: interesting, it could also be that IO registers are not used, i could have a look, are you using `SpiFlashDualQuad` or `SpiFlashSingle`?

DurandA, 2020-05-14:
I am using `SpiFlashSingle` and am currently looking into changing the configuration of the flash chip to enable QIO. I have a work-in-progress fork of @gregdavill [ecpprog](https://github.com/gregdavill/ecpprog) to read/write registers so I will report the results once I am amble to use `SpiFlashDualQuad`.

DurandA, 2020-05-20:
I [finally managed](https://github.com/gregdavill/ecpprog/issues/1#issuecomment-631481837) to enable Quad I/O mode on my flash. So far I haven't observed the same issue as in single I/O mode but I will try with more CPU cores/nextpnr seeds just to make sure.

DurandA, 2020-05-20:
For convenience, here is the [diff between `SpiFlashSingle` and `SpiFlashDualQuad`](https://diffy.org/diff/bokjs94ogn3jzpxbuknj98uxr).

davidcorrigan714, 2020-08-08:
I'm wondering if I'm hitting this bug trying to get things running on the Crosslink-NX so I'm trying to understand a bit on how to debug this. Seems like I'll have to insert a Reveal analyzer and watch the traffic over the wishbone bus? Not 100% convinced this is the issue, and there's probably more than one, but ruling it out seems like my next step.

daveshah1, 2020-08-08:
You can also use the built-in LiteScope. There are some docs here: https://github.com/enjoy-digital/litex/wiki/Use-LiteScope-To-Debug-A-SoC

enjoy-digital, 2021-08-24:
I'm not sure it's worth investigating more on this now that the targets have been moved to LiteSPI. The issue was probably related to the SPIFlash core or IO/Clock constraints.

closed 2021-08-24 by enjoy-digital