enjoy-digital/litex #522: Flashing SPI using SFL with bios has no effect

DurandA, 2020-05-13:
Flashing the firmware to SPI flash using `litex_term --kernel firmware.fbi --flash`  doesn't seem to modify the flash content.

I am investigating on the issue as the `fw` bios command works fine.

```
--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LXTERM] Received firmware download request from the device.
[LXTERM] Flashing firmware/firmware.fbi to 0x40000000 (8376 bytes)...
[LXTERM] Upload complete (2.6KB/s).
[LXTERM] Done.
```

DurandA, 2020-05-13:
It works fine. I did not realize that [`erase_flash_sector()`](https://github.com/enjoy-digital/litex/blob/c96f31a9adaec15c846ea23747d933d64314059d/litex/soc/software/libbase/spiflash.c#L73) uses the 0xD8 command while the [Micron N25Q128A datasheet](https://www.micron.com/-/media/client/global/documents/products/data-sheet/nor-flash/serial-nor/n25q/n25q_128mb_1_8v_65nm.pdf) reference this as a block...

Command | Command Code
--- | ---
SE (sector erase) | 0x20
BE 32K (block erase 32KB) | 0x52
BE (block erase 64KB) | 0xD8

closed 2020-05-13 by DurandA

mithro, 2020-05-13:
FYI -- https://github.com/litex-hub/litespi/blob/master/litespi/modules/modules.py

mithro, 2020-05-13:
And https://github.com/litex-hub/litespi/blob/master/litespi/modules/generated_modules.py

DurandA, 2020-05-13:
@mithro Thanks! Strangely, _LiteSPI_ doesn't define a sector size so I wonder how partial erase is done.

mithro, 2020-05-13:
@DurandA -- LiteSPI is only new, I logged https://github.com/litex-hub/litespi/issues/35 to figure out what is happening there.