enjoy-digital/litex #525: litex_setup.py fails on WIN 10

AnttiLukats, 2020-05-15:
fresh install on fresh win10 PC: litex_setup.py goes as far as:
pythondata_software_compiler_rt, then comes >
[installing litex]
python3: cant't open file 'setup,py'

what todo, how to make the install to proceed?

enjoy-digital, 2020-05-15:
We will need more information/help to be able to investigate. Can you check that the clone of litex has been done successfully? Can you try to run the same command manually?

AnttiLukats, 2020-05-15:
clone OK
well I can not execute the command manuall as in the litex/litex subdir there is no setup.py ?

enjoy-digital, 2020-05-15:
Can you list what you have in the directory?  Can you also provide the full installation logs?

DurandA, 2020-05-15:
Do you have any success by using setup.py directly (or using pip)?

```sh
$ pip install git+https://github.com/m-labs/migen.git
$ pip install git+https://github.com/litex-hub/pythondata-software-compiler_rt.git
$ pip install git+https://github.com/enjoy-digital/litex.git
$ pip install git+https://github.com/litex-hub/litex-boards.git
$ pip install git+https://github.com/enjoy-digital/litedram.git
```

AnttiLukats, 2020-05-15:
pip install litex says that requirement already satisfied and does nothing

AnttiLukats, 2020-05-15:
warning: no previously-included files matching '*.py[cod]' found anywhere in distribution
writing manifest file 'pythondata_software_compiler_rt.egg-info\SOURCES.txt'
running build_ext
Creating c:\users\antti\appdata\local\programs\python\python38\lib\site-packages\pythondata-software-compiler-rt.egg-link (link to .)
pythondata-software-compiler-rt 0.0.post6191 is already the active version in easy-install.pth

Installed b:\github\litex\pythondata-software-compiler_rt
Processing dependencies for pythondata-software-compiler-rt==0.0.post6191
Finished processing dependencies for pythondata-software-compiler-rt==0.0.post6191
[installing litex]...
python3: can't open file 'setup.py': [Errno 2] No such file or directory
Traceback (most recent call last):
  File "litex_setup.py", line 127, in <module>
    subprocess.check_call(                                                                                                                                                                                           File "C:\Users\antti\AppData\Local\Programs\Python\Python38\lib\subprocess.py", line 364, in check_call                                                                                                              raise CalledProcessError(retcode, cmd)                                                                                                                                                                         subprocess.CalledProcessError: Command 'python3 setup.py develop' returned non-zero exit status 2.

DurandA, 2020-05-15:
Can you try forcing reinstall (i.e. `--upgrade` or use `pip uninstall` and then `pip install`)?

AnttiLukats, 2020-05-15:
you mean litex_setup.py update ?
tried that, no change, at same place comes "setup.py" not found error

DurandA, 2020-05-15:
I meant `pip install --upgrade git+https://github.com/litex-hub/pythondata-software-compiler_rt.git` (or `pip3 ...`) and then `pip install --upgrade git+https://github.com/enjoy-digital/litex.git`. If is still fails you can try to remove the package eggs manually.

AnttiLukats, 2020-05-15:
OK, tried out no changes :( but you see the litex subdirectory, there is no setup.py
so the install MUST fail ? i do not understand this...

DurandA, 2020-05-15:
Where is your _litex_ directory located? Is it a subfolder of the current directory from where you are launching the python/pip commands or it is `/lib/python3.8/site-packages/litex` or similar (=Python packages default location on Windows)? Can you remove it and try again?

AnttiLukats, 2020-05-15:
Verzeichnis von B:\GitHub\litex\litex

15.05.2020  15:28    <DIR>          .
15.05.2020  15:28    <DIR>          ..
15.05.2020  15:28                 0 1
13.05.2020  11:07    <DIR>          boards
13.05.2020  11:07    <DIR>          build
13.05.2020  11:07    <DIR>          gen
13.05.2020  11:07    <DIR>          soc
15.05.2020  14:44    <DIR>          tools
13.05.2020  11:07               639 __init__.py

I did GIT clone to B:\Github
and launching from B:\Github\litex\

DurandA, 2020-05-15:
Can you try to remove your clone (`B:\Github\litex\`) and do the above commands (pip will clone the repos automatically):

> I meant `pip install --upgrade git+https://github.com/litex-hub/pythondata-software-compiler_rt.git` (or `pip3 ...`) and then `pip install --upgrade git+https://github.com/enjoy-digital/litex.git`. If is still fails you can try to remove the package eggs manually.

This should be easier than debugging your local installation.

enjoy-digital, 2020-05-15:
The issue is probably to install litex, you should only download `litex_setup.py` (and not clone the repository):
````
$ wget https://raw.githubusercontent.com/enjoy-digital/litex/master/litex_setup.py
$ chmod +x litex_setup.py
$ ./litex_setup.py init install --user (--user to install to user directory)
````

AnttiLukats, 2020-05-15:
case closed. Indeed must only get 1 file and not clone reposotory :( sorry for trouble.

closed 2020-05-15 by AnttiLukats