enjoy-digital/litex #528: build does not complete (WIN10)

AnttiLukats, 2020-05-15:
it seems the build process stops after "memusage" part of gatware files are not generated, etc

log:

python3 -m litex.soc.software.mkmscimg bios.bin --little                                                                                                                                                           python3 -m litex.soc.software.memusage bios.elf /b/GitHub/litexx/litex-boards/litex_boards/targets/soc_basesoc_c10lprefkit/software/bios/../include/generated/regions.ld riscv-none-embed                                                                                                                                                                                                                                             ROM usage: 19.53KiB     (61.04%)                                                                                                                                                                                   RAM usage: 1.63KiB      (40.82%)                                                                                                                                                                                                                                                                                                                                                                                                      make: Leaving directory '/b/GitHub/litexx/litex-boards/litex_boards/targets/soc_basesoc_c10lprefkit/software/bios'                                                                                                                                                                                                                                                                                                                    B:\GitHub\litexx\litex-boards\litex_boards> 

this is how far the build proceeds

AnttiLukats, 2020-05-15:
firmware and most of gatware is generated ok, when adding dummy quartus project file then quartus builds ok

enjoy-digital, 2020-05-15:
Are you passing `--build` to the target? (this is now required to build the bitstream).

AnttiLukats, 2020-05-16:
no I did not, assumed all is still working as it was used to work- will try out monday.

AnttiLukats, 2020-05-18:
it was the missing --build, case closed

closed 2020-05-18 by AnttiLukats