enjoy-digital/litex #540: commit 0db350699 breaks `--with-sdcard` build for upstream LiteX

gsomlo, 2020-05-24:
Attempting to build LiteX with `--with-sdcard` (on the nexys4ddr) fails:

```
litex/litex/boards/targets/nexys4ddr.py --csr-data-width 8 --sys-clk-freq 75e6 --with-ethernet --with-sdcard --cpu-type vexriscv --build
```

results in `error: 'SDREAD_BASE' undeclared` and `'SDWRITE_BASE' undeclared` here:

https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/liblitesdcard/sdcard.c#L39

Those two constants were introduced by commit 0db350699 ("Update Litex bios to handle updated litesdcard."). I can only assume they're present in some downstream (Antmicro?) version of LiteX, and build fine there.

Anything I'm missing? @mateusz-holenko @kamilrakoczy -- anything that should have been part of that commit and somehow fell through the cracks?

mateusz-holenko, 2020-05-25:
In https://github.com/enjoy-digital/litesdcard/pull/7 we extended the LiteSDCard core so it's fully usable in Linux (https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/soc_linux.py#L236).

As a part of this work we extended LiteX BIOS sdcard testing capabilities by adding tests for reading/writing actual data. They require additional memory regions (buffers) - `SDREAD` and `SDWRITE` to run.

It looks we missed the `nexys4ddr` platform that already had support for the SD card and didn't update its' definition. Could you take a look at: https://github.com/antmicro/litex/commit/9de63744e1fec85abd2bff8852198fa48e64d7e8 and preferrably test it? It solves the compilation problems for sure, but we didn't test it on an actual HW to see if everything works as expected.

It requires to extend the `ROM` size:
```
litex/litex/boards/targets/nexys4ddr.py --csr-data-width 8 --sys-clk-freq 75e6 --with-ethernet --with-sdcard --cpu-type vexriscv --build --integrated-rom-size=0x10000
```

Once it's confirmed to work fine, we'll PR the fix to the mainline LiteX.

gsomlo, 2020-05-25:
@mateusz-holenko : thanks, the patch does indeed fix the compilation error. However, testing (from the bios prompt) fails: after (successfully) completing `sdinit`, I write some test string to block `0x10`:

```
litex> sdtestwrite 0x10 foobar
SDWRITE:
0000   66 6f 6f 62  61 72 66 6f  6f 62 61 72  66 6f 6f 62   foob arfo obar foob
0010   61 72 66 6f  6f 62 61 72  66 6f 6f 62  61 72 66 6f   arfo obar foob arfo
0020   6f 62 61 72  66 6f 6f 62  61 72 66 6f  6f 62 61 72   obar foob arfo obar
0030   66 6f 6f 62  61 72 66 6f  6f 62 61 72  66 6f 6f 62   foob arfo obar foob
0040   61 72 66 6f  6f 62 61 72  66 6f 6f 62  61 72 66 6f   arfo obar foob arfo
0050   6f 62 61 72  66 6f 6f 62  61 72 66 6f  6f 62 61 72   obar foob arfo obar
0060   66 6f 6f 62  61 72 66 6f  6f 62 61 72  66 6f 6f 62   foob arfo obar foob
0070   61 72 66 6f  6f 62 61 72  66 6f 6f 62  61 72 66 6f   arfo obar foob arfo
0080   6f 62 61 72  66 6f 6f 62  61 72 66 6f  6f 62 61 72   obar foob arfo obar
0090   66 6f 6f 62  61 72 66 6f  6f 62 61 72  66 6f 6f 62   foob arfo obar foob
00a0   61 72 66 6f  6f 62 61 72  66 6f 6f 62  61 72 66 6f   arfo obar foob arfo
00b0   6f 62 61 72  66 6f 6f 62  61 72 66 6f  6f 62 61 72   obar foob arfo obar
00c0   66 6f 6f 62  61 72 66 6f  6f 62 61 72  66 6f 6f 62   foob arfo obar foob
00d0   61 72 66 6f  6f 62 61 72  66 6f 6f 62  61 72 66 6f   arfo obar foob arfo
00e0   6f 62 61 72  66 6f 6f 62  61 72 66 6f  6f 62 61 72   obar foob arfo obar
00f0   66 6f 6f 62  61 72 66 6f  6f 62 61 72  66 6f 6f 62   foob arfo obar foob
0100   61 72 66 6f  6f 62 61 72  66 6f 6f 62  61 72 66 6f   arfo obar foob arfo
0110   6f 62 61 72  66 6f 6f 62  61 72 66 6f  6f 62 61 72   obar foob arfo obar
0120   66 6f 6f 62  61 72 66 6f  6f 62 61 72  66 6f 6f 62   foob arfo obar foob
0130   61 72 66 6f  6f 62 61 72  66 6f 6f 62  61 72 66 6f   arfo obar foob arfo
0140   6f 62 61 72  66 6f 6f 62  61 72 66 6f  6f 62 61 72   obar foob arfo obar
0150   66 6f 6f 62  61 72 66 6f  6f 62 61 72  66 6f 6f 62   foob arfo obar foob
0160   61 72 66 6f  6f 62 61 72  66 6f 6f 62  61 72 66 6f   arfo obar foob arfo
0170   6f 62 61 72  66 6f 6f 62  61 72 66 6f  6f 62 61 72   obar foob arfo obar
0180   66 6f 6f 62  61 72 66 6f  6f 62 61 72  66 6f 6f 62   foob arfo obar foob
0190   61 72 66 6f  6f 62 61 72  66 6f 6f 62  61 72 66 6f   arfo obar foob arfo
01a0   6f 62 61 72  66 6f 6f 62  61 72 66 6f  6f 62 61 72   obar foob arfo obar
01b0   66 6f 6f 62  61 72 66 6f  6f 62 61 72  66 6f 6f 62   foob arfo obar foob
01c0   61 72 66 6f  6f 62 61 72  66 6f 6f 62  61 72 66 6f   arfo obar foob arfo
01d0   6f 62 61 72  66 6f 6f 62  61 72 66 6f  6f 62 61 72   obar foob arfo obar
01e0   66 6f 6f 62  61 72 66 6f  6f 62 61 72  66 6f 6f 62   foob arfo obar foob
01f0   61 72 66 6f  6f 62 61 72  66 6f 6f 62  61 72 66 6f   arfo obar foob arfo

CMD23: SET_BLOCK_COUNT
00000000 00000000 00000000 00000017 00000900 
CMD24: WRITE_SINGLE_BLOCK
00000000 00000000 00000000 00000018 00000900 
CMD12: STOP_TRANSMISSION
00000000 00000000 00000000 00000018 00000900 

litex> 
```

After that, trying to read the data back fails:

```
litex> sdtestread 0x10
SDREAD (0xb0002000) before read:
0000   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   .... .... .... ....
0010   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   .... .... .... ....
0020   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   .... .... .... ....
0030   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   .... .... .... ....
0040   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   .... .... .... ....
0050   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   .... .... .... ....
0060   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   .... .... .... ....
0070   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   .... .... .... ....
0080   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   .... .... .... ....
0090   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   .... .... .... ....
00a0   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   .... .... .... ....
00b0   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   .... .... .... ....
00c0   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   .... .... .... ....
00d0   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   .... .... .... ....
00e0   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   .... .... .... ....
00f0   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   .... .... .... ....
0100   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   .... .... .... ....
[... 53 lines cut ...]
01c0   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   .... .... .... ....
01d0   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   .... .... .... ....
01e0   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   .... .... .... ....
01f0   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   .... .... .... ....

litex> 
```

This happens both with the 64GB SanDisk card that *used* to work pre-0db350699 (and LiteSDCard commit da5467b), and with the 32GB SanDisk that didn't work even before the latest changes.

The `sdtest` bios command also enters an infinite loop trying to read back the first tested block:

```
Reading and checking block 0 (1/5); errors: 0 in 0 blocks; crc errors: 0; timeouts: 0
CMD23: SET_BLOCK_COUNT
00000000 00000000 00000000 00000018 00400900 
CMD17: READ_SINGLE_BLOCK
00000000 00000000 00000000 00000018 00400900 
  dataevt: 00000000 (check #1000001)
  Repeating
```

ad infinitum...

Also, a somewhat related question: if not the nexys4ddr, which board did you use to test?

mateusz-holenko, 2020-05-26:
We'll grab some `nexys4ddr` and test it then (it might take a few days, though).

As for the HW we developed and tested the litesdcard core on, it was:

* Arty A7 + https://store.digilentinc.com/pmod-microsd-microsd-card-slot/
* Arty A7 + https://github.com/antmicro/arty-expansion-board

enjoy-digital, 2020-05-26:
@mateusz-holenko: thanks for ordering a `nexys4ddr`, it will ease things. I also ordered the microsd PMOD last week and should receive it soon (and also ordered other PMODs,  like the I2S one you are using for the Zephyr project).

mateusz-holenko, 2020-05-27:
It took less time than I thought (turns out I'm a pessimist). @enjoy-digital we had some HW in the office, but due to the current situation (working from home) I expected this to be more problematic than it really was.

Anyway, we were able to test it on HW and prepared another fix: https://github.com/antmicro/litex/tree/nexys4ddr-sdcard. It works for us, but could you give it a try, @gsomlo?

That's our output:
```
litex> sdinit
CMD0: GO_IDLE
CMD8: SEND_EXT_CSD, arg: 0x000001aa
00000000 00000000 00000000 00000008 000001aa 
Accepted voltage: 2.7-3.6V
CMD55: APP_CMD
00000000 00000000 00000000 00000037 00000120 
ACMD41: APP_SEND_OP_COND, arg: 70ff8000
00000000 00000000 00000000 0000003f 00ff8000 
CMD55: APP_CMD
00000000 00000000 00000000 00000037 00000120 
ACMD41: APP_SEND_OP_COND, arg: 70ff8000
00000000 00000000 00000000 0000003f 00ff8000 
CMD55: APP_CMD
00000000 00000000 00000000 00000037 00000120 
ACMD41: APP_SEND_OP_COND, arg: 70ff8000
00000000 00000000 00000000 0000003f 00ff8000 
CMD55: APP_CMD
00000000 00000000 00000000 00000037 00000120 
ACMD41: APP_SEND_OP_COND, arg: 70ff8000
00000000 00000000 00000000 0000003f 00ff8000 
CMD55: APP_CMD
00000000 00000000 00000000 00000037 00000120 
ACMD41: APP_SEND_OP_COND, arg: 70ff8000
00000000 00000000 00000000 0000003f 00ff8000 
CMD55: APP_CMD
00000000 00000000 00000000 00000037 00000120 
ACMD41: APP_SEND_OP_COND, arg: 70ff8000
00000000 00000000 00000000 0000003f 00ff8000 
CMD55: APP_CMD
00000000 00000000 00000000 00000037 00000120 
ACMD41: APP_SEND_OP_COND, arg: 70ff8000
00000000 00000000 00000000 0000003f 00ff8000 
CMD55: APP_CMD
00000000 00000000 00000000 00000037 00000120 
ACMD41: APP_SEND_OP_COND, arg: 70ff8000
00000000 00000000 00000000 0000003f 00ff8000 
CMD55: APP_CMD
00000000 00000000 00000000 00000037 00000120 
ACMD41: APP_SEND_OP_COND, arg: 70ff8000
00000000 00000000 00000000 0000003f c0ff8000 
CMD2: ALL_SEND_CID
0000003f 02544d53 41313647 43001af4 4c0129e9 
CID Register: 0x02544d534131364743001af44c0129e9
Manufacturer ID: 0x254
Application ID 0x4d53
Product name: A16GC
CRC: e9
Production date(m/yy): 9/18
PSN: 001af44c
OID: TM
CMD3: SET_RELATIVE_ADDRESS
00000000 00000000 00000000 00000003 12340500 
CMD10: SEND_CID
0000003f 02544d53 41313647 43001af4 4c0129e9 
CMD9: SEND_CSD
0000003f 400e0032 5b590000 734f7f80 0a40001b 
CSD Register: 0x400e00325b590000734f7f800a40001b
Max data transfer rate: 64 MB/s
Max read block length: 512 bytes
Device size: 14 GB
CMD7: SELECT_CARD
00000000 00000000 00000000 00000007 00000700 
CMD55: APP_CMD
[... 100 lines cut ...]
0080   6f 62 61 72  66 6f 6f 62  61 72 66 6f  6f 62 61 72   obar foob arfo obar
0090   66 6f 6f 62  61 72 66 6f  6f 62 61 72  66 6f 6f 62   foob arfo obar foob
00a0   61 72 66 6f  6f 62 61 72  66 6f 6f 62  61 72 66 6f   arfo obar foob arfo
00b0   6f 62 61 72  66 6f 6f 62  61 72 66 6f  6f 62 61 72   obar foob arfo obar
00c0   66 6f 6f 62  61 72 66 6f  6f 62 61 72  66 6f 6f 62   foob arfo obar foob
00d0   61 72 66 6f  6f 62 61 72  66 6f 6f 62  61 72 66 6f   arfo obar foob arfo
00e0   6f 62 61 72  66 6f 6f 62  61 72 66 6f  6f 62 61 72   obar foob arfo obar
00f0   66 6f 6f 62  61 72 66 6f  6f 62 61 72  66 6f 6f 62   foob arfo obar foob
0100   61 72 66 6f  6f 62 61 72  66 6f 6f 62  61 72 66 6f   arfo obar foob arfo
0110   6f 62 61 72  66 6f 6f 62  61 72 66 6f  6f 62 61 72   obar foob arfo obar
0120   66 6f 6f 62  61 72 66 6f  6f 62 61 72  66 6f 6f 62   foob arfo obar foob
0130   61 72 66 6f  6f 62 61 72  66 6f 6f 62  61 72 66 6f   arfo obar foob arfo
0140   6f 62 61 72  66 6f 6f 62  61 72 66 6f  6f 62 61 72   obar foob arfo obar
0150   66 6f 6f 62  61 72 66 6f  6f 62 61 72  66 6f 6f 62   foob arfo obar foob
0160   61 72 66 6f  6f 62 61 72  66 6f 6f 62  61 72 66 6f   arfo obar foob arfo
0170   6f 62 61 72  66 6f 6f 62  61 72 66 6f  6f 62 61 72   obar foob arfo obar
0180   66 6f 6f 62  61 72 66 6f  6f 62 61 72  66 6f 6f 62   foob arfo obar foob
0190   61 72 66 6f  6f 62 61 72  66 6f 6f 62  61 72 66 6f   arfo obar foob arfo
01a0   6f 62 61 72  66 6f 6f 62  61 72 66 6f  6f 62 61 72   obar foob arfo obar
01b0   66 6f 6f 62  61 72 66 6f  6f 62 61 72  66 6f 6f 62   foob arfo obar foob
01c0   61 72 66 6f  6f 62 61 72  66 6f 6f 62  61 72 66 6f   arfo obar foob arfo
01d0   6f 62 61 72  66 6f 6f 62  61 72 66 6f  6f 62 61 72   obar foob arfo obar
01e0   66 6f 6f 62  61 72 66 6f  6f 62 61 72  66 6f 6f 62   foob arfo obar foob
01f0   61 72 66 6f  6f 62 61 72  66 6f 6f 62  61 72 66 6f   arfo obar foob arfo
```

enjoy-digital, 2020-05-27:
@mateusz-holenko: thanks, i'll also test it on hardware. BTW as some point we should probably rework/simplify the LiteSDCard interface/integration but for now it's probably better to get everything working with Linux, we could do this work later when we'll have a reference implementation (i could do it).

enjoy-digital, 2020-05-27:
@mateusz-holenko: i can confirm this is working on Nexys4DDR. I tested with several SDCard and updated the target: https://github.com/enjoy-digital/litex/commit/2efcf8792557e4a71eb195c8793abbc25d5463d, thanks. Let's wait the results from @gsomlo to close this.

gsomlo, 2020-05-27:
@enjoy-digital @mateusz-holenko : thanks, 23fcf87 fixed it for me as well (vexriscv, 8-bit CSRs).

(next on the agenda, I'll try getting it to work on rocket and/or with 32-bit CSRs, and send a PR about that :) ).

closed 2020-05-27 by gsomlo

gsomlo, 2020-05-27:
hmmm... Although hardcoding `sdread` and `sdwrite` to `0x80002*00` here:

https://github.com/enjoy-digital/litex/blob/master/litex/boards/targets/nexys4ddr.py#L59

is not very portable, as it overlaps with `main_ram` on Rocket...

enjoy-digital, 2020-05-27:
@gsomlo: yes we should change that, i just reused un-modified example to be sure it was working. I can have a look.
@mateusz-holenko: can you confirm that the Linux driver uses the `.dts` generated with `json2dts.py `to get these addresses?

gsomlo, 2020-05-27:
@enjoy-digital If you're going to post a fixup patch, please also

```
s/CSR_MMCM_DRP_WRITE_ADDR/CSR_SDCLK_MMCM_DRP_WRITE_ADDR/
```

here: https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/liblitesdcard/sdcard.c#L93

gsomlo, 2020-05-27:
> @enjoy-digital If you're going to post a fixup patch, please also
> 
> ```
> s/CSR_MMCM_DRP_WRITE_ADDR/CSR_SDCLK_MMCM_DRP_WRITE_ADDR/
> ```
> 
> here: https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/liblitesdcard/sdcard.c#L93

Actually, nvm that, it's a bit more involved than just that :) Please see PR #542 instead.

mateusz-holenko, 2020-05-28:
> @mateusz-holenko: can you confirm that the Linux driver uses the `.dts` generated with `json2dts.py `to get these addresses?

That's correct.

enjoy-digital, 2020-05-28:
@mateusz-holenko: ok thanks, i did the changes to use dynamic mapping for `sdread` and `sdwrite`.