enjoy-digital/litex #554: improve Symbiflow/Arty support

enjoy-digital, 2020-06-02:
Just a few notes from the review/changes while merging https://github.com/enjoy-digital/litex/pull/551:
1) Would it be possible to automatically infer the `IBUF`? (more generally, we are probably doing some assumptions like automatic input clock buffer inference when not specified, so to avoid modifying the code just to support Symbiflow, Symbiflow should probably be able to some simple primitive inference).
2) Would it be possible to constraints the clock input pins and propagate the constraints in the design and through the primities (PLL)? (with the current design, it seems the constraint is put on `clk100_ibuf`and not on the clock input pin as done with Vivado, It also seems the clock output of the PLL is constrainted, so that the constraints are not propagated through the PLL).
3) Do we really need phase support on `add_period_constraint`? (This has not been required with others build back-ends). I could understand it's required for now if because of 2) and the fact that you currently need to constraints internal clocks manually, but not sure it will be needed once clocks will be automatically propagated by Symbiflow.
4) We could do changes to the Vivado build backend if this avoid duplications/changes between Vivado and Symbiflow, like using `.sdc`/`.pcf` if supported by Vivado.
5) IIRC the DDR3 and Ethernet is working with Symbiflow, is it planned to integrate it?
6) Is it possible to install the symbiflow toolchain without CONDA?

cc @mithro, @mglb, @kgugala

mithro, 2020-06-02:
> Just a few notes from the review/changes while merging #551:
>
> 1. Would it be possible to automatically infer the `IBUF`? (more generally, we are probably doing some assumptions like automatic input [...]

Yosys should be able to infer `IBUF`s but it is currently not enabled.

> 2. Would it be possible to constraints the clock input pins and propagate the constraints in the design and through the primities (PLL)? (with the current design, it seems the constraint is put on [...]

This will be supported in the future but it will be a while before constraint propagation works.

> 3. Do we really need phase support on `add_period_constraint`? (This has not been required with others build back-ends). I could understand it's required for now if because of 2) and the fact that [...]

You might be right.

> 4. We could do changes to the Vivado build backend if this avoid duplications/changes between Vivado and Symbiflow, like using `.sdc`/`.pcf` if supported by Vivado.

We want to drop the `.pcf` file and the `.sdc` and `.xdc` file should be merged into together (`xdc` is Xilinx's extension to `sdc`.

> 5. IIRC the DDR3 and Ethernet is working with Symbiflow, is it planned to integrate it?

Yes.

> 6. Is it possible to install the symbiflow toolchain without CONDA?

Yes, but it is a pretty involved to do so.

enjoy-digital, 2020-06-03:
Thanks @mithro, i was just trying to list the few things i saw. For now i suggest that we only support  symbiflow toolchain on the Arty (to avoid applying the workarounds to the others targets) and we'll remove the workarounds progressively as things improve with symbiflow.

mithro, 2020-06-03:
Our priority list looks like the following;

Artix "50T boards"
 1. Arty 35T variant.
 1. NeTV2 50T variant.

Artix "100T boards"
 1. Arty 100T variant.

Artix "200T boards"
 1. Nexys Video.
 1. AC701

Kintex Boards
 1. Genesys 2 (Kintex 325T).
 1. KC705.

We may also end up adding Zynq in there somewhere....

mithro, 2020-06-03:
FYI -- Our goal is also to accept designs targeted at Xilinx Vivado unmodified.

enjoy-digital, 2020-11-04:
@mithro: do you have an update on this? Would it be possible to update the Arty LiteX target to remove the workarounds required for the initial Symbiflow support?

mithro, 2020-11-04:
@kgugala and @acomodi should be able to give you an update.

enjoy-digital, 2020-12-01:
We can now close this since the workarounds have been removed with https://github.com/enjoy-digital/litex/pull/704.

closed 2020-12-01 by enjoy-digital