enjoy-digital/litex #560: lx_server: "Protocol not available" on Mac OS X

derekmulcahy, 2020-06-07:
Running `lxserver --udp --udp-ip 192.168.1.50` on Mac OS X 10.14.6 and I get the following error.

```
LiteX remote server
[CommUDP] ip: 192.168.1.50 / port: 1234 / Traceback (most recent call last):
  File "/usr/local/bin/lxserver", line 11, in <module>
    load_entry_point('litex', 'console_scripts', 'lxserver')()
  File "/Users/derek/Documents/fpga/colorlight/litex/litex/litex/tools/litex_server.py", line 189, in main
    server.open()
  File "/Users/derek/Documents/fpga/colorlight/litex/litex/litex/tools/litex_server.py", line 35, in open
    self.socket.setsockopt(socket.SOL_SOCKET, socket_flags, 1)
```

I think the issue is caused by passing a bit mask to setsocketopt(). For the Mac this is an option name and not a mask of multiple options. I fixed it by changing the RemoteServer.open routine to set the two option flags separately.

```
    def open(self):
        if hasattr(self, "socket"):
            return
        self.socket = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
        if hasattr(socket, "SO_REUSEADDR"):
            self.socket.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
        if hasattr(socket, "SO_REUSEPORT"):
            self.socket.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEPORT, 1)
        self.socket.bind((self.bind_ip, self.bind_port))
        print("tcp port: {:d}".format(self.bind_port))
        self.socket.listen(1)
        self.comm.open()
```

xobs, 2020-06-07:
Apropos of nothing, could you see if `wishbone-tool` works? Binary releases are at https://github.com/litex-hub/wishbone-utils/releases and the command you want ought to be:

`wishbone-tool -s wishbone --ethernet-host 192.168.1.50`

derekmulcahy, 2020-06-07:
@xobs wishbone-tool works

enjoy-digital, 2020-06-20:
Thanks, your solution has been integrated with https://github.com/enjoy-digital/litex/commit/13e0852af48accf941d10e61639a1838ed0f0ca0.

closed 2020-06-20 by enjoy-digital