enjoy-digital/litex #561: LXTERM Image Upload Extremely Slow

nvonbulow, 2020-06-08:
I am trying to upload an image to the DE10-nano using LXTERM on Windows. The issue I'm having here is that it's taking well over an hour to upload a ~120k file over 115200 baud.

Adding in some print statements reveal that only one frame is being written every second, and just about all of that time is being spent on receiving the reply.

It's possible that the issue is either with the FPGA gateware or LXTERM since I tried using a different, precompiled gateware (the linux version), which has a 1M baud rate (but still only a 3.7KB/s write speed...)

rob-ng15, 2020-06-09:
Hi,

I am getting 89kbps on Linux with a 1e6 baud rate on a de10-nano.

Rob.

nvonbulow, 2020-06-09:
Thanks for the comparison @rob-ng15.
I will try it out on Linux to see if there's any difference.

Could you send me the bitstream file you used for your test so I can see if there's any difference? 

Thanks,
Nick

rob-ng15, 2020-06-09:
https://github.com/rob-ng15/de10nano

have uploaded the bitstream to here

NOTE: SERIAL UART LOCATIONS

tx = AH9 Arduino i/o IO14
rx = AG11 Arduino i/o IO15

hoonchoi70, 2020-06-15:
Any update on this issue? I have the same problem. FPGA gateware + LITEX term is so slow for uploading...

enjoy-digital, 2020-06-15:
@hoonchoi70, @nvonbulow: can you provide more information about the FPGA target and the lxterm command used?

rob-ng15, 2020-06-15:
I've not heard back about whether the bitstream I posted worked. Doesn't windows limit uart to 115200 baud?

xobs, 2020-06-15:
Windows supports much faster UARTs:

[image]

Note that higher speeds require platform-specific calls. For example, on Linux you need [termios2](https://github.com/npat-efault/picocom/blob/master/termios2.txt), and on macOS you need a special IOKit call. On Windows it's done via the [SetCommState()](https://docs.microsoft.com/en-us/windows/win32/api/winbase/nf-winbase-setcommstate) call, which supports arbitrary baud rates via the [BaudRate](https://docs.microsoft.com/en-us/windows/win32/api/winbase/ns-winbase-dcb#members) struct member.

enjoy-digital, 2020-06-20:
@nvonbulow, @hoonchoi70 with Linux-on-LiteX-Vexriscv the upload speed should be at least 80KB/s. I'm closing since we don't have enough information to be able to help, we need to know which gateware is used, which board and the lxterm command used.

closed 2020-06-20 by enjoy-digital

derekmulcahy, 2020-07-17:
Although this is long closed I have some information to add.

Using lxterm with a FT2232 device to upload I find a constant speed of 3.7KB/s is reported. This upload speed doesn't change if I use a baud rate of 115200 or 460800 (or anything else). Lxterm is running on MacOS Mojave and I am uploading to vexriscv on an ECP5.

If I disable the CRC calculation using the --no-crc flag to lxterm then I see upload speeds of 10.0KB/s for 115,200, 40.0KB/s for 460,800, 77.6KB/s for 921,600 and 146.5KB/s for 2,000,000 baud. With the CRC calculation enabled I always see 3.7KB/s.

I am happy to run any tests that might narrow it down.

derekmulcahy, 2020-07-17:
In litex_term.py:55 I changed sfl_payload_length to the maximum value of 255 and this increased the 2,000,000 baud upload speed to 15.3KB/s when CRC was calculated and 180.2KB/s with --no-crc.

The issue seems to be the turnaround time for the ACK when in CRC mode. It looks like it manages about 60 data frames per second with ACKs and over 700 frames per second without. The --no-crc mode dispenses with all acknowledgments and just blasts the data down the wire!

hoonchoi70, 2020-07-17:
Thanks for the update. Yes you are right. The CRC was the cause of the slow
down. Right now I am using Litex-Term w/o CRC.
BTW, thanks for the Litex. Really enjoy using it! Thanks.

derekmulcahy, 2020-07-17:
Windows has some UART tuning parameters. This [document](https://www.mxshift.com/assets/usb_m8/usb_m8_users_guide.pdf) might help.