enjoy-digital/litex #566: bios: `.data` section placed in rom, wrongly ends up acting like `.rodata`

gsomlo, 2020-06-14:
The `.data` section of the bios (initialized global variables) is *wrongly* placed in `rom` by the [linker.ld](https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/bios/linker.ld#L55) script.

As a consequence, initialized global variables such as [this one](https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/liblitesdcard/spisdcard.c#L271) end up being impossible to modify, leading to buggy bios behavior (`spisdcardstatus` will stubbornly remain set to `1` on Rocket, no matter how many times one tries to write a `0` to that location!!!).

I tried `s/rom/sram/` in [linker.ld](https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/bios/linker.ld#L55), but that's obviously not a good solution, since it results in a hang after the `LiteX` logo is printed at startup :) Need some help figuring out what the proper fix should look like.

Thanks!

gsomlo, 2020-06-15:
a simple reproducer would look like this:

```
diff --git a/litex/soc/software/bios/main.c b/litex/soc/software/bios/main.c
index dd0989ab..aef599d1 100644
--- a/litex/soc/software/bios/main.c
+++ b/litex/soc/software/bios/main.c
@@ -69,6 +69,16 @@ static void boot_sequence(void)
 	}
 }
 
+static int foo = 42; // testing .data writability
+
+static __attribute__((noinline)) int mod_foo(int bar) {
+	if (bar)
+		foo = 32;
+	else
+		foo = 52;
+	return foo;
+}
+
 int main(int i, char **c)
 {
 	char buffer[CMD_LINE_BUFFER_SIZE];
@@ -95,6 +105,10 @@ int main(int i, char **c)
 	printf(" (c) Copyright 2007-2015 M-Labs\n");
 	printf("\n");
 	printf(" BIOS built on "__DATE__" "__TIME__"\n");
+	mod_foo(1);
+	printf(" testing .data write access, should be 32: %d\n", foo);
+	mod_foo(0);
+	printf(" testing .data write access, should be 52: %d\n", foo);
 	crcbios();
 	printf("\n");
 	printf(" Migen git sha1: "MIGEN_GIT_SHA1"\n");
```

and is fixed by @zyp's PR #567 (tested on nexys4ddr with litex+rocket).

enjoy-digital, 2020-06-16:
Status with #567 merged:

* `vexriscv`: Updated and tested.
* `rocket`: Updated and tested.
* `serv`: Updated and tested.
* `blackparrot`: Identical to `rocket`, not tested.
* `cv32e40p`: Identical to `vexriscv`, not tested.
* `minerva`: Identical to `serv`, not tested.
* `picorv32`: Existing copy code updated, not tested.
* `lm32`: Existing copy code updated, not tested.
* `mor1kx`: Still needs to be updated.
* `microwatt`: Still needs to be updated. Seems to also miss `.bss` initialization.

enjoy-digital, 2020-06-19:
Status with #569 merged:

    vexriscv: Updated and tested.
    rocket: Updated and tested.
    serv: Updated and tested.
    blackparrot: Identical to rocket, not tested.
    cv32e40p: Identical to vexriscv, not tested.
    minerva: Identical to serv, not tested.
    picorv32: Existing copy code updated, not tested.
    lm32: Existing copy code updated, not tested.
    mor1kx: Updated and tested.
    microwatt: Still needs to be updated. Seems to also miss .bss initialization.

ozbenh, 2020-06-21:
https://github.com/enjoy-digital/litex/pull/573 for Microwatt... seems to work in sim :)

enjoy-digital, 2020-06-23:
LM32 has been tested and Microwatt has been updated with #573, so this is now fixed on all supported CPUs.

PR #567 (2020-06-16): bios/linker: Place .data in sram with initial copy in rom.
As per #566, `.data` section is currently located in rom, rendering it read-only. This PR moves `.data` to sram, leaving initial state in rom that needs to be copied to sram during startup. This means that every `crt0.S` needs to be updated to fix this.

* `vexriscv`: Updated and tested.
* `rocket`: Updated and tested.
* `serv`: Updated and tested.
* `blackparrot`: Identical to `rocket`, not tested.
* `cv32e40p`: Identical to `vexriscv`, not tested.
* `minerva`: Identical to `serv`, not tested.
* `picorv32`: Existing copy code updated, [... cut]
closed 2020-06-23 by enjoy-digital