enjoy-digital/litex #572: Wishbone 32-bit/8-bit downconverter malfunctioning

madscientist159, 2020-06-19:
I'm working with an 8-bit Wishbone peripheral, and while LiteX seems to insert a bus width converter automatically for my 32-bit system, I'm noticing a significant malfunction.  It looks like the upper three bytes always read zero, as if they are masked off.  Only addresses ending in 2'b11 return any non-zero data from the 8-bit slave.

Is the 32-bit to 8-bit converter regularly tested (or indeed was it tested at all)?  The slave was already reduced to an extremely simple test block and the malfunction persists:

```
case (transfer_state)
	0: begin
		if (wb_cyc & wb_stb) begin
			data_cycle_type <= wb_we;
			address_reg <= addr;
			transfer_state <= 0;
		end
		wb_ack_reg <= 0;
	end
	1: begin
		wb_dat_r_reg <= 8'h5a;
		wb_ack_reg <= 1;
		transfer_state <= 1;
	end
endcase
```

enjoy-digital, 2020-06-20:
Hi @madscientist159,

can you also share the wrapper you used around and the integration code? I'll do some checks.

xobs, 2020-06-20:
Is this related to the fact that a Wishbone address is 30 bits? For example, here's how I wire up the USB bridge:

https://github.com/im-tomu/valentyusb/blob/master/valentyusb/usbcore/cpu/usbwishbonebridge.py#L260-L271

```python
        # Trim off the last two bits of the address, because wishbone addresses
        # are word-based, and a word is 32-bits.  Therefore, the last two bits
        # should always be zero.
        addr_to_wishbone = [
            self.wishbone.adr.eq(address[2:]),
            self.wishbone.dat_w.eq(data),
            self.wishbone.sel.eq(2**len(self.wishbone.sel) - 1)
        ]
        if relax_timing:
            self.sync += addr_to_wishbone
        else:
            self.comb += addr_to_wishbone
```

enjoy-digital, 2020-06-20:
I just did a quick test with a simple 8-bit Wishbone module:

````python3
        from litex.soc.interconnect import wishbone
        from litex.soc.integration.soc import SoCRegion

        class Dummy8bSlave(Module):
            def __init__(self):
                self.bus = bus = wishbone.Interface(data_width=8)

                # # #

                mem  = Memory(8, 8, init=[i for i in range(8)])
                port = mem.get_port(write_capable=True, async_read=True)
                self.specials += mem, port

                self.submodules.fsm = fsm = FSM(reset_state="IDLE")
                fsm.act("IDLE",
                    If(bus.stb & bus.cyc,
                        If(bus.we,
                            NextState("WRITE")
                        ).Else(
                            NextState("READ")
                        )
                    )
                )
                fsm.act("WRITE",
                    port.we.eq(1),
                    port.adr.eq(bus.adr),
                    port.dat_w.eq(bus.dat_w),
                    bus.ack.eq(1),
                    NextState("IDLE")
                )
                fsm.act("READ",
                    port.we.eq(0),
                    port.adr.eq(bus.adr),
                    bus.dat_r.eq(port.dat_r),
                    bus.ack.eq(1),
                    NextState("IDLE")
                )

        self.submodules.dummy_8b_slave = Dummy8bSlave()
        self.bus.add_slave("dummy_8b_slave", self.dummy_8b_slave.bus, SoCRegion(origin=0x20000000, size=0x100))
````

That's a Wishbone slave with a memory in it and when accessing it from the BIOS we get:

````
litex> mr 0x20000000 16
Memory dump:
0x20000000  00 01 02 03 04 05 06 07 00 01 02 03 04 05 06 07  ................
litex> mw 0x20000000 0x12345678
litex> mr 0x20000000 16
Memory dump:
0x20000000  78 56 34 12 04 05 06 07 78 56 34 12 04 05 06 07  xV4.....xV4.....
````

So it seems to be behaving correctly and as @xobs suggests the issue is probably related to the fact that the wishbone bus is word based and not byte based. (not sure having it word based was a right choice, but i fear changing it now would break lots of designs, so we'll probably keep it as it is but make sure to document it).

madscientist159, 2020-06-20:
@enjoy-digital So far no luck on my side -- here's my wrapper:

```
class TestModule(Module, AutoCSR):
    def __init__(self, platform, endianness="big"):
        self.bus    = bus    = wishbone.Interface(data_width=8, adr_width=30)
        self.specials += Instance("test_wishbone_module",
            i_wb_cyc = bus.cyc,
            i_wb_stb = bus.stb,
            i_wb_we = bus.we,
            i_spi_addr = spi_bus_adr,
            i_wb_dat_w = bus.dat_w,
            o_wb_dat_r = bus.dat_r,
            o_wb_ack = bus.ack,
            o_wb_err = bus.err,
            i_peripheral_reset = ResetSignal('sys'),
            i_peripheral_clock = ClockSignal('sys')
        )
        platform.add_source("test_wb_module.v")
```

and instantiation:

```
            self.submodules.testmodule = TestModule(
                platform       = platform,
                endianness     = self.cpu.endianness
            self.add_csr("testmodule")
            testmodule_size = 64*1024*1024
            testmodule_region = SoCRegion(origin=self.mem_map.get("testmodule", None), size=testmodule_size, cached=False)
            self.bus.add_slave(name="testmodule", slave=self.testmodule.bus, region=testmodule_region)
```

enjoy-digital, 2020-06-20:
Thanks @madscientist159, it seems the assignment to `spi_bus_adr` is missing in the code snippet, can you share it?

madscientist159, 2020-06-20:
@enjoy-digital Sure -- it's a trivial combinatorial statement:

`self.comb += spi_bus_adr.eq(bus.adr)`

It's only there since the actual module will need to adjust the addresses, but for this tests it's a straight 1:1 assignment.

madscientist159, 2020-06-20:
@enjoy-digital I have also determined that bus.adr is not stable until the clock *after* bus.stb, which seems contrary to the Wishbone B4 specification.  If I update the HDL above to wait one cycle, the off-by-one error disappears in the read output but the upper byte masking problem remains:

```
mr 0xc8000000 32
Memory dump:
0xc8000000  00 00 00 83 00 00 00 87 00 00 00 8b 00 00 00 8f  ................
0xc8000010  00 00 00 93 00 00 00 97 00 00 00 9b 00 00 00 9f  ................
```

xobs, 2020-06-20:
@madscientist159 The way Wishbone works is addresses are right-aligned. That means for example the `ADR` line will go "0, 1, 2, ...".

Then the `SEL` signal will tell which of the 8-bit words are valid.

Since Wishbone in litex is 32-bit, `SEL` contains four bits. If the litex Wishbone bus were 64-bits wide, `SEL` would contain 8 bits.

For example, to access all 32 bits of logical address 0, the CPU sets `ADR` to `0` and `SEL` to `0b1111`.

To access all 32 bits of logical address 4, the CPU sets `ADR` to `1` and `SEL` to `0b1111`.

To address 16 bits of logical address 2, the CPU sets `ADR` to `0` and `SEL` to `0b1100`.

Many peripherals ignore `SEL` and assume all accesses are 32-bits. In this case, you can assume the "missing" two bits on `ADR` are always `0`.

This bus is incapable of accessing 32-bits from logical address 2.

The Wishbone spec doesn't say whether addresses are left-aligned or right-aligned, only that they shave off enough bits such that it is aligned to the width of the bus.

enjoy-digital, 2020-06-20:
@madscientist159: sorry i have some difficulties understand the issue in your case, `bus.stb` and `bus.adr` are on the same clock cycle. Can you provide a minimal verilog + integration code + results you are expecting, i could look at it.

madscientist159, 2020-06-20:
@enjoy-digital I traced the masking to a single typo in the Verilog file -- the ack register was not connected.  Normally this would generate a warning, but in the LiteX flow it either did not generate a warning or the warning was not highlighted (ended up in the general logging noise).  Apparently something in the tooling decided to lock that register to 1'b1, which then caused the apparent issues with masking.

That said, I'm still seeing an oddity on accessing addresses ending in 2'b00 only:

```
mr 0xc8000000 32
Memory dump:
0xc8000000  83 81 82 83 83 85 86 87 87 89 8a 8b 8b 8d 8e 8f  ................
0xc8000010  8f 91 92 93 93 95 96 97 97 99 9a 9b 9b 9d 9e 9f  ................
```

```
litex> mr 0xc8000012 1
Memory dump:
0xc8000012  92                                               .

litex> mr 0xc8000000 1
Memory dump:
0xc8000000  92                                               .

litex> mr 0xc8000000 1
Memory dump:
0xc8000000  80                                               .

litex> mr 0xc8000012 1
Memory dump:
0xc8000012  92                                               .

litex> mr 0xc8000001 1
Memory dump:
0xc8000001  81                                               .

litex> mr 0xc8000012 1
Memory dump:
0xc8000012  92                                               .

litex> mr 0xc8000002 1
Memory dump:
0xc8000002  82                                               .

litex> mr 0xc8000012 1
Memory dump:
0xc8000012  92                                               .

litex> mr 0xc8000003 1
Memory dump:
0xc8000003  83
```

Note that any attempt to read from `'bxxxx00` reads instead the earlier stale data once, then on re-read the correct data is returned.  This does not show up for any other pattern of the lower two address bits.

mithro, 2020-06-20:
@madscientist159 - It be helpful to link to the exact code you are using and commands you are running so people can reproduce the error you are seeing.

madscientist159, 2020-06-20:
@mithro Fair enough, I'm working on extracting the example now.

madscientist159, 2020-06-20:
@enjoy-digital @mithro 

Python module, inserted into the versa_ecp5.py file

```
class TestModule(Module, AutoCSR):
    def __init__(self, platform, endianness="big"):
        self.bus    = bus    = wishbone.Interface(data_width=8, adr_width=32)
        self.specials += Instance("test_wishbone_module",
            i_wb_cyc = bus.cyc,
            i_wb_stb = bus.stb,
            i_wb_we = bus.we,
            i_wb_addr = bus.adr,
            i_wb_dat_w = bus.dat_w,
            o_wb_dat_r = bus.dat_r,
            o_wb_ack = bus.ack,
            o_wb_err = bus.err,
            i_peripheral_reset = ResetSignal('sys'),
            i_peripheral_clock = ClockSignal('sys')
        )
        platform.add_source("test_wb_module.v")
```

and instantiation in `BaseSoC`:

```
class BaseSoC(SoCCore):
    mem_map = {
        "hostspiflash": 0xc8000000,
    }
    mem_map.update(SoCCore.mem_map)
    def __init__(self, sys_clk_freq=int(75e6), device="LFE5UM5G-45F-8BG381C", with_ethernet=False, with_hostspiflash=True, toolchain="trellis", **kwargs):
    ...
    ...
    ...
        self.submodules.testmodule = TestModule(
            platform       = platform,
            endianness     = self.cpu.endianness
        self.add_csr("testmodule")
        testmodule_size = 64*1024*1024
        testmodule_region = SoCRegion(origin=self.mem_map.get("testmodule", None), size=testmodule_size, cached=False)
        self.bus.add_slave(name="testmodule", slave=self.testmodule.bus, region=testmodule_region)
```

Verilog file:
```
module test_wishbone_module(
		// Wishbone signals
		input wire wb_cyc,
		input wire wb_stb,
		input wire wb_we,
		input wire [31:0] wb_addr,
		input wire [7:0] wb_dat_w,
		output wire [7:0] wb_dat_r,
		output wire wb_ack,
		output wire wb_err,

                input wire peripheral_reset,
                input wire peripheral_clock
	);

	reg wb_ack_reg = 0;
	assign wb_ack = wb_ack_reg;
	reg [7:0] wb_dat_r_reg = 0;
	assign wb_dat_r = wb_dat_r_reg;

	reg [7:0] wishbone_transfer_state = 0;
	reg wb_data_cycle_type = 0;
	reg [31:0] wb_address_reg = 0;
	always @(posedge peripheral_clock) begin
		if (peripheral_reset) begin
			wb_ack_reg <= 0;
			wishbone_transfer_state <= 0;
		end else begin
			case (wishbone_transfer_state)
				0: begin
					if (wb_cyc && wb_stb) begin
						wb_data_cycle_type <= wb_we;
						wb_address_reg <= wb_addr;
						wishbone_transfer_state <= 1;
					end
				end
				1: begin
					wb_dat_r_reg <= (wb_address_reg[7:0] + 8'h80);
					wb_ack_reg <= 1;

					wishbone_transfer_state <= 2;
				end
				2: begin
					if (!wb_stb) begin
						wb_ack_reg <= 0;
						wishbone_transfer_state <= 0;
					end
				end
			endcase
		end
	end
endmodule
```

This is a very simple system that should just work.  I'm using Microwatt on the Versa board:

`./versa_ecp5.py --device=LFE5UM-45F-8BG381C --cpu-type=microwatt --cpu-variant=standard+ghdl --build`

mithro, 2020-06-20:
Could you point us to a git repo which has the exact code you are running? Manually hand patching files is very error prone.

madscientist159, 2020-06-20:
Yes, I can get that together, will probably be more this evening or so.

enjoy-digital, 2020-06-20:
Testing in simulation (with Vexriscv) i get:
````
litex> mr 0x80000000 128
Memory dump:
0x80000000  80 80 80 80 84 84 84 84 88 88 88 88 8c 8c 8c 8c  ................
0x80000010  90 90 90 90 94 94 94 94 98 98 98 98 9c 9c 9c 9c  ................
0x80000020  a0 a0 a0 a0 a4 a4 a4 a4 a8 a8 a8 a8 ac ac ac ac  ................
0x80000030  b0 b0 b0 b0 b4 b4 b4 b4 b8 b8 b8 b8 bc bc bc bc  ................
0x80000040  c0 c0 c0 c0 c4 c4 c4 c4 c8 c8 c8 c8 cc cc cc cc  ................
0x80000050  d0 d0 d0 d0 d4 d4 d4 d4 d8 d8 d8 d8 dc dc dc dc  ................
0x80000060  e0 e0 e0 e0 e4 e4 e4 e4 e8 e8 e8 e8 ec ec ec ec  ................
0x80000070  f0 f0 f0 f0 f4 f4 f4 f4 f8 f8 f8 f8 fc fc fc fc  ................
````

In your core, you are checking that `wb_stb` goes low at the end of the transfer, which i don't think is required by the Wishbone specification (at least LiteX is not doing so: it would make each Wishbone transfer 3 clock cycles, 2 is already enough...), modifying:

````
                                2: begin
					if (!wb_stb) begin
						wb_ack_reg <= 0;
						wishbone_transfer_state <= 0;
					end
				end
````
 to:
````
				2: begin
					wb_ack_reg <= 0;
					wishbone_transfer_state <= 0;
				end
````

gives

````
litex> mr 0x80000000 128
Memory dump:
0x80000000  80 81 82 83 84 85 86 87 88 89 8a 8b 8c 8d 8e 8f  ................
0x80000010  90 91 92 93 94 95 96 97 98 99 9a 9b 9c 9d 9e 9f  ................
0x80000020  a0 a1 a2 a3 a4 a5 a6 a7 a8 a9 aa ab ac ad ae af  ................
0x80000030  b0 b1 b2 b3 b4 b5 b6 b7 b8 b9 ba bb bc bd be bf  ................
0x80000040  c0 c1 c2 c3 c4 c5 c6 c7 c8 c9 ca cb cc cd ce cf  ................
0x80000050  d0 d1 d2 d3 d4 d5 d6 d7 d8 d9 da db dc dd de df  ................
0x80000060  e0 e1 e2 e3 e4 e5 e6 e7 e8 e9 ea eb ec ed ee ef  ................
0x80000070  f0 f1 f2 f3 f4 f5 f6 f7 f8 f9 fa fb fc fd fe ff  ................
````

madscientist159, 2020-06-20:
@enjoy-digital That made it worse, actually:

```
--============= Console ================--

litex> mr 0xc8000000 256
Memory dump:
0xc8000000  00 80 82 83 82 84 86 87 86 88 8a 8b 8a 8c 8e 8f  ................
0xc8000010  8e 90 92 93 92 94 96 97 96 98 9a 9b 9a 9c 9e 9f  ................
0xc8000020  9e a0 a2 a3 a2 a4 a6 a7 a6 a8 aa ab aa ac ae af  ................
0xc8000030  ae b0 b2 b3 b2 b4 b6 b7 b6 b8 ba bb ba bc be bf  ................
0xc8000040  be c0 c2 c3 c2 c4 c6 c7 c6 c8 ca cb ca cc ce cf  ................
0xc8000050  ce d0 d2 d3 d2 d4 d6 d7 d6 d8 da db da dc de df  ................
0xc8000060  de e0 e2 e3 e2 e4 e6 e7 e6 e8 ea eb ea ec ee ef  ................
0xc8000070  ee f0 f2 f3 f2 f4 f6 f7 f6 f8 fa fb fa fc fe ff  ................
0xc8000080  fe 00 02 03 02 04 06 07 06 08 0a 0b 0a 0c 0e 0f  ................
0xc8000090  0e 10 12 13 12 14 16 17 16 18 1a 1b 1a 1c 1e 1f  ................
0xc80000a0  1e 20 22 23 22 24 26 27 26 28 2a 2b 2a 2c 2e 2f  . "#"$&'&(*+*,./
0xc80000b0  2e 30 32 33 32 34 36 37 36 38 3a 3b 3a 3c 3e 3f  >023246768:;:<>?
0xc80000c0  3e 40 42 43 42 44 46 47 46 48 4a 4b 4a 4c 4e 4f  N@BCBDFGFHJKJLNO
0xc80000d0  4e 50 52 53 52 54 56 57 56 58 5a 5b 5a 5c 5e 5f  ^PRSRTVWVXZ[Z\^_
0xc80000e0  5e 60 62 63 62 64 66 67 66 68 6a 6b 6a 6c 6e 6f  n`bcbdfgfhjkjlno
0xc80000f0  6e 70 72 73 72 74 76 77 76 78 7a 7b 7a 7c 7e 7f  ~prsrtvwvxz{z|~.

litex>
```

Here's the full test project:
https://gitlab.raptorengineering.com/tpearson/litex-boards, branch wishbone_8bit_malfunction

I'm testing with:
`./versa_ecp5.py --device=LFE5UM-45F-8BG381C --cpu-type=microwatt --cpu-variant=standard+ghdl --build`

madscientist159, 2020-06-21:
@enjoy-digital I have confirmed that the exact same HDL in the repository I linked works perfectly in 32-bit mode (just by changing the data width parameter from 8 to 32, and address width parameter from 32 to 30).  This should isolate the fault to either the Wishbone 32-bit/8-bit downconverter or Microwatt's byte handling on the Wishbone bus.

enjoy-digital, 2020-06-21:
@madscientist159: thanks, sorry i've not yet been able to have another look, i'll do tomorrow.

madscientist159, 2020-06-21:
@enjoy-digital No problem.  Now that I know the 32 bit access works correctly, I've cobbled together a crude byte-mode access system based on wb_adr and wb_sel with the peripheral configured for 32-bit data mode.  So far it's working just fine.

enjoy-digital, 2020-06-22:
@madscientist159: i found an issue on the optimization i did to avoid doing the downstream access when `sel=0`. The read datapath was not shifting the data in this case, which is still required.

This is fixed with https://github.com/enjoy-digital/litex/commit/0c0689f444f285550304c21ac463a2ab0fd64e67 and has been tested with the attached: [test_wishbone_module.v](https://github.com/enjoy-digital/litex/files/4812957/test_wishbone_module.v.txt)

````
litex> mr 0xc8000000 16
Memory dump:
0xc8000000  80 81 82 83 84 85 86 87 88 89 8a 8b 8c 8d 8e 8f  ................
````

Can you confirm this is also working on your side?

enjoy-digital, 2020-06-24:
Closing since the issue identified in LiteX has been solved. Please re-open is your are still not able to make it work.

closed 2020-06-24 by enjoy-digital